esp8266/Arduino #7334: Interrupt set to be triggered on RISING, actually gets triggered on CHANGE

domingosl, 2020-05-28:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[?]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Using the Arduino IDE with a Wemos D1 mini, and setting an interrupt as follows:

```cpp

const int stop1Pin = 14;
void ICACHE_RAM_ATTR stopMotor();

void setup() {

  pinMode(stop1Pin, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(stop1Pin), stopMotor, RISING);

}

void stopMotor() {
  //DO SOMETHING
}
```

On stop1Pin I have a normally closed button connected to GND. Running the program for the first time stopMotor does not get called as expected, since stop1Pin is grounded. I then push the button breaking the connection to GND and since stop1Pin is internally pulled up the RAISING condition gets triggered and stopMotor gets executed. BUT, as soon as I release the button, so going back to GND (FALLING), stopMotor gets called again.

I don't think I have a problem in hardware since I've been probing to see that the voltage levels are correct, I tried with regular hardware pullups and not using the soft pullups from the example and event debouncing the button in hardware and software, same results. It behaves like the condition is CHANGE and not RAISING. BTW, using FALLING has the same effect.

earlephilhower, 2020-05-28:
This is 99% a hardware problem.  Physical buttons are incredibly noisy.  There be bouncing as the switch makes/breaks contact that will only be visible on an oscilloscope.

The SW Serial library uses both CHANGE and RISING/FALLING and seems to work well for users.

Code inspection shows that the core itself it passing along the mode (CHANGE/RISING/FALLING) properly to the SDK.
https://github.com/esp8266/Arduino/blob/51daecc236fbff0d1ed30a9c63a3af6e8b5c7392/cores/esp8266/core_esp8266_wiring_digital.cpp#L190

I suggest you look at ways of debouncing using SW in your app.  Calling `stopMotor` as the actual IRQ handler is not going to work w/a physical button.

closed 2020-05-28 by earlephilhower

domingosl, 2020-05-28:
@earlephilhower As I mentioned, Im using debounce (soft and hard), I just simplified the code to make more readable. Let me show you the actual ISR function with software debounce:

```
ICACHE_RAM_ATTR void IRStopMotor() {

  if(millis() - lastIRS < 1000)
    return;

  lastIRS = millis();
  
  counter++;
  digitalWrite(motor1Pin, LOW);
  digitalWrite(motor2Pin, LOW);
  moving = 0;
}
```

As you can see I added an exaggerated 1 second debounce. The counter variable (declared as volatile) is printed over serial in the Loop function to monitor how many times the IRS is called. I push the button (rise), it gets called, I release de button (fall) a few seconds after and it gets called again.

earlephilhower, 2020-05-28:
The debounce code above doesn't work.

Consider what happens after the initial 0->1.  Say `lastIRS = 2000` due to it.  Hold the button for 5 seconds.  Now, you release the button and it jumps around and you get a spurious 0->1 transition.  `millis` is now 2000+5000=7000.  `millis() - lastIRQ` = 5000.  That `if` condition fails, and you re-execute, turning off the motor..

I suggest you look for some canned Arduino debounce code.  The 8266 isn't unique with this issue, all controllers with pushbuttons have the same concern.