esp8266/Arduino #7342: [Feature Request]Light BearSSL client with smaller memory usage for simple https queries

luc-github, 2020-05-30:
- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it. (N/A)
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12 nomcu]
- Core Version:			[2.7.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Doing https use a lot of memory with bearSSL even used as WiFiClientSecure compare to AXTLS
just doing a : `client.connect(host, httpsPort) ` use 27272 octet (54% of available memory, It use 20% more memory than AXTLS) which on sample code works but not on `real` application  because often go OutOfMemory

It would be great to have a `light` bear SSL client for simple insecure request, because current version is need really too much memory compare to AXTLS.

I did some bench on MCVE (enclosed) and on my application (ESP3D)
On sample code which only do the client connect
```
AXTLS all SSL cipher
Free mem before 49704
Free mem before connection 49704, used 0
Free mem after connection 32640, used 17064 = 34% of mem

BearSSSL all SSL cipher
Free mem before 50216
Free mem before connection 44008, used 6208
Free mem after connection 22944, used 27272 = 54% of mem 

BearSSSL basic SSL cipher
Free mem before 50216
Free mem before connection 44008, used 6208
Free mem after connection 22944, used 27272
```

On ESP3D which is a reasonable application :
```
AXTLS all SSL cipher
Free mem before 29600
Free mem before connection 29600, used 0
Free mem after connection 12152, used 17448

BearSSSL all SSL cipher
Free mem before 30008
Free mem before connection 23800, used 6208
Free mem after connection 2952, used 27056
```
with 2K memory available the second call usually fail because apps still need to work or even crash the esp due to low memory available

### MCVE Sketch

```

#include <ESP8266WiFi.h>

// force use of AxTLS (BearSSL is now default)
//#define USING_AXTLS
#ifdef USING_AXTLS
#include <WiFiClientSecureAxTLS.h>
typedef axTLS::WiFiClientSecure TSecureClient;
#else
#include <WiFiClientSecure.h>
typedef WiFiClientSecure TSecureClient;
#endif
const char* ssid = "XXXXXXXXX";
const char* password = "XXXXXXX";

const char* host = "notify-api.line.me";//"api.telegram.org";//"notify-api.line.me";//
const int httpsPort = 443;
uint32_t lastsize;

void ConnectServer()
{
    TSecureClient Notificationclient;
    #ifndef USING_AXTLS
    Notificationclient.setInsecure();
    #endif
    uint32_t newsize = ESP.getFreeHeap();
    Serial.printf("Free mem before connection %d, used %d\n", newsize, lastsize-newsize);
    if (!Notificationclient.connect(host, httpsPort)) {
        newsize = ESP.getFreeHeap();
        Serial.printf("Free mem after failing %d used %d\n", newsize, lastsize-newsize);
        return ;
    }
    newsize = ESP.getFreeHeap();
    Serial.printf("Free mem after connection %d, used %d\n", newsize, lastsize-newsize);
    Notificationclient.stop();
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
    lastsize = ESP.getFreeHeap();
    Serial.printf("Free mem before %d\n", lastsize);
    ConnectServer();
    delay(5000);
}

```

I really hope a light WiFiClientSecure could be done for simple insecure queries not using 54% of memory  - AxTLS is now  deprecated and do not support all websites handcheck (e.g. api.telegram.org gave SSL Error code 40) when Bear SSL works but can not be used within an application due to the huge need of memory

I understand BearSSL may need more memory for full SSL support but I think a lighter client could be a good mitigation for some projects  - not only ESP3D

Thank you

JiriBilek, 2020-05-30:
I doubt it is possible without a kind of trade off. Please have a look at https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html#bearssl-wifi-classes.
The article explains why BearSSL needs so much memory and suggests using MFLN extension if you are short of memory. But the extension must be supported by the TLS layer on the server.

earlephilhower, 2020-05-30:
Sorry, as @JiriBilek described the SSL protocol itself requires a 16KB receive buffer unless MFLN is used.  BearSSL allocates all the memory it needs on creation, to avoid later OOM errors.  axTLS may appear to use less memory, but it does extensive memory allocations and deallocations at runtime which was the cause of many, many weird crashes at runtime.

Look at adding MFLN to your code/server (see the examples) as @JiriBilek mentioned.  OTW there's not much to be done.

closed 2020-05-30 by earlephilhower

earlephilhower, 2020-05-30:
One other thing, and this is an absolutely ugly hack, you can limit the SSL protocols to use only simpler key exchange (but this may make the server reject the connection attempt).  Select the `Basic SSL` option in the menus and then you may be able to reduce the BearSSL stack *manually* from the ~6KB down.  My gut says you've got a max of 1.5KB to save there, but I doubt it, and when it crashes it shall crash in spectacular fashion.

luc-github, 2020-05-30:
@JiriBilek @earlephilhower Thank you I will investigate these directions

earlephilhower, 2020-05-30:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/BearSSL_MaxFragmentLength

luc-github, 2020-05-30:
@earlephilhower  @JiriBilek  I have tested and fit perfectly the needs of ESP3D and messenging API webservers - using MFLN 512 works and use less memory than AXTLS, so  
Thank you again ^_^

earlephilhower, 2020-05-30:
Awesome!  Thanks for the update.