esp8266/Arduino #7343: Exception (28): epc1=0x40203876 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000When i upload the file from the form the nodemcu restarts and then in the serial motinor i get the exception.

saqibmanan329, 2020-05-30:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					NODEMCU (ESP 12-E)
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200|

### Problem Description

Detailed problem description goes here.

Exception (28):
epc1=0x40203876 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcd0 end: 3fffffc0 offset: 01a0
3ffffe70:  3fff014c 3ffffe90 3fff02d0 40205f10  
3ffffe80:  00000001 3ffee4e4 3fff02d0 40202f8d  
3ffffe90:  3fff0100 00000001 3ffffee0 00000001  
3ffffea0:  3ffffee0 3ffefefc 00000000 40206760  
3ffffeb0:  00000003 00000001 3ffefed4 4020fd5a  
[... 142 lines cut ...]
    Serial.print("handleFileUpload Name: "); Serial.println(filename);
    fsUploadFile = SPIFFS.open(filename, "w");            // Open the file for writing in SPIFFS (create if it doesn't exist)
    filename = String();
  } else if(upload.status == UPLOAD_FILE_WRITE){
    if(fsUploadFile)
      fsUploadFile.write(upload.buf, upload.currentSize); // Write the received bytes to the file
  } else if(upload.status == UPLOAD_FILE_END){
    if(fsUploadFile) {                                    // If the file was successfully created
      fsUploadFile.close();                               // Close the file again
      Serial.print("handleFileUpload Size: "); Serial.println(upload.totalSize);
      server.sendHeader("Location","/success.html");      // Redirect the client to the success page
      server.send(303);
    } else {
      server.send(500, "text/plain", "500: couldn't create file");
    }
  }
}

```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2020-05-30:
Please decode the stack and fix the title.  We can't help with what you've given.

saqibmanan329, 2020-05-30:
> 
> 
> Please decode the stack and fix the title. We can't help with what you've given.

https://github.com/esp8266/Arduino/issues/7345
sir, check the updated issue and help.

closed 2020-05-30 by earlephilhower