esp8266/Arduino #7344: WDT Reset even after disabling the watchdog timer

pushpa-raj-bhatta, 2020-05-30:
Hi,
I am trying to do serial communication between arduino uno and my esp8266 module. The arduino part is working fine and responds to commands sent through serial monitor. On the esp part I have a Async web server running on port 80. The esp creates its wifi access point and i connect to it to access the required endpoint i.e. 192.168.4.1/getDataFromArduino

This triggers my following function:

`server.on("/fromArduino", HTTP_GET, [](AsyncWebServerRequest *request){
    ESP.wdtDisable();
    //Send a JSON-formatted request with key "type" and value "request"
    //then parse the JSON-formatted response with keys "a" and "b"
    DynamicJsonDocument doc(1024);
    String a;
    String b;
    //Sending the request
    doc["type"] = "request";
    char buffer[1024];
    //serializeJson(doc, buffer); 
    Serial.println("{\"type\": \"request\"}");
    //Serial.println(Serial.readString());
    //Reading the response
    bool messageReady = false;
    String message = "";
    
    while(messageReady == false) { //blocking but that's ok
      if(Serial.available()){
        message = Serial.readString();
        messageReady = true;
      }
      delay(1000);
    }
    delay(3000)
    //Attempt to deserialize the JSON-formatted message
    DeserializationError error = deserializeJson(doc, message);
    if(error) {
      Serial.print(F("deserializeJson() failed: "));
        Serial.println(error.c_str());
        return;
    }
    a = doc["a"].as<String>();
    b = doc["b"].as<String>();
    
    //Prepare the data for serving it over HTTP
    String output = "a: " + a + "\n";
    output += "b: " + b + "\n";
    //Serve the data as plain text
    request->send_P(200, "text/plain", String(output).c_str());
    ESP.wdtEnable(1);
  });`

Now when i connect to the access point provided by esp and then open the endpoint, it restarts the esp with the following message on serial monitor:

` ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
`

The rst cause:4 means watchdog reset which is caused due to the while loop in esp code, I tried using delay, yield, and also ESP.wdtDisable();

devyte, 2020-05-30:
The AsyncWebserver executes in SYS context, i. e. the callbacks execute as system callbacks, and not as part of the CONT or continuation context where setup/loop execute. All code that runs in SYS must follow the SYS restrictions, i. e. can't be blocking, no yield/delay calls, timing constraints, etc. This is explained in tjw docs.
Closing due to user error.

closed 2020-05-30 by devyte

d-a-v, 2020-05-30:
Remove this `char buffer[1024];` which should not be declared in the stack.
delays are not wise in any callback (and especially the `delay(3000)` without semicolon). If there is not enough data on serial, then they should be buffered in the main loop and passed to the callback when they are valid.

This is not a forum, and the issue template was ignored. Closing. Please use esp8266.com for queries like this.