esp8266/Arduino #7352: Devices Bricking after firmware update

arihantdaga, 2020-06-05:
### Basic Infos

- [ x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [  ] I have tested that the issue is present in current master branch (aka latest git).
- [ x ] I have searched the issue tracker for a similar issue.
- [ x ] If there is a stack dump, I have decoded it.
- [ x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12S]
- Core Version:			[d600cc7 - 7 April]
- Development Env:		[Platformio]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[I am not sure]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

Platformio configuration
```

[env]
platform = espressif8266
platform_packages = framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#d600cc7
framework = arduino
board_build.flash_mode = dio
monitor_speed = 115200
upload_speed = 115200
extra_scripts = pre:scripts/pio_pre.py, scripts/pio_main.py
lib_extra_dirs =
    ${common.shared_libdeps_dir}
```

### Problem Description

Recently i started rolling out the update to few of our devices in production and some of them got bricked :o. All of them Received update, worked for some time, but once after a power cycle(Electricity went and came back), they never booted up. I collected the devices and unless i upload code into them again, it won't boot. It happens randomly in just a few devices, while others work fine. I don't know how to produce MCVE, because my complete projects is little big and based on espurna. But i havent changed anything major in code apart from the Arduino core version and building it with platformio instead of arduino ide.

I checked few issues in github regarding XMC Chip, but i verified that my flash is not from XMC, but from WinBond. 
The other thing is, on running esptool flash_id, it shows flash frequency 26Mhz, but in the platformio i did not specify any flash_frequency. can it affect? 
 Is there any other commit after d600cc7on 7th April which could fix this. 
I took a flash dump of the bricked device before uploading new code. 
https://www.dropbox.com/s/wqn06ppn6n8tok2/flash_contents.bin?dl=0

Can any other info help in identifying the issue?

devyte, 2020-06-05:
The final xmc fix was merged afterwards.
I suggest building your project against latest git with the IDE as a first test. I don't know what PIO uses for defaults, but I remember at least one case where the flash frecuency was not correct there, and one case where the user was sure he was building against a recent version, and then the core version printout to serial proved differently.

arihantdaga, 2020-06-06:
I'll try updating few devices with the latest git core. But the issue is very randomly occurring in just few devices. 
I further verified the core version and the flash speed value from sketch compiled using both platformio and Arduino IDE. 
Both shows same value of flash speed (40 MHz). 
```

[002118] [INIT] CPU chip ID: 0x6FC66E
[002120] [INIT] CPU frequency: 80 MHz
[002123] [INIT] SDK version: 2.2.2-dev(38a443e)
[**002127] [INIT] Core version: c3796a4d** (With latest git core. If i run my older firmware, as expected it shows d600cc7)
[002130] [INIT] Core revision: 3279514189
[002134] 
[002135] [INIT] Flash chip ID: 0x1640EF
**[002139] [INIT] Flash speed: 40000000 Hz** (This is retrieved by this function - ESP.getFlashChipSpeed())
[002142] [INIT] Flash mode: DIO
[002145] 
```

However the crystal value returned by esptool.py flash_id is 26Mhz. Is this value and the value returned by ESP.getFlashChipSpeed() supposed to be same ? 
```
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:6f:c6:6e
Uploading stub...
Running stub...
Stub running...
```

Also to conclude, in my case, previously and now the flash speed was same. And building with IDE or platformio isnt making a difference. 

The major thing which has changed is that earlier, before the OTA, the devices were running core version 961b558a91caf9589cdf0daa76a2aafd4a6ec568 (Somewhere in June 26,2019). And the new ota updates have the core version d600cc7 . I remember that espressif sdk has also been updated in the code. Could that be a problem ? Has anyone else not faced this problem yet ?

arihantdaga, 2020-06-06:
Also to add more info. I have made one more change. Earlier i used to have eagle.flash.4m.1m.ld. 
But in the new OTA, i used the custom partition for using 4 sectors for eeprom. 
```
/*
sketch: 1019KB
fs: 992KB
eeprom: 16KB
*/

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0xfeff0
}

/*
Provide both _SPIFFS_ and _FS_ to be compatible with 2.3.0...2.6.0+ and
any library that is using old _SPIFFS_...
*/

PROVIDE ( _SPIFFS_start = 0x40500000 );
PROVIDE ( _SPIFFS_end = 0x405f8000 );
PROVIDE ( _SPIFFS_page = 0x100 );
PROVIDE ( _SPIFFS_block = 0x2000 );
PROVIDE ( _EEPROM_start = 0x405f9000 );

PROVIDE ( _FS_start = _SPIFFS_start );
PROVIDE ( _FS_end = _SPIFFS_end );
PROVIDE ( _FS_page = _SPIFFS_page );
PROVIDE ( _FS_block = _SPIFFS_block );

INCLUDE "local.eagle.app.v6.common.ld"

```
Is it a valid thing to do to update this flash partition during OTA.

arihantdaga, 2020-06-06:
More weird issues, which i have never seen before are coming. 
In few devices in which i dumped with the latest core (using platformio upload, not ota). The devices are working but if i try to do esptool.py erase_flash, it shows Chip erase completed successfully in 0.0 second, but it does not actually erase the flash. 

Also if i use this command to upload the sketch using esptool 
`esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 write_flash 0x0 mybinary.bin`
It shows upload finished. But when esp boots, it is still running an older sketch and not the latest sketch i just uploaded.

earlephilhower, 2020-06-06:
> Also if i use this command to upload the sketch using esptool
esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 write_flash 0x0 mybinary.bin
It shows upload finished. But when esp boots, it is still running an older sketch and not the latest sketch i just uploaded.

Sounds like you're saying the main Espressif tool for uploads isn't working on your devices.  `esptool.py` is directly from their repo, no changes for us.  That would mean you've got something going on other than the Arduino core.  I suggest you pop over to https://github.com/espressif/esptool and drop them an issue.  You'll need the flash vendor, all the board specs, etc.

arihantdaga, 2020-06-06:
@earlephilhower Thank you. What I meant to do was to share here in this issue just more detail regarding what is happening after updating With the latest core. Trying to upload using the esptool was only a thing I tried and found weird results in the device with latest core after ota. 

I am trying to figure out how to reproduce these issues, so we can pinpoint it, because till now they are only occurring randomly in few devices.

earlephilhower, 2020-06-06:
esptool uploads its own SW to do the flashing (unrelated to the Arduino core), so if you are having trouble with it not actually uploading things, there is something very fundamentally wrong with the boards and looking at the core right now probably isn't the best way forward.

I really do think that if you have boards where esptool says it flashes, but doesn't, then that's worth checking with the upstream tool.  If esptool can't write to flash, with a completely different method than the Arduino core, there's a basic HW thing going on...

devyte, 2020-06-06:
There has been one winbond-related change done in the recent pass, and it's for a 16MB version so it doesn't apply here.
I think @earlephilhower is correct: if esptool.py can't seem to flash a sketch reliably on your device, you need to discuss it at the esptool.py repo, and do that before moving forward with anything else.
I don't see anything actionable here at this time, so I'm closing for now. If you do eventually find more info that points to a core issue, please add it here and we'll consider reopening, or open a new issue and add the details there.

devyte, 2020-06-06:
As an afterthought: check your winbond model/id, and trace code places where decisions are made based on it, e. g. 0x1840EF is for W25Q128.

closed 2020-06-06 by devyte