esp8266/Arduino #7359: Host tests: different Arduino.h contents

mcspr, 2020-06-07:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Host tests
- Core Version:			https://github.com/esp8266/Arduino/commit/e70092ca3e7a974f268521d85833ac58187f4649
- Development Env:		esp8266 Arduino Mock

### Problem Description

As noticed in the #6294 , [host tests Arduino.h](https://github.com/esp8266/Arduino/blob/master/tests/host/common/Arduino.h) should be in sync with the one used to build the real firmware.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

This does not cause any immediate issues, just introduces different build settings for host. Some (but not all) examples are:
- boolean typedef
- implicit `#include <cmath>`
- abs redefinition

This may cause more problems in the future, perhaps both headers should use the common source and adjust platform-specific settings.

d-a-v, 2020-06-13:
@mcspr do you intend to propose a PR ?

mcspr, 2020-06-13:
Yes. I have a somewhat working POC with only a cores/esp8266/Arduino.h, just need some time to actually push it

Significant issue with std::abs <-> abs is still there for me, not really sure why cstdlib / stdlib.h / Arduino.h decides to be included in a different order in mocks and in real hardware builds... But, more in the PR itself, with some code examples.

d-a-v, 2020-06-13:
What happens in mock does not follow any official rule. Bend it if you can, to comply with arduino.

mcspr, 2020-06-14:
For the specific issue with abs, problem is with compiler versions. Even with 7.5.0 used by the CI, the following example based on the Arduino.h assumptions is invalid.
```cpp
extern "C" {
#include <stdlib.h>
#include <stdio.h>
}
#undef abs
#define abs(x) ((x > 0) ? (x) : -(x))

#include <algorithm>
int main() {
    printf("%d\n", std::abs(-1));
}
```
```shell
~> g++ --version
g++ (GCC) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

~> g++ -c -o example.o example.cpp
example.cpp: In function ‘int main()’:
example.cpp:6:16: error: expected unqualified-id before ‘(’ token
    6 | #define abs(x) ((x > 0) ? (x) : -(x))
      |                ^
example.cpp:10:25: note: in expansion of macro ‘abs’
   10 |     printf("%d\n", std::abs(-1));
      |                         ^~~
~> ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++ --version
xtensa-lx106-elf-g++ (GCC) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

~> ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++ -c -o example.o example.cpp
~> # builds just fine
```

GCC 10 PR already exposed this issue, because it includes `<cstdlib>` right from the top, stdlib.h now is a proxy-header from the C++ stdlib and not the original C header, as was expected before. So the original code expects to receive abs definition from the stdlib.h, undef it, load cstdlib (implicitly loaded via algorithm header) and have c++ abs. What happens in CI, it loads cstdlib from the get-go, there is no abs to undef by the Arduino.h, but it still sets up it's own definition.

PR #7377 (2020-10-06): Allow test framework to use cores/esp8266/Arduino.h directly
resolve #7359 , @d-a-v

- Removes 'tests/host/common/Arduino.h' in favor of 'cores/esp8266/Arduino.h'
- Clean-up math imports, comment some reasoning behind define / undef
- Add a small example that does not build with the current master

Some minor stuff
- Esp.h change is cosmetic, we might as well use class body instead of this weird forced inline definition
- I also noticed malloc debug include at the end
```cpp
#ifdef DEBUG_ESP_OOM
// reinclude *alloc redefinition because of <cstdlib> undefining them
// this is mandatory for allo [... cut]
closed 2020-10-06 by d-a-v