esp8266/Arduino #7360: Can not connect to some routers

odilonafonso, 2020-06-08:
#### Platform

- Hardware:				[ESP-12]
- Core Version:			[.6.3]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[LOLIN(Wemos) D1 R2 & mini]
- Flash Mode:			[dio]
- Flash Size:                    [2MB/1MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[--before default_reset --after hard_reset]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:              [921600]

### Problem Description
ESP is unable to connect to some routers.
On some routers it works perfectly; others do not.
I was unable to identify the difference between the routers that explains this differentiated operation.
I tried to change the ssid, password.
It does not connect and after 250 seconds it connects and obtain a strange IP. It does not appear to be a LAN IP.

### Sketch

#include <ESP8266WiFi.h>        // Include the Wi-Fi library

const char* ssid     = "CLARO2G";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "mypassword";     // The password of the Wi-Fi network

void setup() {
   Serial.begin(115200);         // Start the Serial communication to send messages to the computer
   delay(10);
   Serial.println('\n');
  
   Serial.println("\n\n");
   WiFi.begin(ssid, password);             // Connect to the network
   Serial.print("Connecting to ");
   Serial.print(ssid); Serial.println(" ...");

   int i = 0;
   int ws;
   while ( (ws = WiFi.status()) != WL_CONNECTED) { // Wait for the Wi-Fi to connect
      delay(2000);
      Serial.print(++i); Serial.print(String(ws)); Serial.print(' ');
   }

   Serial.println('\n');
   Serial.println("Connection established!");  
   Serial.print("IP address:\t");
   Serial.println(WiFi.localIP());         // Send the IP address of the ESP8266 to the computer
}

void loop() {}

### Debug Messages

mode : sta(5c:cf:7f:36:89:dc) + softAP(5e:cf:7f:36:89:dc)
add if0
Connecting to CLARO2G ...
wifi evt: 8
wifi evt: 2
16 26 scandone
state: 0 -> 2 (b0)
36 state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CLARO2G, channel 6
dhcp client start...
wifi evt: 0
46 56 66 76 86 96 106 116 126 136 wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
146 156 166 176 186 196 206 216 226 236 246 256 wifi evt: 7
......
...... (similar debug msgs)
......
wifi evt: 7
wifi evt: 7
2426 2436 2446 2456 2466 2476 2486 2496 ip:169.254.221.137,mask:255.255.0.0,gw:0.0.0.0
wifi evt: 3
2506 

Connection established!
IP address:	169.254.221.137
wifi evt: 7
wifi evt: 7

odilonafonso, 2020-06-08:
This is a debug messages for another router that I connect with the same code (changing ssid).

bcn 0
del if1
usl
mode : sta(5c:cf:7f:36:89:dc)
add if0
Connecting to PONESA-3G ...
wifi evt: 8
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with PONESA-3G, channel 7
dhcp client start...
wifi evt: 0
ip:192.168.2.103,mask:255.255.255.0,gw:192.168.2.1
wifi evt: 3

Connection established!
IP address:	192.168.2.103
pm open,type:2 0

odilonafonso, 2020-06-08:
I accessed the configuration of the CLARO2G router, to which the ESP was not being able to connect and manually placed it to use channel 7.
Got connection!
But I didn't understand why.

bcn 0
del if1
usl
mode : sta(5c:cf:7f:36:89:dc)
add if0
Connecting to CLARO2G ...
wifi evt: 8
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CLARO2G, channel 7
dhcp client start...
wifi evt: 0
ip:192.168.0.113,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 3

Connection established!
IP address:	192.168.0.113
pm open,type:2 0

Tech-TX, 2020-06-08:
The biggest issue I've had is the modem over-driving the router.   I think the default WiFi.setOutputPower(x) is set to 17dBm,  and I  had to drop that to 5dBm to get reliable connection to my router. 

That's one thing to check,  In any case.

odilonafonso, 2020-06-08:
I did not know this method WiFi.setOutputPower (x).
Please know where I find Wifi class documentation with the methods implemented for ESP8266?

devyte, 2020-06-08:
@odilonafonso readthedocs, but it's probably incomplete.
The recommendation is always to look at a function/class interface to see what is available for use. 
That means look at the .h file.

devyte, 2020-06-08:
The times I have seen this it's due to a bad wifi config on board. You have to wipe the flash when downloading the sketch.

odilonafonso, 2020-06-09:
Grateful for the @devyte  help, but I didn't understand what you mean. I have already tried to rewrite the firmware with the three options offered by the IDE:

"Only sketch"
"Sketch + Wifi settings"
"All flash Contents".

I've already done another test; without modifying anything on the router, just adding any channel in the call to "Wifi.begin ()" and that sometimes works, other times it doesn't.
On the other hand, one of my routers accepts connection requests in any way that ESP is configured.

I really don't know what to conclude, apparently it is a hardware failure and not a software failure.
I just want to be sure about that.

I'm thinking of adding a loop to the call Wifi.begin () adding one channel after another, until I get a connection.

Tech-TX, 2020-06-09:
> I did not know this method WiFi.setOutputPower (x).
> Please, do you know where I find Wifi class documentation with the methods implemented for ESP8266?

In my code it's usually
**WiFi.setOutputPower (5);**
(5dBm) due to my router.  At higher levels I get dropped packets.  I *think* the allowed range for output power is 0 to 20.5, default 17.

TD-er, 2020-06-09:
Just an idea... would it make sense to make this output power dynamic, based on the RSSI of the AP you want to connect to?

odilonafonso, 2020-06-09:
> @odilonafonso readthedocs, but it's probably incomplete.
> The recommendation is always to look at a function/class interface to see what is available for use.
> That means look at the .h file.

I always look at the .h source and often the .cpp

odilonafonso, 2020-06-10:
I made some changes. I added a channel to Wifi.begin ().
Apparently it is not used.

### New sketch

#include <ESP8266WiFi.h>        // Include the Wi-Fi library

const char* ssid     = "CLARO2G";         // The SSID (name) of the Wi-Fi network you want to connect to
const char* password = "meiadeseda";     // The password of the Wi-Fi network

void setup() {
   Serial.begin(115200);         // Start the Serial communication to send messages to the computer
   delay(10);
   Serial.println("\n\n");
   WiFi.mode(WIFI_STA);

   int i = 0;
   int ws;
   long tini = millis();
   long tnow;
   #define CHINI 2
   #define CHEND 9
   #define INTERVAL 5000
   int chan = CHINI;
   int retrys;
   #define NTRYS 3
   while (true) {
      Serial.println("Connecting to " + String(ssid) + " Channel:" + String(chan));
      WiFi.begin(ssid, password, chan);             // Connect to the network
      retrys = 0;
      while ( (ws = WiFi.status()) != WL_CONNECTED && retrys < NTRYS) { // Wait for the Wi-Fi to connect
         tnow = millis();
         if ( tnow - tini > INTERVAL ) {
            ++i;
            Serial.println("Not connected after " + String(i*INTERVAL/1000) + "secs status:" + String(ws));
            tini = tnow;
            ++retrys;
         }
         yield();
      }
      if (ws == WL_CONNECTED) break;
      chan = (chan == CHEND ? CHINI : chan+1);
   }

   Serial.println('\n');
   Serial.println("Connection established!");  
   Serial.print("IP address:\t");
   Serial.println(WiFi.localIP());         // Send the IP address of the ESP8266 to the computer
}

void loop() {}

### Debug messages

SDK:2.2.2-dev(38a443e)/Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af

Connecting to CLARO2G Channel:2
scandone
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CLARO2G, channel 6
dhcp client start...
wifi evt: 0
Not connected after 5secs status:6
Not connected after 10secs status:6
pm open,type:2 0
Not connected after 15secs status:6
Connecting to CLARO2G Channel:3
sl
scandone
state: 5 -> 2 (b0)
usl
state: 5 -> 2 (7c0)
wifi evt: 1
STA disconnect: 7
reconnect
state: 2 -> 0 (0)
sl
Not connected after 20secs status:6
scandone
state: 0 -> 2 (b0)
usl
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CLARO2G, channel 6
dhcp client start...
wifi evt: 0
Not connected after 25secs status:6
Not connected after 30secs status:6
Connecting to CLARO2G Channel:4
......

I don't know why I get this message:
`connected with CLARO2G, channel 6`

Why ? Why does the message say ESP is connected?
I dont understand

TD-er, 2020-06-10:
It would make the source a lot more readable if you place 3 back-ticks before and after the source code block.

And the AP you connect to determines the WiFi channel, not the client that tries to connect to it.
When calling `WiFi.begin` with a MAC address and channel, you force the connect routine to try those first.
Typical use case for that is when you picked the best AP from a WiFi scan or lost connection and want to make a quick reconnect to the last known AP.

devyte, 2020-06-15:
@odilonafonso in your case, Wifi.begin() means connect the ESP's wifi station + establish a dhcp IP address. Your last log says `connected with CLARO2G, channel 6`, which means the wifi connection was established, but then I see that the ESP's dhcp client times out waiting for your dhcp server to offer.
Such things are support questions, please request peer help at a community forum like esp8266.com or stackoverflow.
Closing.

closed 2020-06-15 by devyte

odilonafonso, 2020-06-16:
@devyte Why did you close? The problem has not been solved.

> @odilonafonso in your case, Wifi.begin() means connect the ESP's wifi station + establish a dhcp IP address. Your last log says `connected with CLARO2G, channel 6`, which means the wifi connection [...]

The message `connected with CLARO2G, channel 6` is not mine. Is not from my sketch.
I receive "not connected".

I am just questioning why the message "connected to CLARO2G" arrives from "somewhere" and meanwhile I get "disconnected" staus.
How to solve this?

odilonafonso, 2020-06-16:
@devyte this 

> And I still don't understand what the debug message is:
> 
> > `connected with CLARO2G, channel 6`
> 
> That must have probably been sent by the ESP8266WiFi library.
> How can the library send this message and at the same time return status 6?

This seems to me to be a software problem. There is no point in the library sending a "CONNECTED" log message and at the same time sending "DISCONNECTED" status.

TD-er, 2020-06-16:
As a matter of fact, I do have also lots of reports where some units enter some kind of loop where they constantly report disconnect events and only remain connected for a very short while.
So that's why I'm also following this discussion here.

Right now I don't have any clue as to where I have to look for a fix to this.
This report states it depends on node vs. AP, but I've also seen reports of some nodes unable to be used and other that can be used with the same build.

devyte, 2020-06-16:
@TD-er that sounds like a different thing. The logs here show a dhcp timeout.

devyte, 2020-06-16:
@odilonafonso I didn't say the message comes from your sketch. What I said was:
> in your case, Wifi.begin() means connect the ESP's wifi station + establish a dhcp IP address

Both steps must complete to have a correct WiFi status. The first part (wifi connection) produces the connect message on channel 6. The second part (getting a dhcp IP from your dhcp server) times out and fails, so then the connection process resets and restarts.
I am telling you the "what". For help with "how" to investigate and fix it I said:
> please request peer help at a community forum like esp8266.com or stackoverflow.

odilonafonso, 2020-06-16:
What I can't understand is why Wemos is able to connect 100% of the time to other routers and with this (CLARO2G), the connection is not established regularly.
And to this router, to which I am unable to connect regularly, with other devices (notebooks, smartphones) the connection is made without problems, thus ruling out being a problem in the router.
Well, @devyte, if you think that this is not a problem to be reported on this forum, OK, let's disregard my problem.
But I doubt that another forum is more specific than that forum to solve this problem.

TD-er, 2020-06-16:
> @TD-er that sounds like a different thing. The logs here show a dhcp timeout.

I will try to add some logging to see if 'my' problem is actually different.
As soon as I know more, I will open an issue for it.

devyte, 2020-06-16:
@odilonafonso what I am telling you is that you have a dhcp timeout. Why that happens with the esp is something that you need to investigate, e. g. with wireshark, which is off topic here, because this is an issue tracker and not a general support forum. If the result of that investigation is that you see something wrong in our core (emphasis on this last part), then open an issue here, explain your findings, and we'll look into how to fix it. 
BTW, there have been several reports of compatibility issues between the esp and various routers. From what I've seen, it's usually some router parameter that the esp doesn't like. What that means is that having other devices connect successfully to your router doesn't prove much beyond that the router is alive and responsive. 
Having said that, I don't remember off the top of my head compatibility issues related to dhcp, but that's why you have to investigate your case with your specific router.

d-a-v, 2020-06-16:
Netdump might be useful to monitor what is sent and received on the esp.

odilonafonso, 2020-06-16:
I tried with two other wemos. Same problem.
It works with one router, it doesn't work with the other.

### Debug log ###
bcn 0
del if1
usl
mode : sta(84:f3:eb:58:41:62)
add if0
Connecting to CLARO2G Channel:2
wifi evt: 8
scandone
no CLARO2G found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
wifi evt: 2
Not connected after 5secs status:6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

> `**connected with CLARO2G, channel 1**`

dhcp client start...
wifi evt: 0
Not connected after 10secs status:6
Not connected after 15secs status:6
Connecting to CLARO2G Channel:3
pm open,type:2 0
scandone
state: 5 -> 2 (b0)
usl
state: 5 -> 2 (7c0)
wifi evt: 1
STA disconnect: 7
reconnect
state: 2 -> 0 (0)

What means this log message  ? 
Why WiFi.status() did return WL_DISCONNECTED ?
The Wifi log message is clear: `**connected with CLARO2G, channel 1**`
Why connected to **channel 1**, when I request the use of channel: 2?

I used the third argument from WiFi.begin (). I specified the channel to use.

Okay, it could be something else to be investigated, but it's clear from the documentation:

WiFi.begin (ssid, pass, **channel**)

The channel must be used. Or not ?

 
### WiFi.status() Returns ###
    WL_NO_SHIELD: assigned when no WiFi shield is present;
    WL_IDLE_STATUS: it is a temporary status assigned when WiFi.begin() is called and remains active until 
                    the number of attempts expires (resulting in WL_CONNECT_FAILED)
                    or a connection is established (resulting in WL_CONNECTED);
    WL_NO_SSID_AVAIL: assigned when no SSID are available;
    WL_SCAN_COMPLETED: assigned when the scan networks is completed;
    WL_CONNECTED: assigned when connected to a WiFi network;
    WL_CONNECT_FAILED: assigned when the connection fails for all the attempts;
    WL_CONNECTION_LOST: assigned when the connection is lost;
    WL_DISCONNECTED: assigned when disconnected from a network; 

    Library statuses – Constants   Value
      WL_NO_SHIELD         255
      WL_IDLE_STATUS       0
      WL_NO_SSID_AVAIL     1
      WL_SCAN_COMPLETED    2
      WL_CONNECTED         3
      WL_CONNECT_FAILED    4
      WL_CONNECTION_LOST   5
      WL_DISCONNECTED      6

odilonafonso, 2020-06-16:
What means: 

**error: pll_call exceeds ems!!!**

### Debug log ###
connected with CLARO2G, channel 1
dhcp client start...
wifi evt: 0
Not connected after 40secs status:4
Not connected after 45secs status:4
Connecting to CLARO2G Channel:5
**error: pll_cal exceeds 2ms!!!**
sl
scandone
state: 5 -> 2 (b0)
usl
state: 5 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 204
Not connected after 50secs status:4
sl
scandone
state: 0 -> 2 (b0)
usl
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with CLARO2G, channel 1
dhcp client start...
wifi evt: 0
Not connected after 55secs status:4
Not connected after 60secs status:4
Connecting to CLARO2G Channel:6
**error: pll_cal exceeds 2ms!!!**
sl
scandone

d-a-v, 2020-06-16:
`error: pll_cal exceeds 2ms!!!`: #2566 and [others](https://github.com/esp8266/Arduino/search?q=pll_cal&type=Issues)

TD-er, 2020-06-16:
@odilonafonso 
The channel parameter is useless when trying to connect a STA interface to an access point.
You simply cannot set the channel of the STA interface on the ESP8266 or ESP32. (except when running in promiscuous mode, but that's a completely different topic)

It can be used as a hint to speed up connecting, but in the end it is the access point you try to connect to that determines the channel to be used.

odilonafonso, 2020-06-16:
> `error: pll_cal exceeds 2ms!!!`: #2566 and [others](https://github.com/esp8266/Arduino/search?q=pll_cal&type=Issues)
> 
> > The channel must be used. Or not ?
> 
> You don't need to specify it.
> 
> > What means this log message ?
> 
> They are generated by the closed source firmware.
> 
> What version of the arduino core do you use (first line displayed on serial).
**Core:2.6.3=20603000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-16-ge23a07e/BearSSL:89454af**
> (well it seems to be 2.6.3. Can you work with latest release (currently 2.7.1) ?)
**Ok, I will try this version.**
> Did you performed a full flash-erase (given you have brand new chips/boards)?
**Yes, I tried the 3 options**
> Did you try another way to power your esp ?
**No. What other way?**
> What is your esp ? afaik wemos d1 does not exist with 2MB flash chips.
**"LOLIN(WEMOS) D1 R2 & Mini"**
**Flash Size:4MB(FS:2MB OTA:~1019Kb)**

odilonafonso, 2020-06-16:
> The channel parameter is useless when trying to connect a STA interface to an access point.
> You simply cannot set the channel of the STA interface on the ESP8266 or ESP32. (except when running in promiscuous mode, but that's a completely different topic)

Well, then why is there this option in the function call?
When should this argument (channel) be used?
This was just my attempt, I do not believe it is the cause of the bug.

whogarden, 2022-09-27:
Hello odilonafonso
Same pb with my ESP8266 12F, I can't connect to a new Internet service provider router.
But an ESP32 can work without pb!
Did you solved your pb?

whogarden, 2022-09-28:
Using the sketch New sketch on 10 Jun 2020.
Can you tell me what to add?
Thank you.

TD-er, 2022-09-28:
You mean the code mentioned here in this comment?
https://github.com/esp8266/Arduino/issues/7360#issuecomment-641668043

whogarden, 2022-09-29:
Thank you TD-er
add : 
``
//  change Radio Type from 802.11n to 802.11g
//  PHY_MODE_11B    = 802.11b
//  PHY_MODE_11G    = 802.11g
//  PHY_MODE_11N    = 802.11n
WiFi.setPhyMode(WIFI_PHY_MODE_11G);
``
And it works.