esp8266/Arduino #7365: My Arduino has stopped working recently  - this appears to be an issue with esp board packages

rosco9999, 2020-06-10:
Arduino IDE has been working fine, I cant remember exactly when I last used it but certainly within 5 days.  Now the splash screen appears then disappears.

When I remove the arduino15 directory this cures the problem, however when I try to add the Esp board back in I get the following output and the issue returns again - I am trying to add the board with this URL http://arduino.esp8266.com/stable/package_esp8266com_index.json

**I suspect there may have been an update to the json file recently that is causing the problem.**

this is the error I am getting when I search for the board:

java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base$9.onIndexesUpdated(Base.java:1400)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:148)
	... 1 more
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base.openBoardsManager(Base.java:1418)
	at processing.app.Base$10.actionPerformed(Base.java:1441)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

MachineThing, 2020-06-10:
I'm getting this problem too.

d-a-v, 2020-06-10:
With HTTPS too ?
 https://arduino.esp8266.com/stable/package_esp8266com_index.json

rosco9999, 2020-06-10:
I can get to the json page ok in a browser, but when I add it to the custom boards in the ide it craps out for me.

Just checked with the https - same story for me

https://arduino.esp8266.com/stable/package_esp8266com_index.json

jandrassy, 2020-06-10:
it is not a problem with esp8266 boards package

rosco9999, 2020-06-10:
Thanks Juraj – I think you may be right, I just tried another custom board url and I get the same issue.  I have reinstalled java also.  Strange that this issue just appears without any package installs on my computer.  I will investigate further and post back if I find anything useful.

Thanks for the prompt reply’s everybody – appreciated!!

James

rosco9999, 2020-06-10:
This looks like it is fixed now – Weird  perhapsa problem somewhere else in the chain.

Thanks anyway everybody

James