esp8266/Arduino #7374: esp8266 wifi BearSSL Secure Server Class

alirezash12345, 2020-06-13:
I want to use setBufferSizes for BearSSL Secure Server Class which is mentioned here: https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/server-class.html I am not sure how to do it and I couldn`t find any document about this application. can annyone help me about this?

earlephilhower, 2020-06-13:
The template was ignored and this is a "How do I..." question.  Closing.  Please refer to the MFLN example.

closed 2020-06-13 by earlephilhower

alirezash12345, 2020-06-14:
How do I use  setBufferSizes(int recv, int xmit) in ESP8266WebServer for serving to clients?
I need to use esp8266 as a server and I am using ESP8266WebServer for serving to clients. In this case I don`t know where to use setBufferSizes(int recv, int xmit) so that  I can set the size of memory buffers when serving to clients.