esp8266/Arduino #7376: Error running cers-from-mozilla.py

DrakoPD, 2020-06-13:
### Platform

- Hardware:				[ESP-12]
- Development Env:		[VS Code: Platformio]
- Operating System:		[Windows 10]

### Problem Description

I already add openssl to the path (like this issue #7300 ), but I get a new error:

```
Traceback (most recent call last):
  File "certs-from-mozilla.py", line 68, in <module>
    call( arCmd )
  File "C:\Users\<myuser>\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Users\<myuser>\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\<myuser>\AppData\Local\Programs\Python\Python38-32\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] El sistema no puede encontrar el archivo especificado
```
It says is missing "ar" or "arCMD", but i don't find any references to this files on internet.

lollotek, 2020-06-13:
@DrakoPD 
Here someone asked for similar issue. the ar.exe is on tools folder od esp8266 library.
https://github.com/esp8266/Arduino/issues/7300#issuecomment-627772485

DrakoPD, 2020-06-13:
@lollotek 
thanks, I don't understand what he was saying. Do I need to add to the path?

DrakoPD, 2020-06-13:
I already found the folder, is different a little bit for me because I'm using PlarformIO on VS code and windows:

%userprofile%\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\bin

Now, cers-from-mozilla.py runs without error.

closed 2020-06-13 by DrakoPD