esp8266/Arduino #7380: request new function in EEPROM.h

MrCrazy1376, 2020-06-15:
Hello 
can you add "forceCommit" function into EEPROM.h library
because i always map EEPROM array with pointer into my structure and had to edit commit function because _dirty flag
i tried to use extend eeprom class to my class but it throw run time crash

something like this

```
bool EEPROMClass::forceCommit() {
  _dirty = true;
  return commit();
}
```

sorry about my bad English :)

earlephilhower, 2020-06-15:
Why not make a pull request (PR) for this, @MrCrazy1376 ?

devyte, 2020-06-15:
Before making a pull request, can you please clarify your use case? What you propose shouldn't be needed. Also, a forced commit is possible by just calling getDataPtr() followed by commit().

MrCrazy1376, 2020-06-16:
> Why not make a pull request (PR) for this, @MrCrazy1376 ?

I'm new in github world
And github banned my country :(

MrCrazy1376, 2020-06-16:
> Before making a pull request, can you please clarify your use case? What you propose shouldn't be needed. Also, a forced commit is possible by just calling getDataPtr() followed by commit().

I just use getDataPtr in beginning of program in setup function cuz eeprom array never change and i need fast access to my data 
So getDataPtr set _dirty flag for once and i can update my data for first time after that commit always don't work because i didn't use getDataPtr or write function and always edit data directly with my pointer
So that's why i edit commit function and remove _dirty check if

devyte, 2020-06-16:
Your usage isn't correct. The correct usage is: you getDataPtr(), you modify data with it, you commit(). Next time around, you repeat the three steps. You don't getDataPtr() to store the internal EEPROM pointer and reuse it, that is pointless. Also, it's dangerous, because the fact that the internal ptr doesn't change is an implementation detail specific to the current implementation, and there is no guarantee that it will be true in the future. 
Closing due to user error.

closed 2020-06-16 by devyte