esp8266/Arduino #7382: "Wemos D1 Mini Pro" can't search my WiFi SSID

brnyza, 2020-06-15:
#### Platform
- Hardware: Wemos D1 Mini Pro
- Core Version: 2.7.1
- Development Env: Arduino IDE 1.8.12
- Operating System: [Windows]

### Problem Description
It's very instable, my SSID appears only sometimes, as we see in the serial monitor:
If I use "D1 mini" with the same example, it allways find my SSID ("ARITA").
I need to know why "D1 Mini Pro" is not working like "D1 mini".

### Sketch
WiFiScan.ino (basic example of ESP8266Wifi)

### Debug Messages
```
10:50:36.057 -> scan done
10:50:36.057 -> 4 networks found
10:50:36.057 -> 1: Tech_D0059199 (-79)*
10:50:36.057 -> 2: NET_OLIVEIRA 2G (-83)*
10:50:36.057 -> 3: SHAN LON (-72)*
10:50:36.091 -> 4: #NET-CLARO-WIFI (-71) 
10:50:36.091 -> 
10:50:41.079 -> scan start
10:50:43.204 -> scan done
10:50:43.204 -> 3 networks found
10:50:43.204 -> 1: Tech_D0059199 (-78)*
10:50:43.204 -> 2: NET_OLIVEIRA 2G (-85)*
10:50:43.238 -> 3: #NET-CLARO-WIFI (-65) 
10:50:43.238 -> 
10:50:48.253 -> scan start
10:50:50.378 -> scan done
10:50:50.378 -> 2 networks found
10:50:50.378 -> 1: Tech_D0059199 (-79)*
10:50:50.378 -> 2: NET_OLIVEIRA 2G (-86)*
10:50:50.378 -> 
10:50:55.373 -> scan start
10:50:57.505 -> scan done
10:50:57.505 -> 3 networks found
10:50:57.505 -> 1: Tech_D0059199 (-76)*
10:50:57.505 -> 2: NET_OLIVEIRA 2G (-80)*
10:50:57.540 -> 3: ARITA (-65)*   //<<<<< This is my wifi!
10:50:57.540 -> 
10:51:02.537 -> scan start
10:51:04.666 -> scan done
10:51:04.666 -> 1 networks found
10:51:04.666 -> 1: Tech_D0059199 (-81)*
10:51:04.666 -> 
```

pittgo, 2020-07-21:
I am having the exact same issue with  bunch of Wemos D1 Mini V3.0.0 WIFI I recently purchased from Ali Express. When i scan the networks it shows a few other networks around me however never finds my network .. The scan changes from no networks to three .  I took the boards to another network location to see if it was one of my router settings however the same  ... it wont find the SSID of the closest network.

Hardware: Wemos D1 Mini ( clone )
Core Version: 2.7.2
Development Env: Arduino IDE 1.8.9
Operating System: [Windows]

brnyza, 2020-07-22:
I still haven't found a solution. It occurs in more than one unit.

pittgo, 2020-07-22:
I’ve tested three

earlephilhower, 2020-07-22:
For these RF issues a common problem is an older WiFi config area.  In the IDE select "Tools->Erase Flash->Erase All Flash" and re-upload to check if that's the cause.

pittgo, 2020-07-22:
started clean several times   

its odd as it sees a my printer and my nieghbours wifi

set up a windows 10 hotspot  - it sees that however cannot log in to it

the only common thread to SSIS visibility is WPA2  and PSK

I have changes to b, n and c individually
i have changed channels from 1 to 11 individually

Here is the serial output

Wifi scan started
Wifi scan ended
1 networks found
1) DIRECT-C0-HP OfficeJet Pro 8720 ch:2 -86dBm (28% ) MAC:A0:8C:FD:5D:47:5C Encryption:WPA2 / PSK

Wifi scan started
Wifi scan ended
3 networks found
1) Debrajay ch:1 -81dBm (38% ) MAC:BC:3E:07:FF:FE:A8 Encryption:WPA2 / PSK
2) DIRECT-4E-HP OfficeJet Pro 8710 ch:11 -65dBm (70% ) MAC:40:B0:34:7C:F9:4F Encryption:WPA2 / PSK
3) GEORGIESPC ch:11 -27dBm (100% ) MAC:80:56:F2:8A:FF:98 Encryption:WPA2 / PSK

Wifi scan started
Wifi scan ended
2 networks found
1) DIRECT-C0-HP OfficeJet Pro 8720 ch:2 -85dBm (30% ) MAC:A0:8C:FD:5D:47:5C Encryption:WPA2 / PSK
2) Debrajay ch:1 -77dBm (46% ) MAC:BC:3E:07:FF:FE:A8 Encryption:WPA2 / PSK

Wifi scan started
Wifi scan ended
2 networks found
1) DIRECT-4E-HP OfficeJet Pro 8710 ch:11 -63dBm (74% ) MAC:40:B0:34:7C:F9:4F Encryption:WPA2 / PSK
2) GEORGIESPC ch:11 -27dBm (100% ) MAC:80:56:F2:8A:FF:98 Encryption:WPA2 / PSK

Wifi scan started
Wifi scan ended
3 networks found
1) Debrajay ch:1 -80dBm (40% ) MAC:BC:3E:07:FF:FE:A8 Encryption:WPA2 / PSK
2) DIRECT-4E-HP OfficeJet Pro 8710 ch:11 -65dBm (70% ) MAC:40:B0:34:7C:F9:4F Encryption:WPA2 / PSK
3) GEORGIESPC ch:11 -57dBm (86% ) MAC:80:56:F2:8A:FF:98 Encryption:WPA2 / PSK

George

pittgo, 2020-07-22:
However it wont find my home SSID and its a strong signal

G

devyte, 2020-07-22:
For the users who can't see a specific ssid at all, but can see other ssids: I do hope you all know that the ESP is 2.4GHz only, so it won't see an ssid served on the 5GHz band. Your cell phone or laptop are likely dual band, so will see all ssids of both bands, unless you have ancient devices. Many recent ISP-provided modems and routers serve wifi on 5GHz by default due to the higher bandwidth, so the ESP won't see them at all. 

For those who have intermittent results: it's not just a matter of reflashing, you have to "Erase All Flash" like @earlephilhower said above, to force a wifi recalibration. If you have this problem after OTA update, then you'll either have to pull the device and flash it over USB, or try out the experimental #6965 Proof of Concept PR that implements the equivalent wifi config wipe programmatically for the OTA case.

If that doesn't work, you could try:
- lower the Tx power (see readthedocs)
- change the phy mode between n/g/b (see readthedocs). Be aware there are some downsides to that, e.g.: in some routers when one device connects in 802.11g mode, then it forces _all_ devices to downgrade to 802.11g mode.
- change some of your router settings (e.g.: lower Tx power)

If none of the above works, the problem is most likely one of the following:
- bad board batch (common with the on board antennas, especially with the wemos d1 pro, and I see the wemos d1 mini v3 is similar, similar would happen with the ESP07 some time ago)
- bad power supply or peripheral (could be injecting noise to the board, e.g.: some nrf shields are known to suck)
- some incompatibility with your router.  Low level wifi details are handled within Espressif's SDK, which is closed source, so we don't have access to poke around.

Whichever the case from all of the above, there isn't much we can do here in the core. On my own boards (four original wemos d1 mini pro from a while back and a bunch of wemos d1 mini r2) I don't have scan problems problems.

I'm closing for now, but feel free to continue discussion. If you do eventually find a core problem, please open a new issue, follow the instructions, and reference this one for context.

closed 2020-07-22 by devyte

pittgo, 2020-07-22:
Thank you for your reply!!

2.4 is live and well on my network as many other 2.4 periferals are active 

Erased sketch completely before USB reflash several times with several different sketches

Reduced Tx power on the router to medium and low  and tried sever lower channel numbers in concert with that change

Changed the phy to g  - n and b individually

several different power supplies used 

Took units complete to a different site to see if it would scan their SSID   different router and WAN

So that leaves me with BAD clones   

again thanks 

G

villevarn, 2020-11-06:
@pittgo did you give up with bad clones? 
I have 10 different Wemos D1 Mini clones V3.0 - from different suppliers (!) some with the ceramic antenna and some with the "normal antenna". Chip on the board reads ESP8266EX. 
All 10 units has the same issues as described in this thread. 

-> Live in an apartment complex. Own Wi-Fi runs 5GHz + 2.4GHz. There are 40 networks visible on my phone/PC - mixture of 5GHz / 2.4GHz. 
My clones can see 1 of these networks, which is one quite far away (opposite apartment, ground floor). Cannot identify any specifics to this network - GHz, channel, etc. is shared by other routers on the network. 

Other than the WiFi, it seems the units work fine. 
-> We can see some networks on multiple *different* units. Is this really a 'bad clone' issue? 
 -- Not always the WiFi shows and not in every scan. Power issue? Tried with external power and other supplies, no difference. 
-> Wi-Fi does work, kind of, as I can at least see one network. Maybe the fault is a piece of hardware used on the clone? 

Currently I have reflashed to ESP8266_AT_Bin_V1.6.2 and the AT commands work, except when listing networks, none is returned except the one mentioned. 

It seems like there a quite a bunch of these units with WiFi issues out there but no solution. I don't like throwing things away (and definitely will not, but will use them as non-wifi devices). But I also think that there must be 'some way' to fix this, either by replacing a component or by software/re-flash with specific settings.

pittgo, 2020-11-06:
Yes   I kept my 10 in a bag hoping that there would be some solution offered some day lol    I was successful in getting a full credit from Ali Express thankfully and the vendor removed them from their offering 

 

I may have a use for them in a non connected function   just not seeing one yet lol

 

Thanks

 

George

kaushal-10, 2021-11-16:
The problem seems to be occurring mostly on the version 3 boards. 
I face the same issue. 
I don't have a solution for that but there is a way these boards can be useful. 
These Wemos boards are mostly able to connect to each other. So you can create an Access point with one and then connect other boards with it for any projects that use wireless communication without the use of internet.

kaushal-10, 2021-11-26:
Recently i have been learning about micropython, and guess what!! These boards that seem faulty or seem to be useless, can still be used as micropython devices and again, yes they can connect to themselves, so you can make a complete network using just these same kind of (Wemos mini version 3) boards.