esp8266/Arduino #7383: HTTP_UPDATE_FAILED Error (-1) when I add more headers to httpupdate

DrakoPD, 2020-06-16:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I need to add two more headers for my update, I added directly to the httpupdate library:

```
http.addHeader(F("Accept"), "application/vnd.github.v3+json");
http.addHeader(F("authorization"), "Bearer my token");
```

Because I need to download an update from a private GitHub repository. 
I can connect and get the code with http.GET() and print through serial (with HTTP code 200).

But if I try do it in the Httpupdate library, I get `HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused`.

Is there a way to solved this? or a way to pass the code that I download by .GET() for update? if exist something like `ESP.update(http.GET())`.

I had configured certs and NTP time.

earlephilhower, 2020-06-16:
@DrakoPD, please post a full MCVE and full debug logs, as requested in the template.  W/o them, we can't really do anything for you.

DrakoPD, 2020-06-17:
@earlephilhower Done. I updated my question with more code.

earlephilhower, 2020-06-17:
Couple of things right off the bat:
* I assume the `/*  ...*/` block is not really commented out in your code
* `bool mfln = client.probeMaxFragmentLength("https://api.github.com", 443, 1024);` is wrong and will always fail.  Just pass in the hostname, not a URL.  api.github.com
* `    BearSSL::WiFiClientSecure client;` <- Use a global for this because it is a very large structure to put on the stack.

Can you max out the debug logging (2nd from last in the menu, dumps a ton more stuff including BearSSL and TCP stuff) and re-capture?

earlephilhower, 2020-06-17:
Another quick thing to check is to ass a `.setInsecure` (for testing only) before passing in the client.  That would allow you to narrow down whether it's the SSL negotiation failing or something else, since the insecure mode will not check anything and always connect if it can.

DrakoPD, 2020-06-17:
> I assume the /* ...*/ block is not really commented out in your code
Is commented, I only used for test the connection to GitHub, and its work.

> Just pass in the hostname, not a URL. api.github.com
I don't understand, that mean just "Github"?

> Can you max out the debug logging...
I already have in this mode, also, I clean all BearSSL and TCP, just let the wifi client an set insecure. The problem was that I accidentally deleted the line `http.begin()` inside the library.

Now I get the first error when I add more headers (before of the mistake of http.begin):
`HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed`

DrakoPD, 2020-06-17:
Also, the debug console is too long now, do I really need to put all here?

earlephilhower, 2020-06-17:
If it gets to `HTTP_UPDATE_FAILED Error (-106): Verify Bin Header Failed` then it is most likely an issue in your code or .BIN file since the connection was made, SSL was negotiated, and something was transferred back.  At this point, your best bet it to go to esp8266.com or https://gitter.im/esp8266/Arduino to get help with your setup.

Given the info here, I'm closing as it's not looking like the core at this point.

closed 2020-06-17 by earlephilhower

DrakoPD, 2020-06-17:
I really don't like esp8266.com, it have an annoying publicity and we have to wait for the approbation of any post (from 6 to 24 hours), making difficult have a stable communication with others.