esp8266/Arduino #7387: Destructor of String may cause crash if allocation failed

TD-er, 2020-06-17:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Since I now have finally the stack decoding working, I've come across a number of crashes in my code related to running out of memory.

One of them seems to be related to the destructor of String.
I don't have a MCVE yet, but I've come to the conclusion it must have something to do with creating a String object where the heap allocation failed during construct.

Part of the error before the stack dump:
```
Exception (3):
epc1=0x40101059 epc2=0x00000000 epc3=0x4026aceb excvaddr=0x40070268 depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40101059 in umm_free_core at umm_malloc.cpp:?
  epc3=0x4026aceb in uart_write_char at ??:?

>>>stack>>>

ctx: cont
sp: 3fff27d0 end: 3fff2bc0 offset: 0190
```

A part of the stack:

```
0x40101317 in free at ??:?
0x40101317 in free at ??:?
0x40266790 in String::~String() at ??:?
0x4028184d in operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc:48
0x4022cabc in ControllerDelayHandlerStruct<MQTT_queue_element>::addToQueue(MQTT_queue_element const&) at ??:?
0x40241af2 in MQTT_queueFull(unsigned char) at ??:?
```

A bit more context.
The String object is part of a struct (containing multiple String objects) that is created and stored in a `std::list` via `emplace_back` and removed from it via `pop_front`.

During the destruct of `String` it calls `invalidate`:
```c++
void String::invalidate(void) {
    if(!isSSO() && wbuffer())
        free(wbuffer());
    init();
}
```

If the construction of the string failed, I would expect the returned pointer of the alloc call to be a `null_ptr`.
Otherwise I cannot explain what may have happened here.

TD-er, 2020-06-17:
I've been looking through the code of WString.cpp a bit more and found this in the `changeBuffer` code:

```c++
char *newbuffer = (char *) realloc(isSSO() ? nullptr : wbuffer(), newSize);
    if (newbuffer) {
```

When looking for a description of what realloc does, [see cplusplus.com realloc](http://www.cplusplus.com/reference/cstdlib/realloc/) and specifically the description for C++11 when you set the size to 0:

> If size is zero, the return value depends on the particular library implementation: it may either be a null pointer or some other location that shall not be dereferenced.

So it looks like it is possible to create a String object which has `setSSO(false)` and the buffer is not a `null_ptr` but still should not be dereferenced.
That is what `invalidate` tries to do.

So it looks like the check for the pointer should be extended with a check for the `newSize`.
Or am I missing something here?

I guess you may also need to set sso to true as a call to realloc with size 0 will de-allocate the previous pointer if there was any address set to it.

TD-er, 2020-06-17:
OK, that was a false conclusion as the `newSize` never can be 0.

Will hunt further....

earlephilhower, 2020-06-19:
@TD-er, while I'm not going to guarantee that `String` is safe under all memory conditions, I would actually be more suspicious of the `std::list` which does allocations itself.  Unless you've enabled C++ exceptions, the std::list could be just point to garbage somewhere that is being tried to access as a `String`.

TD-er, 2020-06-19:
So you're suggesting not to use `emplace_back`, but rather construct some move operator and use that?
For example `push_back` a simple version of the object and then fill that using move operators? (`String` has a move operator for example)

earlephilhower, 2020-06-19:
Can't really comment one way or the other, but without enabilng C++ exceptions when you hit OOM the std:: library is going to go very bad, very fast.  Easy thing would be to enable C++ exceptions and run the failing pattern and see if you get an OOM error thrown by the list implementation...

TD-er, 2020-06-19:
Hmm enabling C++ exceptions is probably not going to help to fit the sketch in the 1M sketch limit....

earlephilhower, 2020-06-20:
If you use the Tools->Exceptions->Disabled(new can abort) then the add'l flash should be relatively small and it will still catch when a `new` bombs out or some other exception occurs.

TD-er, 2020-06-20:
I've been working on this issue today (without enabling exceptions for now) and I think you're on the right track with the remark about the behavior of `std::list`
I did change the behavior of how the structures are added to the list. (and map)

For example, first add a default instance of the struct to the list using `emplace_back` and then fill the `String` objects in that struct.
Did this in a few spots in the software and so far it is running a lot more stable.
The only place where I got this crash again is where I have not changed behavior like this.

And the units are stress tested a lot right now, for several hours.
They are forwarding large messages (up-to 1 kByte), splitting them, re-arranging them (parts are received out of order from several nodes) and sent (via splitting) to the next node.
At the same time running all tasks and the end node is merging all packets again and forwarding it to MQTT.
Traffic to MQTT is roughly 50 kbps and the packets are sent via ESP-now over several hops.
On top of that it is also spamming a lot of logs to the serial port + a http stream of JSON formatted logs.

So this is a lot of std::list and std::map structures filled with structs containing one or more String objects.

And it is working stable now, while still being very responsive to http calls.  (and operating several sensors + OLED display)

> If you use the Tools->Exceptions->Disabled(new can abort) then the add'l flash should be relatively small and it will still catch when a new bombs out or some other exception occurs.

Still I think it would be a good idea to catch these `new` exceptions.
Any idea how it can be set in PlatformIO?

earlephilhower, 2020-06-20:
Add this to the g++ lines however PIO wants it:  `-fno-exceptions -DNEW_OOM_ABORT`

The core will catch any `throw` exceptions and abort with an Unhandled Exception error and do a stack dump, so you'll be able to backtrace where the call that failed was (normally).

earlephilhower, 2020-07-29:
I'm not seeing anything to be done here, so I'm closing.  If there's a specific case you find where it's not OOM, please open a new issue w/MCVE so we can dig into it.

closed 2020-07-29 by earlephilhower