esp8266/Arduino #7389: Tasmota/ESP8266 Drops WiFi Connection - Cannot Reconnect

gwrichard, 2020-06-18:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I am having tissues with my Tasmota devices dropping the Wi-Fi connection. They are flashed to 8.3.1.

I have disabled band steering on my router. Changing to separate 2.4 and 5Ghz frequencies seemed to help (instead of being blocked after Tasmota reboot, it is stable for a couple of days now).

Eventually the Tasmota devices drop their Wi-Fi connection (usually after a day or two) and fail to reconnect. The only remedy is to reboot the router.

Following are my router details:

    Firmware Number: 18.1.c.0462-950-RB
    Modem Make Model: Technicolor DJA0231

Tasmota devices are the ONLY devices on my network that experience this problem. All other devices do not have trouble connecting or reconnecting.

I was advised by the Tasmota Discord channel to post the issue here, since they said the issue is not Tasmota, but rather the Arduino module.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

devyte, 2020-06-18:
@gwrichard we have certain rules here for reporting issues. One of them, as shown in the issue template above, is that a MCVE is required, where only core code is used.
If the Tasmota guys are correct and your issue is a core one, then indeed you need to report here. However, you need the MCVE, so either you need to reduce your code to a pure core example, or the Tasmota guys need to help you do it. Without that, we can't help.

earlephilhower, 2020-07-29:
No MCVE, no response for ~2 weeks.  Closing as nothing to do here.

closed 2020-07-29 by earlephilhower