esp8266/Arduino #7392: undefined reference to `sntp_get_timezone'

pulsartronic, 2020-06-20:
Hello, upgrading from 2.6.3 to 2.7.1 the following code does not compile anymore ...

```cpp
#include <sntp.h>

void setup() {
	sntp_stop();
	sntp_setservername(0, "example.pool.com");
	sntp_set_timezone(0);
	sntp_init();
	const char* host = sntp_getservername(0);
	uint8_t tz = sntp_get_timezone();
	uint32_t ts = sntp_get_current_timestamp();
}

void loop() {

}
```

It throws this error:

```
/home/pulsartronic/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/example.ino.cpp.o:(.text.setup+0x4): undefined reference to `sntp_get_timezone'
/home/pulsartronic/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/example.ino.cpp.o: in function `setup':
/home/pulsartronic/development/pulsartronic/Arduino/LoRaWANGatewaySC/example/example.ino:3: undefined reference to `sntp_get_timezone'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
```

Context:
SO: Ubuntu 18.04
IDE: Arduino IDE 1.8.12
Lib: esp8266 2.7.1

Am I doing something wrong ? ... any guide would be really apreciatted

luc-github, 2020-06-20:
seems it is not present in lwip v2, you need to use v1.4

pulsartronic, 2020-06-20:
> seems it is not present in lwip v2, you need to use v1.4

Where do i specify it ? ... is it a #define or something ?

thanks

luc-github, 2020-06-20:
in arduino ide check the esp8266 compilation settings

d-a-v, 2020-06-20:
It has been removed [in an optimization process](https://github.com/esp8266/Arduino/issues/6993) because this function was an import from espressif timezone api (not part of lwIP), which was broken, and not used in the arduino API. 
You can switch to the esp8266 arduino official `configTime()` API that recently evolved (#7234, #6993, #6373). The official NTP api is presented in the [NTP-TZ-DST.ino](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino) example.

Now, you basically *only* need to use `configTime(`[`TZ_Europe_London`](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h)`, "example.pool.com");` to get always right and proper time.

If you really *need to know* time relative to UTC, you can do that with
```
time_t local, utc;
localtime(&local);
gmtime(&utc);
```
You'll get local time and UTC time in respective above `time_t` structures.

d-a-v, 2020-08-15:
@pulsartronic lwIP 1.4 is removed in master  and won't be available in next releases.
Is it OK with `localtime` and `gmtime` ?

pulsartronic, 2020-08-15:
@d-a-v sorry ... yes it is ok to me, thank you very much

closed 2020-08-16 by devyte

pulsartronic, 2020-09-01:
@d-a-v hello
time_t local, utc;
localtime(&local);
gmtime(&utc);

local and utc end up having the exact same value, no matter what timezone is configured with
configTime(TZ_Europe_Madrid, "pool.ntp.org");

have you tried ? ... is it intended ?

d-a-v, 2020-09-01:
Yes I tried
No it is not intended, I had got different values,
like when the ntp-tz-dst example is executed.
Is the time you get local or UTC or something else ?

pulsartronic, 2020-09-01:
Hi, this is the output of the example

localtime: isdst=1 yday=244 wday=2 year=120 mon=8 mday=1 hour=19 min=5 sec=30
gmtime:    isdst=0 yday=244 wday=2 year=120 mon=8 mday=1 hour=17 min=5 sec=30
clock:     660s / 83065000ns
millis:    660083
micros:    660083087
gtod:      1598979930s / 286306us
time:      1598979930
timezone:  CET-1CEST,M3.5.0,M10.5.0/3
ctime:     Tue Sep  1 19:05:30 2020
sntp0:     pool.ntp.org (147.156.7.26) IPv6: No Reachability: 1

localtime and gmtime have the right values when reading from tm structure, like this
```cpp
time_t lt = time(nullptr);
tm* ltm = localtime(&lt);

time_t gt = time(nullptr);
tm* gtm = gtmtime(&gt);
```
yet "lt" and "gt" have the same value, it is the UTC value ... am i doing something wrong ?
In fact if i omit the "= time(nullptr);" values have no sense 
I am trying to get local time in seconds since 1970, maybe it is not the correct way 

should gtod and time be the same ?

d-a-v, 2020-09-01:
> yet "lt" and "gt" have the same value, it is the UTC value ... 

https://linux.die.net/man/2/time

https://linux.die.net/man/3/localtime

 I should revisit the example to use only one call to `time()` and add comments.

> I am trying to get local time in seconds since 1970, maybe it is not the correct way

There's no proper/direct way for that to my knowledge.
One possible way is to use these functions and their inverse 
https://linux.die.net/man/3/timegm while tweaking TZ to local or UTC

jeanjmichel, 2024-08-15:
Guys, I'm trying lear about IoT using a ESP8266 and Azure, but it's really tough, the last version of the libraries do not work and show tons of errors.

After some downgrades I have installed:

Board:
esp8266 2.7.4

Libraries:
ArduinoJson 5.13.5
AzureIoTHubMQTTClient 0.2.3
Time 1.6.1

And when I try compile my code I receive this error: "\libraries\AzureIoTHubMQTTClient\NtpClientLib_ESP.cpp.o:(.text._ZN9NTPClient7getTimeEv+0x2c): undefined reference to `sntp_get_timezone'".

What I must do now to put my code to run, some tip?

Thanks.