esp8266/Arduino #7395: SPISlave::setData just drops anything over 32 bytes.

sneakyimp, 2020-06-20:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					other
- Core Version:			b4c28e74d66c5386f0b540379050d62bfbd0d4ca
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			other
- Flash Mode:			other
- Flash Size: 		4MB

### Problem Description
This is a fundamental source code problem with [SPISlave](https://github.com/esp8266/Arduino/blob/master/libraries/SPISlave/src/SPISlave.cpp), which is used in your basic [SPISlave_Test](https://github.com/esp8266/Arduino/blob/master/libraries/SPISlave/examples/SPISlave_Test/SPISlave_Test.ino) sketch, and applies to any module or IDE that would make use of these libraries. Your SPISlave class has a hard-coded limitation of 32 bytes for the **SPISlave.setData()** function:

```c++
void SPISlaveClass::setData(uint8_t * data, size_t len)
{
    if(len > 32) {
        len = 32;
    }
    hspi_slave_setData(data, len);
}
```

If you pass this function a data buffer with more than 32 bytes, wouldn't it be courteous to *throw an exception* so that a developer might understand where they went wrong? Imagine how frustrating it would be to try sending more data only to have it clipped, without warning or explanation, to 32 bytes?

earlephilhower, 2020-06-21:
> ...If you pass this function a data buffer with more than 32 bytes, wouldn't it be courteous to throw an exception so that a developer might understand where they went wrong...

Arduino, by default, doesn't allow exceptions, so throwing an exception is unfortunately not an option.

The API presently doesn't have a return code.  If `SPISlave` isn't implementing an Arduino class with a defined API, we'd be happy to get a PR adding a `bool` return code to this and any other functions which can fail.

devyte, 2020-06-21:
The method is called setData, not write or send or similar. The reason is that it sets the hw data buffer for sending, and that hw buffer is 32bytes max. There is no point of failure in setting it. I believe that when the buffer is empty (data sending is done) the callback onDataSent is called, where you would setData with the next 32 bytes. 
So, yes: it's not obvious that you're supposed to provide 32 bytes at a time, but the odd method naming should provide a hint. I'm not spi-savvy, but it took me 15s to figure that out.
As for exceptions, our core supports C++ exceptions, but you have to enable building with them manually, because like @earlephilhower said, Arduino doesn't use them by default, and generic Arduino code isn't designed with them in mind. However, enabling them will blow up your bin size, there's still work to be done to improve that. 
So no, throwing isn't an option for now. I would think an assert would do, given that providing more than 32bytes is really a programming error that must be fixed.

> wouldn't it be courteous

This is a community-driven project, not company driven, so the code you see is contributed by users who volunteer their efforts and time freely. If you think an improvement is called for, I suggest a courteous PR.

earlephilhower, 2020-06-21:
Closing per the above description.

closed 2020-06-21 by earlephilhower