esp8266/Arduino #7396: env: python3: No such file or directory

brycem223, 2020-06-20:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			unknown
- Development Env:		Arduino IDE 1.8.13
- Operating System:		MacOS Mojave 10.14.6

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Mode:			qio
- Flash Size: 		        4MB (FS:3MB OTA:~512KB)
- lwip Variant:                 v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:            115200

### Problem Description

I have not been able to upload any sketches to my ESP8266 board.
When I try uploading or verifying the sketch in Arduino, I get this error.

I have updated the default Python version to 3.8.3 and set it as the global default using pyenv, but that did not seem to change anything.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);    
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);                
  digitalWrite(LED_BUILTIN, HIGH);  
  delay(2000);          

```

### Debug Messages

```
env: python3: No such file or directory
exit status 127
/Users/bryce/Desktop/Arduino.app/Contents/Java/arduino-builder returned 127
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

More error info from Arduino:

Arduino: 1.8.13 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), Port: "/dev/cu.SLAB_USBtoUART", 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:3MB OTA:~512KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

earlephilhower, 2020-06-21:
Please update to git head, rerun `get.py` per the instructions, and retry.  A standalone version of Python is now included to avoid the numerous path and version issues seen on the Mac.

brycem223, 2020-06-21:
Sorry, I'm a student and am still fairly new to all this. Would you care to elaborate on what you mean by "update to git head"? I gave it a shot but the error remains.

devyte, 2020-06-21:
@brycem223 in that case, why did you add this check:
> I have tested that the issue is present in current master branch (aka latest git).

? 
Install instructions for latest git are in readthedocs and mentioned in the front README, along with a link to readthedocs. 

Closing due to already fixed.

closed 2020-06-21 by devyte