esp8266/Arduino #7398: Large stack usage in core and MDNS libraries

earlephilhower, 2020-06-22:
I've been playing locally with GCC10 and the `-Wstack-usage=` option which emits a compile-time warning when stacks are larger than the size given.  There's only 4K total stack to play with so I set the warning limit to 300 bytes.

This isn't an exhaustive list (need to do a local CI and aggregate warnings), but I'm seeing very high use in the flash_hal and MDNS:

Flash_write() has a 512 byte buffer allocated on the stack in the case of an unaligned write, which seems pretty massive and ripe for reduction:
````
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\flash_hal.cpp: In function 'int32_t flash_hal_write(uint32_t, uint32_t, const uint8_t*)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\flash_hal.cpp:102:9: warning: stack usage is 576 bytes [-Wstack-usage=]
  102 | int32_t flash_hal_write(uint32_t addr, uint32_t size, const uint8_t *src) {
      |         ^~~~~~~~~~~~~~~
````

MDNS (old and new) have stack usages up to almost 700 bytes which might explain some issues seen at runtime.
````
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp: In member function 'uint8_t esp8266::MDNSImplementation::MDNSResponder::_ZNK7esp826618MDNSImplementation13MDNSResponder17_replyMaskForHostERKNS1_16stcMDNS_RRHeaderEPb$part$0(const esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRHeader&, bool*) const':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp:1991:9: warning: stack usage is 304 bytes [-Wstack-usage=]
 1991 | uint8_t MDNSResponder::_replyMaskForHost(const MDNSResponder::stcMDNS_RRHeader& p_RRHeader,
      |         ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp: In member function 'uint8_t esp8266::MDNSImplementation::MDNSResponder::_ZNK7esp826618MDNSImplementation13MDNSResponder20_replyMaskForServiceERKNS1_16stcMDNS_RRHeaderERKNS1_14stcMDNSServiceEPb$part$0(const esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRHeader&, const esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, bool*) const':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp:2069:9: warning: stack usage is 576 bytes [-Wstack-usage=]
 2069 | uint8_t MDNSResponder::_replyMaskForService(const MDNSResponder::stcMDNS_RRHeader& p_RRHeader,
      |         ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_parseQuery(const esp8266::MDNSImplementation::MDNSResponder::stcMDNS_MsgHeader&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp:179:6: warning: stack usage is 608 bytes [-Wstack-usage=]
  179 | bool MDNSResponder::_parseQuery(const MDNSResponder::stcMDNS_MsgHeader& p_MsgHeader)
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_ZN7esp826618MDNSImplementation13MDNSResponder15_processAnswersEPKNS1_16stcMDNS_RRAnswerE$part$0(const esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRAnswer*)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp:749:6: warning: stack usage is 320 bytes [-Wstack-usage=]
  749 | bool MDNSResponder::_processAnswers(const MDNSResponder::stcMDNS_RRAnswer* p_pAnswers)
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_parseResponse(const esp8266::MDNSImplementation::MDNSResponder::stcMDNS_MsgHeader&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp:628:6: warning: stack usage is 320 bytes [-Wstack-usage=]
  628 | bool MDNSResponder::_parseResponse(const MDNSResponder::stcMDNS_MsgHeader& p_MsgHeader)
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\ESP8266mDNS_Legacy.cpp: In member function 'void Legacy_MDNSResponder::MDNSResponder::_parsePacket()':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\ESP8266mDNS_Legacy.cpp:567:6: warning: stack usage is 688 bytes [-Wstack-usage=]
  567 | void MDNSResponder::_parsePacket()
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSHostDomain(const char*, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp:1367:6: warning: stack usage is 320 bytes [-Wstack-usage=]
 1367 | bool MDNSResponder::_writeMDNSHostDomain(const char* p_pcHostname,
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(const esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp:1410:6: warning: stack usage is 320 bytes [-Wstack-usage=]
 1410 | bool MDNSResponder::_writeMDNSServiceDomain(const MDNSResponder::stcMDNSService& p_Service,
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_PTR_IP4(IPAddress, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp:1517:6: warning: stack usage is 560 bytes [-Wstack-usage=]
 1517 | bool MDNSResponder::_writeMDNSAnswer_PTR_IP4(IPAddress p_IPAddress,
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_PTR_TYPE(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp:1550:6: warning: stack usage is 544 bytes [-Wstack-usage=]
 1550 | bool MDNSResponder::_writeMDNSAnswer_PTR_TYPE(MDNSResponder::stcMDNSService& p_rService,
      |      ^~~~~~~~~~~~~
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_SRV(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp:1725:6: warning: stack usage is 320 bytes [-Wstack-usage=]
 1725 | bool MDNSResponder::_writeMDNSAnswer_SRV(MDNSResponder::stcMDNSService& p_rService,
      |      ^~~~~~~~~~~~~
````

Crypto.c has some large stacks (which might be unavoidable given the algorithm, but we may want to consider moving it to heap or refactoring the code:
````
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Crypto.cpp: In function 'void* {anonymous}::createBearsslHmac(const br_hash_class*, const void*, size_t, const void*, size_t, void*, size_t)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Crypto.cpp:67:7: warning: stack usage is 432 bytes [-Wstack-usage=]
   67 | void *createBearsslHmac(const br_hash_class *hashType, const void *data, const size_t dataLength, const void *hashKey, const size_t hashKeyLength, void *resultArray, const size_t outputLength)
      |       ^~~~~~~~~~~~~~~~~
[... cut ...]
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Crypto.cpp: In function 'void experimental::crypto::chacha20Poly1305Kernel(int, void*, size_t, const void*, const void*, size_t, const void*, void*, const void*, size_t)':
C:\Users\earle\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Crypto.cpp:507:6: warning: stack usage is 464 bytes [-Wstack-usage=]
  507 | void chacha20Poly1305Kernel(const int encrypt, void *data, const size_t dataLength, const void *key, const void *keySalt, const size_t keySaltLength,
      |      ^~~~~~~~~~~~~~~~~~~~~~
````

TD-er, 2020-06-22:
That's a really nice feature.
Just tested it here and I do indeed see some warnings in my own code too (gosh, surprise...)

I just wonder, what does the parameter exactly state, as I cannot find the documentation on it right now.
It looks like it describes "stack usage".  (and not stack free)

Most of the warnings I get are these:
```
warning: stack usage might be unbounded [-Wstack-usage=]
```

Is it right to interpret these as the stack increase when entering the function?
After all, the compiler cannot predict the total stack usage at run time, or at least not for all code (e.g. recursive calls or array sizes determined at runtime)

earlephilhower, 2020-06-22:
Those could be where you have an explicit `alloca()` in the code, but I think it's more commonly something like:
````
void fn(char bufsize) {
   char buf[bufSize];
   ....
}
````

There are other warning options (stack-frame size) which I *think* ignore this kind of setup and only look for static size assignments, but I did not dig into it.

TD-er, 2020-06-22:
Well I've been playing a bit more with this flag and I think it is a very good tool for the problems I'm seeing.
So thanks for mentioning it :)

This has been a good month.
First having a working stack decoder (in the serial monitor) and now the compiler telling me where I may run into stack overflows.
I can only dream of what can be next...

d-a-v, 2020-06-22:
What stack decoder are you using ?

TD-er, 2020-06-22:
@d-a-v 
In the PlatformIO environment I use this:

```
monitor_filters           = esp8266_exception_decoder
```

For this you need to have `platform = espressif8266@2.5.2`

earlephilhower, 2020-10-17:
Latest master report:

````
In file included from /home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/ESP8266WebServer.h:306,
                 from /home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino:42:
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/Parsing-impl.h: In member function 'bool esp8266webserver::ESP8266WebServerTemplate<ServerType>::_parseForm(esp8266webserver::ESP8266WebServerTemplate<ServerType>::ClientType&, const String&, uint32_t) [with ServerType = WiFiServer]':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src/Parsing-impl.h:361:6: warning: stack usage might be unbounded [-Wstack-usage=]
  361 | bool ESP8266WebServerTemplate<ServerType>::_parseForm(ClientType& client, const String& boundary, uint32_t len){
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAP(const char*, const char*, int, int, int)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp:94:6: warning: stack usage is 304 bytes [-Wstack-usage=]
   94 | bool ESP8266WiFiAPClass::softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection) {
      |      ^~~~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::connectWiFiMulti(uint32_t)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp:278:13: warning: stack usage might be unbounded [-Wstack-usage=]
  278 | wl_status_t ESP8266WiFiMulti::connectWiFiMulti(uint32_t connectTimeoutMs)
      |             ^~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/BearSSLHelpers.cpp: In function 'bool BearSSL::SigningVerifier_verify(BearSSL::PublicKey*, UpdaterHashClass*, const void*, uint32_t)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/BearSSLHelpers.cpp:877:17: warning: stack usage might be unbounded [-Wstack-usage=]
  877 | extern "C" bool SigningVerifier_verify(PublicKey *_pubKey, UpdaterHashClass *hash, const void *signature, uint32_t signatureLen) {
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp: In function 'void BearSSL::br_ssl_client_base_init(br_ssl_client_context*, const uint16_t*, int)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:918:15: warning: stack usage might be unbounded [-Wstack-usage=]
  918 |   static void br_ssl_client_base_init(br_ssl_client_context *cc, const uint16_t *cipher_list, int cipher_cnt) {
      |               ^~~~~~~~~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp: In function 'void BearSSL::br_ssl_server_base_init(br_ssl_server_context*, const uint16_t*, int)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:944:15: warning: stack usage might be unbounded [-Wstack-usage=]
  944 |   static void br_ssl_server_base_init(br_ssl_server_context *cc, const uint16_t *cipher_list, int cipher_cnt) {
      |               ^~~~~~~~~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp: In member function 'size_t BearSSL::WiFiClientSecure::write(Stream&)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp:309:8: warning: stack usage is 320 bytes [-Wstack-usage=]
  309 | size_t WiFiClientSecure::write(Stream& stream) {
      |        ^~~~~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_readRRAnswer(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRAnswer*&)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:459:6: warning: stack usage is 320 bytes [-Wstack-usage=]
  459 | bool MDNSResponder::_readRRAnswer(MDNSResponder::stcMDNS_RRAnswer*& p_rpRRAnswer)
      |      ^~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSHostDomain(const char*, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1376:6: warning: stack usage is 320 bytes [-Wstack-usage=]
 1376 | bool MDNSResponder::_writeMDNSHostDomain(const char* p_pcHostname,
      |      ^~~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp: In member function 'bool esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(const esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&)':
/home/earle/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp:1419:6: warning: stack usage is 320 bytes [-Wstack-usage=]
[... cut ...]
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Updater.cpp:197:6: warning: stack usage is 320 bytes [-Wstack-usage=]
  197 | bool UpdaterClass::end(bool evenIfRemaining){
      |      ^~~~~~~~~~~~
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/time.cpp: In function 'void setTZ(const char*)':
/home/earle/Arduino/hardware/esp8266com/esp8266/cores/esp8266/time.cpp:187:6: warning: stack usage might be unbounded [-Wstack-usage=]
  187 | void setTZ(const char* tz){
      |      ^~~~~
````