esp8266/Arduino #7402: LittleFS linked in while not included

TD-er, 2020-06-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I do not include LittleFS.h (even commented out the one that's between #ifdef wrappers to make sure) and still the LilltleFS library seems to be included in the build.
It is taking quite some space, so it would be nice to have it not included unless I want it to be.

[image]
These are the libraries present in the inspect build.
As you can see, the LittleFS takes up over 20 kB.

By the way, SDFS is also present and I'm pretty sure (not yet 100% sure) I don't include that one too.
ESP8266SdFat is probably included by SDFS, so I imagine that one will be gone too when SDFS has been resolved.

TD-er, 2020-06-24:
Update:
The SDFS stuff is removed by mentioning it in the `lib_ignore`, but that doesn't seem to work with LittleFS
```
lib_ignore                = ${regular_platform.lib_ignore}, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS, LittleFS  
```

[image]

TD-er, 2020-06-24:
Found it, I have to use `LittleFS(esp8266)` in the `lib_ignore`.

earlephilhower, 2020-06-24:
Didn't we have this before?  IIRC the issue was it was always being included because of OTA calling "StopLittleFS(); StopSPIFFS();" and there was a bit of `weak` magic added to the core to short-circuit the littlefs or spiffs if there was no call in the code to XXX.begin()?

@d-a-v, does this ring a bell?

earlephilhower, 2020-06-24:
> Found it, I have to use LittleFS(esp8266) in the lib_ignore.

So is there not a problem now, @TD-er ?

d-a-v, 2020-06-24:
@earlephilhower I checked and littlefs is not linked when building an empty sketch with arduino builder.
It may have to do with PIO buildscripts.

TD-er, 2020-06-24:
> > Found it, I have to use LittleFS(esp8266) in the lib_ignore.
> 
> So is there not a problem now, @TD-er ?

Not entirely sure.
The lib can be excluded via the lib_ignore, but what if there is a more fundamental issue here (or in PlatformIO) that may suddenly include other libs you don't even know about.

earlephilhower, 2020-06-24:
That's got to be a PIO issue, AFAICT.

If there was something specific to the core (i.e. it made a call to littlefs) then it couldn't even *link* if you ignored the lib.  Might want to check `-fgc-sections` (sp?) in the link stage to have it throw out unused code.  Not sure how to do that w/PIO.

TD-er, 2020-06-25:
I've looked into that a while ago and could not find anything that made the build much smaller using those compiler flags.
I wonder, could it be that building in debug mode (as in not release) is not performing such a last check at link time to see if code is unused?
If so then the library showing in the PIO inspect window might be a false positive.

At least using the `lib_ignore` flag does speed up compile time, so it is good to use it anyway.

Misiu, 2020-06-25:
@TD-er I think this bug is related: https://github.com/platformio/platformio-core/issues/3561

earlephilhower, 2020-06-27:
Given the PIO issue from @Misiu and the fact that if you explicitly tell PIO to drop the lib everything works, I can't see anything on the core side to be changed.

Closing, but if something comes back from PIO on this do please let us know.

closed 2020-06-27 by earlephilhower