esp8266/Arduino #7406: SPI Slave, stops recieving data after recieving a zero even if more data is transmitted

ben-watkinson, 2020-06-25:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
Hi,

I'm using the SPI_slave example to send data from an arduino uno to an esp8266 board. the majority of the time this works fine however if one of the bytes in the 32 byte packet is a zero the rest of the bytes after it in the packet also appear as zero. Here is the data i am sending from the uno in hex:

[1,11,12,22,33,34,44,55,56,66,77,78,88,99,9A,AA,BB,BC,CC,DD,DE,EE,FF,F0,0,11,12,22,33,34,44,20]
 
I added this to the SPISlave.ondata function to output the data to the serial monitor 

    Serial.print("data = ");
    for(int i = 0; i<len; i++)
    {
      unsigned int check = (uint8_t) data[i];    
      Serial.print(check,HEX); 
      Serial.print(",");
    }
    Serial.print("\n");

the serial monitor output on the esp8266 side is this:

data = 1,11,12,22,33,34,44,55,56,66,77,78,88,99,9A,AA,BB,BC,CC,DD,DE,EE,FF,F0,0,0,0,0,0,0,0,0,

If i change the zero in the transmitted packet to any other number the whole packet is recieved fine.

Why does the SPI stop recieving once it recieves a zero?
Is there anyway to keep reading in bytes after a zero?

Thanks

Ben

JiriBilek, 2020-06-27:
If you are using the SpiSlave_Master example (Arduino side), check you are transferring the data as uint8_t * and not as char *. See https://github.com/esp8266/Arduino/blob/52f4cc8de06671a44a27886f4df84777731cb4a6/libraries/SPISlave/examples/SPISlave_Master/SPISlave_Master.ino#L86

earlephilhower, 2020-06-27:
Thanks, @JiriBilek, that's my guess as well since the ESP hardware and SPI code don't care what the received data is when sending to the app, they just empty a hardware FIFO from what I can see.

@ben-watkinson, please go over your Uno code and if it's really sending in `uint8_t *` then post your Uno code here and we can eyeball it.  Otherwise, there's nothing we can do.  If that turns out to be the problem, please drop a note and close here.

ben-watkinson, 2020-06-29:
Hi,

I have tried changing the send function from sending a char * to a uint8_t * but unfortunately that didnt change anything. here is the full arduino uno code:

```
#include <SPI.h>

uint8_t test_packet[32] = {1,0x11,0x12,0x22,0x33,0x34,0x44,0x55,0x56,0x66,0x77,0x78,0x88,0x99,0x9A,0xAA,0xBB,0xBC,0xCC,0xDD,0xDE,0xEE,0xFF,0xF0,0x00,0x11,0x12,0x22,0x33,0x34,0x44,32};

class ESPMaster 
{
  private:
    uint8_t _ss_pin;

  public:
    ESPMaster(uint8_t pin): _ss_pin(pin) {}
    void begin() 
    {
      pinMode(_ss_pin, OUTPUT);
      digitalWrite(_ss_pin, HIGH);
    }

    uint32_t readStatus() 
    {
      digitalWrite(_ss_pin, LOW);
      SPI.transfer(0x04);
      uint32_t status = (SPI.transfer(0) | ((uint32_t)(SPI.transfer(0)) << 8) | ((uint32_t)(SPI.transfer(0)) << 16) | ((uint32_t)(SPI.transfer(0)) << 24));
      digitalWrite(_ss_pin, HIGH);
      return status;
    }

    void writeStatus(uint32_t status) 
    {
      digitalWrite(_ss_pin, LOW);
      SPI.transfer(0x01);
      SPI.transfer(status & 0xFF);
      SPI.transfer((status >> 8) & 0xFF);
      SPI.transfer((status >> 16) & 0xFF);
      SPI.transfer((status >> 24) & 0xFF);
      digitalWrite(_ss_pin, HIGH);
    }

    void readData(uint8_t * data) 
    {
      digitalWrite(_ss_pin, LOW);
      SPI.transfer(0x03);
      SPI.transfer(0x00);
      for (uint8_t i = 0; i < 32; i++) 
      {
        data[i] = SPI.transfer(0);
      }
      digitalWrite(_ss_pin, HIGH);
    }

    void writeData(uint8_t * data, size_t len) 
    {
      uint8_t i = 0;
      digitalWrite(_ss_pin, LOW);
      SPI.transfer(0x02);
      SPI.transfer(0x00);
      while (len-- && i < 32) 
      {
        SPI.transfer(data[i++]);
      }
      while (i++ < 32) 
      {
        SPI.transfer(0);
      }
      digitalWrite(_ss_pin, HIGH);
    }

    String readData() 
    {
      char data[33];
      data[32] = 0;
      readData((uint8_t *)data);
      return String(data);
    }

    void writeData(const uint8_t * data) 
    {
      writeData((uint8_t *)data, strlen(data));
    }
};

ESPMaster esp(SS);

void send(const uint8_t * message) 
{
  //Serial.print("Master: ");
  //Serial.println(message);
  esp.writeData(message);
  delay(10);
  //Serial.print("Slave: ");
  //Serial.println(esp.readData());
  //Serial.println();
}

void setup() 
{
  Serial.begin(115200);
  SPI.begin();
  SPI.setClockDivider(SPI_CLOCK_DIV128); 
  esp.begin();
  delay(1000);
  Serial.print("setup done\n");
  //send("Hello Slave!");
}

void loop() 
{
    for(int i = 0; i<32; i++)
    {
      unsigned int check = (uint8_t) test_packet[i];
      Serial.print(check,HEX);
      Serial.print(",");
    }
    Serial.print("\n");
    
    
    delay(10);
    send(test_packet);
  }
  Serial.print("\n\n");
  delay(1000);
  
}
```
thanks you for your help so far

earlephilhower, 2020-06-29:
````
    void writeData(const uint8_t * data) 
    {
      writeData((uint8_t *)data, strlen(data));
    }
````

Your UNO code is not writing anything after the first 0 because your Uno is using `strlen` which will, of course, stop at your array's first 0-byte.

Closing.

closed 2020-06-29 by earlephilhower

ben-watkinson, 2020-06-29:
I didnt spot that sorry. thank you for your help