esp8266/Arduino #7407: Analog read with Arduino and Nodemcu Esp8266

thiagoconsales, 2020-06-26:
Hello,

I'm developing a project that uses the A0 port to receive the sensor data, the problem is that the measurements I did with Arduino Uno are very different when I use Nodemcu, and I need to use nodemcu in the project.

Below is a measurement chart using ECG module AD8232 using Arduino and NodeMCU.

Nodemcu:

[image]

Arduino:
[image]

The Arduino readings is better than Nodemcu. 

Bellow a simple test I did in order to find the problem:

[image]

Any idea what is happen?

Thanks!

JiriBilek, 2020-06-27:
Most probably you are stuck on the fact, that ESP8266 ADC pin input voltage range is 0 to 1V.

earlephilhower, 2020-06-27:
Interesting use case, @thiagoconsales, but as @JiriBilek noted the ADC on the 8266 and the AVR are very different.  The 8266 has a 0-1V, low bandwidth, low quality hardware ADC.  The A0 input itself is also modified by some boards to give a 3.3V range via an R-divider network (potentially changing the frequency response and the impedance), but you need to talk w/your board manufacturer or search online for details.

That said, this is not the place for general discussion like this.  Use the https://esp8266.com/forum or https://gitter.im/esp8266/Arduino IRC-like to get general help on projects.

closed 2020-06-27 by earlephilhower