esp8266/Arduino #7408: ESP 01 gives Leaving... Hard resetting via RTS pin...

JohanJomy, 2020-06-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-01
- Core Version:			latest git hash or date
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT
- Flash Size: 		        1MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:               115200

### Problem Description

I was trying to make a home automation project with ESP 01 but it shows:

Leaving...
Hard resetting via RTS pin...

and does not connect with my blink app. 

The ESP 01 has a red led and blue led, the red one is always on and the blue one blinks when uploading.
I even tried the wifi scan from the examples and nothing gets printed on the serial monitor. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

char auth[] = "DDdmF***********jp1LzuADltzCgG";

char ssid[] = "wifi";
char pass[] = "password";

void setup()
{
  Serial.begin(9600);

  Blynk.begin(auth, ssid, pass);
}

void loop()             
{
  Blynk.run();
}

### Debug Messages

esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5*:0*:91:*a:50:5a
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 285968 bytes to 209324...
Wrote 285968 bytes (209324 compressed) at 0x00000000 in 18.3 seconds (effective 124.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

earlephilhower, 2020-06-27:
That "Hard resetting..." message is expected, normal behavior.

There is no core issue here, only a hardware, setup, or Blynk one.  This isn't the forum for those, please use https://esp8266.com or https://gitter.im/esp8266/Arduino or the Blynk repo for general debug and bringup discussion.

closed 2020-06-27 by earlephilhower

HMZain007, 2020-06-27:
Well I think the issue is with your ESP so try a hard reset, hope it will work fine then. 
You can consult the following link for more info

https://electronicsinnovation.com/hard-resetting-via-rts-pin-fixed-explained/#:~:text=Explanation%3A%20Hard%20resetting%20via%20RTS%20pin,-If%20you%20want&text=UART%20Means%20%E2%80%93%20Universal%20Asynchronous%20Receiver,with%20each%20other%20over%20UART.