esp8266/Arduino #7411: No connect to WiFi

LuisPignataro, 2020-06-29:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				WeMos d1 mini (ESP8266EX)
- Core Version:			2.7.1
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			|Wemos D1 mini
- Flash Size: 		4MB
- Upload Using:		SERIAL

[env:d1_mini]
platform = espressif8266@2.3.3
board = d1_mini
framework = arduino
upload_speed = 921600

### Problem Description

I have two d1 mini boards, with the same program.
The problem happens only with one of them, so I rule it out as the router (I provided it with 3 different access points).
If I put tasmota v 8.2 it works.
I am testing with the "WifiClient" example.
I erase falsh too.
The SSID and password are correct.

On some occasions after a long time it connects.
I have forced the use of 11N, persistent (false), and WiFi.disconnect ().
Unsuccessfully.
Also using the 2.6.3 core version with the same result.

##Code

#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "XXX"
#define STAPSK  "XXX"
#endif

const char* ssid     = STASSID;
const char* password = STAPSK;

const char* host = "djxmmx.net";
const uint16_t port = 17;

void setup() {
  //WiFi.persistent(false);
  Serial.begin(9600);

  // We start by connecting to a WiFi network
  
  WiFi.disconnect();
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  Serial.println(WiFi.status(), DEC);
  Serial.setDebugOutput(true);
  //WiFi.setPhyMode(WIFI_PHY_MODE_11N);
  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    //Serial.println("--------------------------------------------------");
    //WiFi.printDiag(Serial);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  
  }

### Debug info in console

scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect

### Using WiFi.printDiag(Serial);
AP is configured for channel 11.  ESP8266 iterates for all channels, but not connect.
--------------------------------------------------
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (10): XXX
Passphrase (10): XXX
BSSID set: 0
state: 2 -> 0 (2)
reconnect
--------------------------------------------------

ZakCodes, 2020-06-29:
If the software works on one of your boards, and they're the same model, it's probably a hardware issue. You'd probably be better off asking your question on the [esp8266 forum](https://www.esp8266.com/). There could be a problem with the antenna on your board or something.

The formatting of this issue isn't properly done. You should have used the preview option when creating the issue to make sure it looks fine like it was mentioned in the issue template.

Here's what your markup should have been: 
~~~md
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:	WeMos d1 mini (ESP8266EX)
- Core Version: 2.7.1
- Development Env: Platformio
- Operating System: Windows

### Settings in IDE

- Module: Wemos D1 mini
- Flash Size: 4MB
- Upload Using: SERIAL
- Upload Speed: 921600

platformio.ini: 
```ini
[env:d1_mini]
platform = espressif8266@2.3.3
board = d1_mini
framework = arduino
upload_speed = 921600
```

### Problem Description

I have two d1 mini boards, with the same program.
The problem happens only with one of them, so I rule it out as the router (I provided it with 3 different access points).
If I put tasmota v 8.2 it works.
I am testing with the "WifiClient" example.
I erase falsh too.
The SSID and password are correct.

On some occasions after a long time it connects.
I have forced the use of 11N, persistent (false), and WiFi.disconnect ().
Unsuccessfully.
Also using the 2.6.3 core version with the same result.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>

#ifndef STASSID
#define STASSID "XXX"
#define STAPSK "XXX"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const char* host = "djxmmx.net";
const uint16_t port = 17;

void setup() {
    //WiFi.persistent(false);
    Serial.begin(9600);

    // We start by connecting to a WiFi network

    WiFi.disconnect();
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    Serial.println(WiFi.status(), DEC);
    Serial.setDebugOutput(true);
    //WiFi.setPhyMode(WIFI_PHY_MODE_11N);
    /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
    would try to act as both a client and an access-point and could cause
    network-issues with your other WiFi-devices on your WiFi-network. */
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(1000);
        //Serial.println("--------------------------------------------------");
        //WiFi.printDiag(Serial);
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop() {

}
```

### Debug Messages

```txt
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
Using WiFi.printDiag(Serial);
AP is configured for channel 11. ESP8266 iterates for all channels, but not connect.
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 1
Auto connect: 1
SSID (10): XXX
Passphrase (10): XXX
BSSID set: 0
state: 2 -> 0 (2)
reconnect
```
~~~

Feel free to use this and modify it to update your issue's description.

LuisPignataro, 2020-06-29:
Tanks @ZakCodes, I edited the description, sorry.

It is possible that the problem is related to the hardware, I am testing it.

Perhaps this will serve to have a better documentation of the result of Serial.setDebugOutput (true);

The codes are not well documented. Or at least I can't find them.

state: 0 -> 2 (b0)
state: 2 -> 0 (2)

If you could help me with this, I thank you in advance.

The problem started after updating the firmware and it works perfect with tasmota, so while it could be a hardware related problem, the software is not totally separate from this problem.

Maybe tasmota solves something that I am not considering by software, however if it were a pure hardware problem, tasmota would work.
It is important to clarify that I do the tests without modifying anything of the hard. In fact it is very simple.

earlephilhower, 2020-06-29:
Please enable the `Tools->Erase Flash->All Flash Contents` setting in the IDE and then re-flash.  The SDK blob has private RF tuning settings which are not consistent between SDK versions, and sometimes this causes issues like what you're seeing.

LuisPignataro, 2020-06-29:
Hi, @earlephilhower , is enabled. Erase all flash doesn't work. Neither did WiFi.persistent(false);

LuisPignataro, 2020-06-29:
Update: Using WiFi.RSSI() I could see that the signs are very low. before connection. About 35bd.
After connecting to the network, the signal rises to 60db.
I am using an access point just 2 meters from the device, but with a power of 10%. Only so that my brain does not fry.
Increasing the antenna gain begins to work.
This does not explain the reason why Tasmota works smoothly.

earlephilhower, 2020-06-30:
The reported db is *negative* so -35db >>> -60db (almost 30x stronger).  35db is kind of insanely strong, and -60db is not too bad.

> I am using an access point just 2 meters from the device, but with a power of 10%

That's the issue here as far as I can see.  Nothing to do with the Arduino core, more to do with xmit and receive power levels and the distance involved.

For general talk about power levels, please use https://esp8266.com or https://gitter.im/esp8266/Arduino .  There are folks who can help you there with adjusting xmit and receive gains.

closed 2020-06-30 by earlephilhower