esp8266/Arduino #7418: Can't rewrite the last char in a file using seek()

ardubit, 2020-06-30:
### Problem Description

Can't rewrite the last char in a file. Is it a bug or where am I wrong?

```
fileObject = SPIFFS.open(fileName, "a+"); // Open for appending (writing at end of file)
    if (fileObject)
    {
#ifdef SERIAL_OUT
      Serial.print(F(" -> Created File Size: "));
      Serial.println(fileObject.size());
      Serial.print(F(" -> Position: "));
      Serial.println(fileObject.position());
#endif
      // Moves current position in a file from end -offset
      bool seekResult = fileObject.seek(1, SeekEnd);  
      if (seekResult)
      {
#ifdef SERIAL_OUT
        int position = fileObject.position(); 
        Serial.print(F(" -> Current Position: "));
        Serial.println(position);
#endif

        String jsonEnd;
        jsonEnd = " ],\n";
        jsonEnd += " \"distance\": ";
        jsonEnd += String(distanceKm) + "\n";
        jsonEnd += "}\n";

        fileObject.print(jsonEnd);
        fileObject.close();
```

Output:
```
 -> Created File Size: 739
 -> Position: 0
 -> Current Position: 738
```

Thus, the current position is -1 char, but after close() it's still writing data without removing the last char.

devyte, 2020-07-01:
SPIFFS is deprecated in our core, and so we're not accepting issues for now. The alternative is LittleFS, see examples. 
We would accept a PR for this, if you investigate the problem yourself and propose a fix. 
The relevant exit point from our core's code is here:
https://github.com/esp8266/Arduino/blob/83166f948bedff85543b1cb5532b905746744df7/cores/esp8266/spiffs_api.h#L416
You can see there that it's a thin layer over the SPIFFS lib. The entry point to the SPIFFS lib is here:
https://github.com/esp8266/Arduino/blob/799c0f67743facfab2867900c4e07cb77440d355/cores/esp8266/spiffs/spiffs_hydrogen.cpp#L600

Closing due to deprecated feature.

closed 2020-07-01 by devyte

ardubit, 2020-07-02:
The solution is next: "a+" open for appending is forcing to write at the end of the file, with "r+" seek() is working well.