esp8266/Arduino #7422: DNS lookup results in ip 255.255.255.255 randomly

gsieben, 2020-07-02:
Hi Guys,

My extensive program for ESP8266 has a big problem. The connection to the MQTT server fails frequently due to error -2 (network problem). If I used the IP address and not the hostname for the connection, the whole thing worked. This made me to do a DNS lookup right after the network connection. The result. 

Seems like 50% the DNS request fails.I use the command: "DNSRequest("galaxia"); (Galaxia is the name of the server I want to reach). 

 I get the IP 255.255.255.255 back. If this happened after the starup, it will not correct itself. Even if I start the same request after 10 minutes, it ends in 255.255.255.255 instead of 10.7.7.247. Is there a workaround for this bug? 

The corresponding code:
```
void MQTTSetup() {
    if (settings.mqtt.Broker == "null" || settings.mqtt.Broker.isEmpty()) {
        // The Broker is unkonwn
        Serial.println("MQTT: WARNING: Der Broker ist nicht bekannt.");

      } else {
        **DNSRequest(settings.mqtt.Broker.c_str());**           // Test-> DNS lookup ends up in -> 255.255.255.255 instead of 10.7.7.247 (50%)

        Serial.print("MQTT: Verbinde mit Broker: '" + settings.mqtt.Broker + "' Port: '" + settings.mqtt.Port + "'");  

        if (!MQTTclient.connected()) {
```

TD-er, 2020-07-03:
Can you also share the code of this `DNSRequest` function?

earlephilhower, 2020-07-03:
@gsieben you have ignored the issue template and are missing a MCVE.  W/o those answers and an MCVE to run, there's nothing we can do here.

Please collect the requested information and build a MCVE so that others can reproduce a failing run, and open a new issue with the if you're still hitting a failure in the core.  You can also try https://esp8266.com or https://gitter.im/esp8266/Arduino .

closed 2020-07-03 by earlephilhower