esp8266/Arduino #7425: Problem connect to WIFI

Shazy946, 2020-07-04:
#### Platform

- Hardware:					ESP-12
- Core Version:			2.5
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			other
- Flash Size: 		4MB
- lwip Variant:     v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   115200

### Problem Description

Hello guys! I'm new to GitHub and I have a problem. 3 days ago, I bought a kit with esp8266 and I have a problem. They offered me a tutorial with many projects, and the internet connection does not work. I add the code in Arduino, I load it on esp8266, but the internet connection doesn't appear. I tried it on my phone and laptop, but it doesn't work. I'll add the code given by them for the internet connection and for control below an LED from web. Maybe you can help me. I apologize for my English, but I don't know English ...Thanks!

CODE:

#include <ESP8266WebServer.h>

const char* ssid = "ssid";
const char* password = "password";

ESP8266WebServer server(80);

void setupServer()
{ 
  server.on("/", htmlIndex);
  server.begin();
  Serial.println("HTTP server started");
}

void htmlIndex() 
{

  int replyCode = 200;
  String contentType = "text/plain";
  String message = "Hello world!";
  server.send(replyCode, contentType, message);
}

void connectToWiFi()
{ 
  Serial.println("Connecting to the WiFi");
  WiFi.mode(WIFI_STA);

  
  WiFi.begin(ssid, password);

  Serial.println("Waiting for connection");
  
  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup() 
{
 
  Serial.begin(115200);
  
  delay(1000);

  connectToWiFi();
  setupServer();
}

void loop() 
{

  server.handleClient();
}

CODE FOR WIFI+LED

#include <ESP8266WebServer.h>

const char* ssid = "............";
const char* password = "........";

ESP8266WebServer server(80);

const int LED = D6;

String storage = "";

String page =
R"(
<html lang='en'>
    <head>
        <title>Play with an LED</title>
        <meta name='viewport' content='width=device-width, initial-scale=1'>
        <link rel='stylesheet' href='https://use.fontawesome.com/releases/v5.1.0/css/all.css'/>
        <link rel='stylesheet' href='https://stackpath.bootstrapcdn.com/bootstrap/4.1.2/css/bootstrap.min.css'/>
        <script src='https://code.jquery.com/jquery-3.3.1.min.js'></script>
        <style>
            .css_on 
            {
                color: orange;
            }
            .css_off 
            {
                color: black;
            }
        </style>
    </head>
    <body>
    <div class='container-fluid text-center'>
        <div class='row text-center'>
            <div class='col-sm-12' style='color:red;font-size:7vh'>
                <p>Let's play with a bulb.</p>
            </div>
        </div>
        
        <div class='row'>
            <div  class='col-sm-12'>
                <i id='id_bulb' class='fas fa-lightbulb css_off' style='font-size:25vh'></i>
            </div>
        </div>
            <br>
        <div class='row'>
            <div  class='col-sm-12' style='font-size:7vh'>
                <p>Click the button:</p>
            </div>
        </div>
        
        <div class='row'>
            <div  class='col-sm-12'>
                <input type='button' class='btn btn-primary' id='id_button' value='Off'  style='font-size:7vh'>
            </div>
        </div>
    </div>
    
    <script>
    $(document).ready(function(){
        $('#id_button').click(function(){
            var current_state = $("#id_bulb").hasClass("css_off");
                
            if(current_state == true)
            {
                $.ajax({
                  url:'/led',
                  type: 'POST',
                  data: {state: "On"},
                });
                $("#id_button").val("On");
                $("#id_bulb").removeClass("css_off").addClass("css_on");
            }
            else
            {
                $.ajax({
                  url:'/led',
                  type: 'POST',
                  data: {state: "Off"},
                });
                $("#id_button").val("Off");
                $("#id_bulb").removeClass("css_on").addClass("css_off");
            }
        });
    });
    </script>
    </body>
</html>
)";

void led()
{

  storage = server.arg("state");
  
  if(storage == "On")
  {
   
    digitalWrite(LED, HIGH);
  }
  else 
  {
   
    digitalWrite(LED, LOW);
  }

  
   server.send(200,"text/html", "ok");
}

void htmlIndex() 
{
  
  server.send(200, "text/html", page);
}

void connectToWiFi()
{ 
  Serial.println("Connecting to the WiFi");
 
  WiFi.mode(WIFI_STA);

  
  WiFi.begin(ssid, password);

 
  Serial.println("Waiting for connection");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void setupServer()
{ 
 
  server.on("/", htmlIndex);
  server.on("/led", led);

  
  server.begin();
  
 
  Serial.println("HTTP server started");
}

void setup() 
{
  
  pinMode(LED, OUTPUT);

 
  Serial.begin(115200);
  
  
  delay(1000);
  
  
  connectToWiFi();
  setupServer();

  
  delay(4000);
}

void loop()
{
  
  server.handleClient();
}

d-a-v, 2020-07-04:
Try to upgrade to version 2.7.1.
In any case, first thing first, use the option "Erase All Flash" in the tool menu.

Shazy946, 2020-07-05:
> Try to upgrade to version 2.7.1.
> In any case, first thing first, use the option "Erase All Flash" in the tool menu.

Am incercat cum ai spus...Am facut upgrade la 2.7.1 si am setat "Erase All Flash" in the tool menu, dar tot nu-mi apare conexiunea la internet.

Shazy946, 2020-07-05:
> Try to upgrade to version 2.7.1.
> In any case, first thing first, use the option "Erase All Flash" in the tool menu.

I tried as you said ... I upgraded to 2.7.1 and set "Erase All Flash" in the tool menu, but my internet connection still doesn't appear.

d-a-v, 2020-07-05:
Just to be sure, the examples above, did you try them "as-is" or did you adapt `ssid` and `password` to your environment ?

Shazy946, 2020-07-05:
Those 2 codes are offered by the company from which I bought the kit, so the codes are good, and I changed to ssid and pass with the ones I wanted.

Shazy946, 2020-07-05:
> Just to be sure, the examples above, did you try them "as-is" or did you adapt `ssid` and `password` to your environment ?

I don't want to advertise, but here's their guide, read lesson 15. Thank you!

[Guide.pdf](https://github.com/esp8266/Arduino/files/4875312/Guide.pdf)

TD-er, 2020-07-05:
Just glanced through the code and came across this part:
```c++
String page =
R"( .....
```
Don't know that syntax.

Just curious, what kind of macro is that?

earlephilhower, 2020-07-05:
Apropos of nothing, @TD-er, it's a GCC multiline raw string literal.

Shazy946, 2020-07-06:
The first time I thought it was a broken chip, so I bought another one, but it's the same problem, so I tried to use another code that I attach below, with this code it worked, but I need to know about my problem because the whole guide is based on that code ...

A functional code:

`

#include <ESP8266WiFi.h>

const char* ssid = "Testy";
const char* password = "12345678";

WiFiServer server(80);

void setup() {
  delay(1000);
  WiFi.softAP(ssid, password); 
  server.begin();
  delay(2000);
   
  pinMode(BUILTIN_LED, OUTPUT);
  delay(100);
  digitalWrite(BUILTIN_LED, HIGH); 
}

void loop() {
  
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  
 
  while(!client.available()){
    delay(1);
  }
  
  
  String req = client.readStringUntil('\r');
  client.flush();
  
 
  if (req.indexOf("/LED_1/off") != -1)
    digitalWrite(BUILTIN_LED, HIGH); 
  else if (req.indexOf("LED_1/on") != -1)
    digitalWrite(BUILTIN_LED, LOW); 
  else if (req.indexOf("/") != -1)
    ; 
  else {
    client.stop();
    return;
  }
  client.flush();
  
  String HTML = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html> <h1>DIY HowToDoIt</h1> <h2>ESP8266 Web Server</h2>";
  HTML += "<h3>Blue LED is now ";
  HTML += (digitalRead(BUILTIN_LED))?"OFF":"ON</h3>"; 
  HTML += "<h3>Blue LED <button onclick=\"window.location.href='/LED_1/on'\">LED ON</button>&nbsp;<button onclick=\"window.location.href='/LED_1/off'\">LED OFF</button></h3></html>";
 

  client.print(HTML);
  delay(10);
  
}

`

PRINT SCREEN WITH FUNCTIONAL CODE: https://i.imgur.com/pB6BzHZ.png

P.S-  After my tests, I discovered that if I change in my code "WiFi.begin (ssid, password);" with "WiFi.softAP (ssid, password);" my WIFI network appears, but it does not connect to Serial Monitor.

PRINT SCREEN WITH THAT LINE CHANGED AND SERIAL MONITOR: https://imgur.com/a/MjATJ7n

d-a-v, 2020-07-06:
@Shazy946 you jumped to "38. Lesson 33" p274 of your guide.

esp8266 has two modes: it can create a WiFi access point "=AP mode",
or it can connect to your already existing access point (=STA mode).

Your "working sketch" is in AP mode.

The initial sketch is in STA mode. In this mode, no WiFi network is created. It is the most common mode (like your computer or your phone connecting to WiFi). ESP in that case can connect to the internet and is reachable through its IP address.

Shazy946, 2020-07-06:
> @Shazy946 you jumped to "38. Lesson 33" p274 of your guide.
>
> esp8266 has two modes: it can create a WiFi access point "=AP mode",
> or it can connect to your already existing access point (=STA [...]

Aaaaaa.I understand! You're right. I haven't tried to add my WiFi network to ssid and pass so far, I just tried to add some new ones believing that it will have to create a new network. Thank you, now it works!

devyte, 2020-07-06:
I don't see a core issue here so far.
For general support and "how do I" type discussions, please refer to esp8266.com or stackoverflow. 
Closing.

closed 2020-07-06 by devyte