esp8266/Arduino #7442: Error in compile after  "Upgrade to GCC 10.1 toolchain" commit.

mbm60, 2020-07-09:
#### Platform

- Hardware:			[ESP-12]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]

### Problem Description

After "Upgrade to GCC 10.1 toolchain" commit, arduino cannot compile any programs and displays this message:

**xtensa-lx106-elf-g++: error: unrecognized command line option '-std=gnu++17'**

Gizmodo, 2020-07-09:
Try to use -std=c++17

earlephilhower, 2020-07-09:
Don't change the platform options, that's not a good idea.

Update the toolchain using the `tools/get.py`.  `git` won't automatically do it for you, you need to do it yourself.

mbm60, 2020-07-09:
> 
> 
> Don't change the platform options, that's not a good idea.
> 
> Update the toolchain using the `tools/get.py`. `git` won't automatically do it for you, you need to do it yourself.

Thanks for your reply. I've already done this but the error still exists.

earlephilhower, 2020-07-10:
Please change to the toolchain directory (`tools/xtensa-lx106-elf/bin/`) and run `xtensa-lx106-elf-gcc --version`

You should get the following output:
````
xtensa-lx106-elf-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
````

I think something failed on the `get.py` update for you and you're still at gcc4.x.

Another option:  if you installed from the Arduino IDE as well as from git, you will have 2 copies of the toolchain and Arduino won't use the git version.  Simply remove the ESP8266 package from the boards manager in this case and restart.

mbm60, 2020-07-10:
Yes i had git and stable version. i removed stable version and it's working fine now.
Thanks for your help.

closed 2020-07-10 by mbm60