esp8266/Arduino #7443: Wemod D1 mini + MicroSD Shield => Loops in "Initializing SD card..."

AKTanara, 2020-07-09:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[WEMOS D1 mini]
- Core Version:			[Sorry, DN where to find it...]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 10]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

HW:
I'm using (Wemos D1 mini) + (MicroSD Shield for Wemos D1 mini) + (8GB HCSD formatted as FAT32) with these Connections:
D5                  CLK
D6                  MISO
D7                  MOSI
D8                     CS

SW:
(Arduino IDE) + ( ESP8266 2.7.2 in Board manager) + (WEMOS D1 mini selected as Board) + Example: SD(ESP8266) as code 

But it loops over and over again as you can see in debug...

What I tried with no success:
Using older versions of ESP8266 Board managers like 2.0.0
Changing MicroSD

I cant even know if the SPI connection is established or not!...

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <SPI.h>
#include <SD.h>

const int chipSelect = 8;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  Serial.print("Initializing SD card...");
  SPI.begin();

  // see if the card is present and can be initialized:
  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    // don't do anything more:
    return;
  }
  Serial.println("card initialized.");
}

void loop() {
  // make a string for assembling the data to log:
  String dataString = "";

  // read three sensors and append to the string:
  for (int analogPin = 0; analogPin < 3; analogPin++) {
    int sensor = analogRead(analogPin);
    dataString += String(sensor);
    if (analogPin < 2) {
      dataString += ",";
    }
  }

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("datalog.txt", FILE_WRITE);

  // if the file is available, write to it:
  if (dataFile) {
    dataFile.println(dataString);
    dataFile.close();
    // print to the serial port too:
    Serial.println(dataString);
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
}
```

### Debug Messages

```
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
Initializing SD card...
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
Initializing SD card...
```

devyte, 2020-07-09:
You seem to be confusing the gpio numbering with the board numbering. You said CS=D8, but your sketch says CS=8, they are not the same thing. I believe gpio 8 is forbidden (flash data pin), so no wonder you're getting crashes.
Closing due to user error.

closed 2020-07-09 by devyte

AKTanara, 2020-07-09:
Thanks for your quick reply and explanation. Can you please tell me which number should be chosen for CS?

Many Thanks in advance...

devyte, 2020-07-09:
I already did: you're trying to use 8 when it should be D8. Or any of the free pins, really.
The Di defines are convenience macros that map from the D numbering to the GPIO numbering. 
Please study the pin numbering for these boards, it sounds like you haven't looked that up. Google is your friend.

AKTanara, 2020-07-10:
Thanks again,
Just for the record I found out that : D8 in WEMOS D1 mini is equivalent of GPIO 15. So i should have either used
const int chipSelect = D8;
or
const int chipSelect = 15;
which both work fine.

And for the others who might be looking for here is full mapping of most ESP8266 based boards and some recommendations:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

Regards