esp8266/Arduino #7445: Exception (9): epc1=0x4020a730 epc2=0x00000000 epc3=0x00000000 excvaddr=0x72696662 depc=0x00000000

skp1996, 2020-07-10:
Basic Infos
Hardware
Hardware: ESP-12

Description
When I send a data from Mobile App created using MITApp Inventer to NodeMCU it crashes after certain time probably 1 Minute and giving the error "
Exception (9):
epc1=0x4020a730 epc2=0x00000000 epc3=0x00000000 excvaddr=0x72696662 depc=0x00000000
and again restarts after certain time

Settings in IDE
Module: NodeMCU 1.0 (ESP-12E Module)
Flash Size: 4MB
CPU Frequency: 80Mhz
Flash Mode: qio
Flash Frequency: 40Mhz
Upload Using: SERIAL
Reset Method: nodemcu

Sketch
// The Code is modified and written by Alok Shaw, the creator of the Youtube channel : www.youtube.com/roboshala and runs an educational blog : www.roboshala.com

#include <ESP8266WiFi.h>
#include<FirebaseArduino.h>

#define FIREBASE_HOST "my-project-1507454246873.firebaseio.com"      //Your Firebase Project URL goes here without "http:" , "\" and "/"
#define FIREBASE_AUTH "0mLHS8rBIlJLMkChxafjZIjrHPyXoEXYwY7QRe0c"       //Your Firebase Database Secret goes here
#define WIFI_SSID "JioFiber"                                               //your WiFi SSID for which yout NodeMCU connects
#define WIFI_PASSWORD "Srv.kp14"                                      //Password of your wifi network 

#define Relay1 12 //D6
int val1;

#define Relay2 14 //D2
int val2;

#define Relay3 4  //D1
int val3;

#define Relay4 5 //D5
int val4;

void setup() 
{
  Serial.begin(115200);                                                   // Select the same baud rate if you want to see the datas on Serial Monitor
  pinMode(Relay1,OUTPUT);
  pinMode(Relay2,OUTPUT);
  pinMode(Relay3,OUTPUT);
  pinMode(Relay4,OUTPUT);

  digitalWrite(Relay1,HIGH);
  digitalWrite(Relay2,HIGH);
  digitalWrite(Relay3,HIGH);
  digitalWrite(Relay4,HIGH);

  ESP.eraseConfig();
  WiFi.begin(WIFI_SSID,WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status()!=WL_CONNECTED){
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.setInt("S1",0);                     //Here the varialbe"S1","S2","S3" and "S4" needs to be the one which is used in our Firebase and MIT App Inventor
  Firebase.setInt("S2",0); 
  Firebase.setInt("S3",0); 
  Firebase.setInt("S4",0); 
  
}
void firebasereconnect()
{
  Serial.println("Trying to reconnect");
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  }

void loop() 
{
  if (Firebase.failed())
      {
      Serial.print("setting number failed:");
      Serial.println(Firebase.error());
      firebasereconnect();
      return;
      }
      
  val1=(bool)Firebase.getString("S1").toInt();                                        //Reading the value of the varialble Status from the firebase
  
  if(val1==1)                                                             // If, the Status is 1, turn on the Relay1
     {
      digitalWrite(Relay1,LOW);
      Serial.println("light 1 ON");
    }
    else if(val1==0)                                                      // If, the Status is 0, turn Off the Relay1
    {                                      
      digitalWrite(Relay1,HIGH);
      Serial.println("light 1 OFF");
    }

  val2=(bool)Firebase.getString("S2").toInt();                                        //Reading the value of the varialble Status from the firebase
  
  if(val2==1)                                                             // If, the Status is 1, turn on the Relay2
     {
      digitalWrite(Relay2,LOW);
      Serial.println("light 2 ON");
    }
    else if(val2==0)                                                      // If, the Status is 0, turn Off the Relay2
    {                                      
      digitalWrite(Relay2,HIGH);
      Serial.println("light 2 OFF");
    }

   val3=(bool)Firebase.getString("S3").toInt();                                        //Reading the value of the varialble Status from the firebase
  
  if(val3==1)                                                             // If, the Status is 1, turn on the Relay3
     {
      digitalWrite(Relay3,LOW);
      Serial.println("light 3 ON");
    }
    else if(val3==0)                                                      // If, the Status is 0, turn Off the Relay3
    {                                      
      digitalWrite(Relay3,HIGH);
      Serial.println("light 3 OFF");
    }

   val4=(bool)Firebase.getString("S4").toInt();                                        //Reading the value of the varialble Status from the firebase
  
  if(val4==1)                                                             // If, the Status is 1, turn on the Relay4
     {
      digitalWrite(Relay4,LOW);
      Serial.println("light 4 ON");
    }
    else if(val4==0)                                                      // If, the Status is 0, turn Off the Relay4
    {                                      
      digitalWrite(Relay4,HIGH);
      Serial.println("light 4 OFF");
    }    
  ESP.;
}

In Serial Monitor

light 1 ON
light 2 ON
light 3 ON
light 4 ON

Exception (9):
epc1=0x4020a730 epc2=0x00000000 epc3=0x00000000 excvaddr=0x72696662 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffbf0 end: 3fffffc0 offset: 0190
3ffffd80:  00000001 40103a66 3ffed1d8 3ffffe40  
3ffffd90:  0000000b 3fff0188 3ffefeac 40204cac  
3ffffda0:  00001078 000001bb 3ffefeac 40205b74  
3ffffdb0:  3ffe8852 0000003b 3ffefeac 40205fa0  
3ffffdc0:  0000003b 3ffee98c 00000020 40100a0f  
3ffffdd0:  3ffffe18 0000003b 3ffffe00 40206ac8  
3ffffde0:  3ffee96c 3ffffe00 3ffe8852 3ffe87ca  
3ffffdf0:  3ffee96c 3ffffe60 3ffefea4 4020432f  
3ffffe00:  3fff01c4 003b003f 00fffe60 3fff03d4  
3ffffe10:  0036003f 00fee98c 3fff013c 0027002f  
3ffffe20:  00fe8822 3ffee98c 3fffff28 402038a8  
3ffffe30:  3ffee96c 3ffee98c 3fffff28 40202d06  
3ffffe40:  3fff0148 4024f0bd 00000057 3ffe87a3  
3ffffe50:  3ffefcfc 00000000 0000000a 00000002  
3ffffe60:  3fff0188 40203318 00000020 40100a44  
3ffffe70:  00004018 3fffff4c 3fffff38 3ffee968  
3ffffe80:  3ffffec8 3ffffec8 3ffe8822 402033b4  
3ffffe90:  3ffffecc 3fffff7c 00000002 3ffee96c  
3ffffea0:  3fffff4c 3ffffec0 3fffff28 402030be  
3ffffeb0:  3ffffec0 3fffff70 3ffefeac 4020340a  
3ffffec0:  3ffefea4 3ffefbdc 3ffee98c 3ffeeae0  
3ffffed0:  3fffff00 3ffee968 3fffff7c 402033b4  
3ffffee0:  3fffff58 3fffff00 3fffff28 3ffeeae0  
3ffffef0:  00000000 3fffff70 3ffee968 40203a63  
3fffff00:  3ffefea4 3ffefbdc 696c004e 40208885  
3fffff10:  3ffe87ad 4024f0bd 3fffff00 00000000  
3fffff20:  402060b8 3ffee9b0 4020abc8 3ffefea4  
3fffff30:  3ffefbdc 00000000 3ffee98c 3ffee98c  
3fffff40:  00000000 00000000 3fffff7c 3ffefe88  
3fffff50:  00000000 3fffff7c 3ffe8727 40206afe  
3fffff60:  007a1200 3ffee968 3ffee968 4020125a  
3fffff70:  00000000 3ffee92c 80fee968 40003153  
3fffff80:  00000000 82000000 00000001 4010017c  
3fffff90:  3fffdad0 00000000 3ffeeaa0 3ffeeae0  
3fffffa0:  3fffdad0 00000000 3ffeeaa0 40207970  
3fffffb0:  feefeffe feefeffe 3ffe857c 40100cd1  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
connecting..
Please help me. I am stuck with this problem from days.

Tech-TX, 2020-07-10:
Firstly,  this is a section for issues with the core and library files,  not a "my code breaks,  help"  forum. 
Second,  you need to install the exception decoder,  as that raw dump is worthless.   
https://github.com/me-no-dev/EspExceptionDecoder

These are not the droids you're looking for...

closed 2020-07-10 by earlephilhower

skp1996, 2020-07-14:
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4020a7fc: HTTPClient::connected() at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 475
EXCVADDR: 0x00000035

Decoding stack results
0x401009f3: free(void*) at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x40204cac: HTTPClient::disconnect(bool) at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 434
0x40205bd4: HTTPClient::end() at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 425
0x40204362: FirebaseHttpClientEsp8266::end() at C:\Users\Sourav Kumar Purohit\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseHttpClient_Esp8266.cpp line 56
0x40202fad: FirebaseCall::~FirebaseCall() at C:\Users\Sourav Kumar Purohit\Documents\Arduino\libraries\firebase-arduino-master\src\Firebase.cpp line 152
0x40202ffc: FirebaseGet::~FirebaseGet() at C:\Users\Sourav Kumar Purohit\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h line 101
0x40203acb: FirebaseArduino::getString(String const&) at C:\Users\Sourav Kumar Purohit\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp line 113
0x40206b62: String::String(char const*) at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\WString.cpp line 36
0x40201381: loop() at C:\Users\Sourav Kumar Purohit\Desktop\FirebaseAutomation\Firebase_HomeAutomation/Firebase_HomeAutomation.ino line 112
0x4010017c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 177
0x402079d4: loop_wrapper() at C:\Users\Sourav Kumar Purohit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 197

ExceptionDecoder Will show this result. I couldnot understand this, please help me because I am new to this