esp8266/Arduino #7448: mdns responder stops responding

goatzillax, 2020-07-12:
- Hardware:					ESP8266
- Core Version:			Uh, 2.7.2?
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			Yes
- Flash Size: 		4MB
- lwip Variant:   v2 lower memory
- Reset Method:		?
- Flash Frequency:	?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600 (serial upload only)

### Problem Description

The mDNS responder is very flaky.

It will usually blast out the records upon boot but after that the responses are a crapshoot.

I have a bash shell which just keeps running a lookup every second.

Sun Jul 12 03:41:05 CDT 2020
esp8266.local	192.168.2.112

...

Failed to resolve host name 'esp8266.local': Timeout reached
Sun Jul 12 03:43:24 CDT 2020

That may correspond to the 120s TTL.

After that, it's a crapshoot.  Sometimes it will send a response.  Sometimes it will say it is sending an answer but it never apparently sends it, and sometimes it will never send an answer at all.

In this case, the mDNS responder didn't even respond until:

04:13:07.500 -> [MDNSResponder] _callProcess (1923416, triggered by: 192.168.2.110)
04:13:07.500 -> [MDNSResponder] _parseMessage (Time: 1923417 ms, heap: 49376 bytes, from 192.168.2.110(5353), to 224.0.0.251(5353))
04:13:07.533 -> [MDNSResponder] _readRRQuestion
04:13:07.533 -> [MDNSResponder] _readRRQuestion esp8266.local Type:0x0001 Class:0x0001 Multicast
04:13:07.533 -> [MDNSResponder] _replyMaskForHost: 0x1
04:13:07.533 -> [MDNSResponder] _parseQuery: Host reply needed 0x1
04:13:07.533 -> [MDNSResponder] _readRRQuestion
04:13:07.533 -> [MDNSResponder] _readRRQuestion esp8266.local Type:0x001C Class:0x0001 Multicast
04:13:07.533 -> [MDNSResponder] _parseQuery: Host reply needed 0x1
04:13:07.567 -> [MDNSResponder] _parseQuery: Sending answer(0x1)...
04:13:07.567 -> [MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
04:13:07.567 -> [MDNSResponder] _prepareMDNSMessage
04:13:07.567 -> [MDNSResponder] _prepareMDNSMessage: ID:0 QR:1 OP:0 AA:1 TC:0 RD:0 RA:0 R:0 QD:0 AN:1 NS:0 AR:0
04:13:07.567 -> [MDNSResponder] _writeMDNSAnswer_A (192.168.2.112)
04:13:07.567 -> [MDNSResponder] _parseMessage: Done (Succeeded after 64 ms, ate 168 bytes, remaining 49208)

Like 30 minutes later.

It responded again at 4:15, and again at 4:17, and then it went silent again.

All of this was visible from both wireshark and the serial debug.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Uh, I'm just using the included example.

mDNS_Web_Server.cpp

### Debug Messages

d-a-v, 2020-08-12:
@goatzillax mDNS is known to have some issues.
It is under reconstruction. Next version is a complete rewrite, will be interface agnostic and is [there](https://github.com/Labor-Et-Ars/ESP8266-Arduino/tree/LEAmDNS2x).

rrelande, 2024-01-25:
I am having the same issue. my configuration is 
Arduino esp8266 3.2.0
as i look into the component, it seems to use the LEAmDNS and not the legacy mDNS. 

reading above, this seems to be an old issue on the legacy mDNS which should be solved by the LEAmDNS. 
However VS Code is clearly showing that LEAmDNS is in use.