esp8266/Arduino #7449: Serial communication and  the sweet old ets Jan  8 2013,rst cause:4, boot mode:(3,6)

dhajnes, 2020-07-12:
#### Platform

- Hardware:				[ESP8266EX] (chip)
- Core Version:			[N/A]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 7]

### Settings in IDE

- Module:  	[NodeMCU spinoff- HW-628 V1.1]
- Flash Mode:	[qio]
- Flash Size: 	[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:  [nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:	[80Mhz]
- Upload Using:	[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Hi, I've looked through some similar issues, but I did not find any helpful info, thus I am starting a new Issue. Otherwise, please redirect me to one, that I have possibly missed. Starting to be a lil desperate here. :D

I am trying to do the following:
* Sense information from sensors on Arduino UNO
* Send them through RX TX serial com to NodeMCU (spinoff- HW-628 V1.1) with ESP8266EX chipset
* Make the NodeMCU send this to an app and show graphs etc. (**not in the scope of this question!**)

I have got a problem with the second bulletpoint. :-1: 
So I remade the code for debugging purposes, so instead of arduino sending the data to the NodeMCU, I am trying to communicate with it, to see, whether is everything ok. **And boy, it is not!**
I upload my code to the MCU and opened the Serial Monitor to try and send some info down the line. But this showed up:
` 
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v39c79d9b
~ld
`
*Cause 4* means that the WatchDog is resetting my MCU beacuse of some *h/w watchdog trigger*. There were some comments on various issues in the *rst cause 4* talking about "Accessing area close to NULL operator could lead into watchdog reset", but I am not aware of any bitwise operations in my code. 

Or else I have no idea how could I come close to some NULL operators?! But I am kind of a novice in matter of communication of systems, so my code might be actually horribly written. :laughing: 

1.  I expect to see an asterisk ('*') in the serial monitor, which would be read by the arduino and it would send an asterisk ('*') back and start sending data byte after byte.
2. After seeing the asterisk ('*'), I would try sending data in, manually, to see whether the communication protocol is working. BUT. I did not manage to even see the asterisk. :-1: 

Here is my code. I tried to be as close to MCVE as possible.

```cpp

#include <WiFi.h>
#include <SoftwareSerial.h>
#include <ESP8266WebServer.h>
#include <stdlib.h>

char* ssid     = "Eufloria";
char* password = "";
WiFiServer server(80);

// data arrays init
const int sample_length = 10;
int data_indx = 0;
int data_temp[sample_length];
int data_moist1[sample_length];
int data_moist2[sample_length];
int data_light[sample_length];
int data_RH[sample_length];
// end of data arrays init

int data_counter = 0;
unsigned char comIN = 0;
char buf[11];
//bool showdata = false;
bool showdata = true; // "= true" is only operative for debugging, normally "= false"
bool eOF = false;

void setup()
{

  pinMode(LED_BUILTIN, OUTPUT); //Declare Pin mode
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  server.begin();
  Serial.begin(115200);
}

void loop() {

  delay(10000); // for debugging purposes, the "showdata = true" is set from the
                // beginning, the wifi comm is not yet coded
                // waits for 10 seconds, so I have time to open the serial monitor

  if(showdata == true){
    Serial.print('*');
  
    while (Serial.available()) {
      comIN = Serial.peek();
      switch (comIN){
        case '*': // signals start of transmission
        comIN = Serial.read();
        break;

        case 'x': // signals end of a subsegment of data
        comIN = Serial.read();
        data_counter++;
        break;

        case '#': // signals end of data transmission
        comIN = Serial.read();
        eOF = true;
        data_indx = 0;
        break;
  
        default: // other characters are only values
          // debug signal - not even reached :/
          digitalWrite(LED_BUILTIN, HIGH);  
          delay(100);              
          digitalWrite(LED_BUILTIN, LOW);    
          delay(100);
          
          comIN = Serial.read();

          // reconstructing chars into char array, then to an int via atoi()
          int char_n = 0;
          while(comIN != ';'){
            buf[char_n] = comIN;
            char_n++;
            comIN = Serial.read();
          }
          buf[char_n] = '\0';
          int dato = atoi(buf);

          // saving incoming data into arrays; data_counter increases for every "x" incoming,
          // which divides data into subcategories below
          switch(data_counter){
            case 0:
              Serial.println("temp");
              data_temp[data_indx] = dato;
            break;

            case 1:
              Serial.println("moist1");
              data_moist1[data_indx] = dato;
            break;

            case 2:
              Serial.println("moist2");
              data_moist2[data_indx] = dato;
            break;

            case 3:
              Serial.println("light");
              data_light[data_indx] = dato;
            break;

            case 4:
              Serial.println("RH");
              data_RH[data_indx] = dato;
            break;
            
            }
            data_indx++;  
            comIN = Serial.read(); // just reads the remaining ';' symbol
           
      }
      
      if(eOF == true){ // if reached the '#' symbol, the transmission is over,
                       // print some of what you've got
                       
        Serial.println("Printing the first array:");
        for(int i = 0; i < sample_length; i++){
          Serial.println(data_temp[i]);
          }
        Serial.println("Printing the first array:");
        for(int i = 0; i < sample_length; i++){
          Serial.println(data_RH[i]);
          }
        break;
      }
    }
  } 
}

I did not manage to squeeze out any debug messages out of this program. I tried to upload a example sketch from the library to test, whether its not a HW problem or problem with the cable. It is not, I have uploaded the sketch, even connected to the wifi hotspot it created. So the upload procedure etc shoul be intact.

I will be very thankful for any information on this matter, that could help to resolve my desperate situation. Thank you! :heart: 

Andrej

earlephilhower, 2020-07-12:
@dhajnes, this tracker is for issues inside the core.  For help debugging your own code, please use https://www.esp8266.com or https://gitter.im/esp8266/Arduino

closed 2020-07-12 by earlephilhower

dhajnes, 2020-07-12:
@earlephilhower I believe, it is not as a code debuggin issue, as "what behaviour causes the cause 4, with watchdog" issue.
I believe there is something I have missed while programming, that is specific for the ESP chip. And thats why I decided to make a thread on this forum...

devyte, 2020-07-12:
Off the top of my head: replace while(Serial. available()) with if(Serial. available()) for starters, and improve error handling with a paranoid mindset. 
To be clear, this is not a discussion forum for general help or even general questions, but an issue tracker for tracking issues in the core. @earlephilhower 's advice is on point.