esp8266/Arduino #7450: HTTPS POST <body> is lost during redirect ESP2866/Arduino

GamerClassN7, 2020-07-13:
Hi,
i am working on custome SimpleHome (SmartHome Server) project i run in to the issue. If i want to use HTTPs, ESP lose request payload i realy try to do every possible alternative and nothing worked can someone sugest what i am dooing wrong ? i am running code on ESP2866 custome borad based on NodeMCU. Thanks for any answer.

### Description:
ESP is not able to send POST body on redirect, but if i test API with Insomnia of Curl everithing works fine.
### Server Response:
[image]

### Sketch:  https://raw.githubusercontent.com/Simple-Home/SmartHome-Firmwares/master/firmwares/PlatformIO/S20/src/main.cpp

liebman, 2020-07-13:
Some "spec" context for discussion: https://softwareengineering.stackexchange.com/questions/99894/why-doesnt-http-have-post-redirect

MarkyAD, 2020-07-14:
I see lots of 401 errors there. 401 = Unauthorized, likely a permissions problem. Your code should be checking http response codes and not trying to parse json objects when there's no object there to parse. This seems like user error and not a core issue.

GamerClassN7, 2020-07-15:
@MarkyAD That is the problem post body is lost during redirect from dodz contains token so it is not authorized on server.

MarkyAD, 2020-07-15:
You should've filled out the issue template you were presented with when you made this post, otherwise the devs probably wont even look at it. If you think this is a core problem then close this and make another post with all the needed info filled in, including the smallest possible complete sketch which demonstrates the problem.

closed 2020-07-20 by GamerClassN7