esp8266/Arduino #7457: ESP8266 httpUpdate binary does not fit flash Size

krupis, 2020-07-15:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:		                [ESP8266 (Node MCU-0.9)]
- Core Version:			[latest code from GIT]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Node MCU-0.9]
- Flash Mode:			[SPI]
- Flash Size: 		[4MB(FS:1MB OTA:~1019KB]
- lwip Variant:   [|v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|
- Upload Using:		[OTA&SERIAL]
- Upload Speed:   [115200]

### Problem Description

I am trying to upload sketch to my ESP8266 module OTA. I have managed to upload simple blink.bin file which is (256KB size), however, when I am trying to upload my actual sketch which is slightly larger (277KB) , I am getting error:
CALLBACK:  HTTP update process started
CALLBACK:  HTTP update fatal error code -107
HTTP_UPDATE_FAILD Error (-107): New Binary Does Not Fit Flash Size

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#ifndef APSSID
#define APSSID "ID"
#define APPSK  "PSWD"  
#endif

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(APSSID, APPSK);

}

void update_started() {
  Serial.println("CALLBACK:  HTTP update process started");
}

void update_finished() {
  Serial.println("CALLBACK:  HTTP update process finished");
}

void update_progress(int cur, int total) {
  Serial.printf("CALLBACK:  HTTP update process at %d of %d bytes...\n", cur, total);
}

void update_error(int err) {
  Serial.printf("CALLBACK:  HTTP update fatal error code %d\n", err);
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    // Add optional callback notifiers
    ESPhttpUpdate.onStart(update_started);
    ESPhttpUpdate.onEnd(update_finished);
    ESPhttpUpdate.onProgress(update_progress);
    ESPhttpUpdate.onError(update_error);

    t_httpUpdate_return ret = ESPhttpUpdate.update(client, "192.168.100.150",80,"/uploads/dev1.bin");
    // Or:
    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}

```

### Debug Messages
 Executable segment sizes:
IROM   : 268340          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27432   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1268  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1232  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25360 )         - zeroed variables      (global, static) in RAM/HEAP 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\petrikas.lu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\petrikas.lu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.3 in folder: C:\Users\petrikas.lu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.2\libraries\ESP8266httpUpdate 
"C:\\Users\\petrikas.lu\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-size" -A "C:\\Users\\petrikas.lu\\AppData\\Local\\Temp\\arduino_build_249815/httpUpdate_test.ino.elf"
Sketch uses 298272 bytes (28%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27860 bytes (34%) of dynamic memory, leaving 54060 bytes for local variables. Maximum is 81920 bytes.
C:\Users\petrikas.lu\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1/python3 C:\Users\petrikas.lu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.2/tools/upload.py --chip esp8266 --port COM12 --baud 115200 erase_flash --before default_reset --after hard_reset write_flash 0x0 C:\Users\petrikas.lu\AppData\Local\Temp\arduino_build_249815/httpUpdate_test.ino.bin 
esptool.py v2.8
Serial port COM12
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:05:23:03
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 9.9s
Compressed 302432 bytes to 220568...
Wrote 302432 bytes (220568 compressed) at 0x00000000 in 19.5 seconds (effective 124.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
CALLBACK:  HTTP update process started
CALLBACK:  HTTP update fatal error code -107
HTTP_UPDATE_FAILD Error (-107): New Binary Does Not Fit Flash Size
CALLBACK:  HTTP update process started
CALLBACK:  HTTP update fatal error code -107
HTTP_UPDATE_FAILD Error (-107): New Binary Does Not Fit Flash Size
CALLBACK:  HTTP update process started
..............

Debug messages go here
```

earlephilhower, 2020-07-26:
I am unable to reproduce this at all.  There is a chance that you've built for an ESP8266 flash map that doesn't have enough free space in it for your second app.  Check the `Tools->Flash Size` and make sure the OTA area is there.

I've just ran your test on a d1 mini 4MB (FS:1MB, OTA~1019k) and upgraded to a >500kb example from my ESP8266Audio library.  It pulled the update from my server and rebooted and started playing just fine.

````
[HTTP-Client][sendRequest] type: 'GET' redirCount: 0
[hostByName] Host: 192.168.1.8 is a IP!
:ref 1
[HTTP-Client] connected to 192.168.1.8:80
[HTTP-Client] sending request header
-----
GET /dev1.bin HTTP/1.0
Host: 192.168.1.8
User-Agent: ESP8266-http-Update
Connection: close
x-ESP8266-Chip-ID: 2340006
x-ESP8266-STA-MAC: 60:01:94:23:B4:A6
x-ESP8266-AP-MAC: 62:01:94:23:B4:A6
x-ESP8266-free-space: 2830336
x-ESP8266-sketch-size: 314240
x-ESP8266-sketch-md5: e0c134db4d1e1d88f912158ba90dedd6
x-ESP8266-chip-size: 4194304
x-ESP8266-sdk-version: 2.2.2-dev(38a443e)
x-ESP8266-mode: sketch
Content-Length: 0

-----
:wr 434 0
:wrc 434 434 0
:ack 434
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK
'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sun, 26 Jul 2020 19:37:08 GMT
'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache/2.4.29 (Ubuntu)
'
[HTTP-Client][handleHeaderResponse] RX: 'Last-Modified: Sun, 26 Jul 2020 19:37:07 GMT
'
[HTTP-Client][handleHeaderResponse] RX: 'ETag: W/"83430-5ab5d54dde92f"
'
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: bytes
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 537648
'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close
'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream
'
[HTTP-Client][handleHeaderResponse] RX: '
'
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 537648
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 537648
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2830336
[httpUpdate]  - current Sketch Size: 314240
CALLBACK:  HTTP update process started
[httpUpdate] runUpdate flash...
:pd 4, 2144, 270
:rpi 266, 4
sleep disable
[begin] roundedSize:       0x00084000 (540672)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x0004D000 (315392)
[begin] _startAddress:     0x0027C000 (2605056)
[begin] _currentAddress:   0x0027C000 (2605056)
[begin] _size:             0x00083430 (537648)
CALLBACK:  HTTP update process at 0 of 537648 bytes...
CALLBACK:  HTTP update process at 0 of 537648 bytes...
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
Header: 0xE9 1 2 40
CALLBACK:  HTTP update process at 4096 of 537648 bytes...
:c 1, 536, 2144

....

:rch 536, 536
:rch 1072, 536
:rch 1608, 536
CALLBACK:  HTTP update process at 528384 of 537648 bytes...
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
CALLBACK:  HTTP update process at 532480 of 537648 bytes...
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:rch 1072, 310
:rcl pb=0x3fff0484 sz=1382
CALLBACK:  HTTP update process at 536576 of 537648 bytes...
:c 1, 536, 1382
:c 1, 536, 846
:c0 1, 310
CALLBACK:  HTTP update process at 537648 of 537648 bytes...
CALLBACK:  HTTP update process at 537648 of 537648 bytes...
CALLBACK:  HTTP update process at 537648 of 537648 bytes...
Staged: address:0x0027C000, size:0x00083430
[httpUpdate] Update ok
[HTTP-Client][end] tcp is closed
CALLBACK:  HTTP update process finished
state: 5 -> 0 (0)
rm 0
....
````

closed 2020-07-26 by earlephilhower

KimAnhDuc, 2024-10-17:
I use ESP8266 and
I get an error when updating the firmware HTTP_UPDATE_FAILED Error (-107): New Binary Does Not Fit Flash Size while Sketch uses 326240 bytes (31%) of program storage space. Maximum is 1044464 bytes. Is there any way to fix it? Thank

vvlklvv, 2025-02-20:
See source code for ESP8266HTTPUpdate::Update

                    if (buf[0] == 0xe9) {
                        uint32_t bin_flash_size = ESP.magicFlashChipSize((buf[3] & 0xf0) >> 4);

                        // check if new bin fits to SPI flash
                        if(bin_flash_size > ESP.getFlashChipRealSize()) {
                            DEBUG_HTTP_UPDATE("[httpUpdate] New binary does not fit SPI Flash size\n");
                            _setLastError(HTTP_UE_BIN_FOR_WRONG_FLASH);
                            http.end();
                            return HTTP_UPDATE_FAILED;
                        }
                    }

This error occurs only in one case - when the memory size of the chip is smaller than the size for which the file is assembled. Inside the file, 3 bytes contains the memory size of the chip for which it is assembled. The memory map is not important, OTA, SPIFFS and so on do not matter! The file size also is not important! The total chip size only checked. For example, you upload file to 2 MB chip that is assembled for a 4 MB chip.