esp8266/Arduino #7466: Doubt about EEPROM and OTA

franciscogimeno2000, 2020-07-17:
Doubt about EEPROM and OTA

Hello my question is:
When OTA is done on a device, the EEPROM values (which are not really in the eeprom but in the program area) are overwritten? 

If so, how do we avoid removing them when doing OTA?
Where is the EEPROM data exactly?

Greetings and thank you
Francisco

Tech-TX, 2020-07-17:
AFAIK the only 'EEPROM values'  are the WiFi parameters,  and they aren't overwritten unless you do 'erase all'

franciscogimeno2000, 2020-07-17:
Hello, I mean the data that we keep with the eeprom instruction.

earlephilhower, 2020-07-17:
For general questions please use https://esp8266.com forums or https://gitter.im/esp8266/Arduino .  As the template you removed says, this area is for issues with the core and not general coding questions.

closed 2020-07-17 by earlephilhower