esp8266/Arduino #7470: Wifi connection very slow and not usable for esp8266 board core 2.6 and 2.7.  Works up to 2.5.2

jplavoiemtl, 2020-07-19:
Basic Infos

Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

Problem Description

Basic wifi sketch for esp8266 from board core ESP8266 is very slow to connect to wifi up to 20 seconds or more and sometimes no wifi connection at all.  The wifi is not stable in core 2.6 and 2.7.  It is working fine up to core 2.5.2 that I have to use because of this.

d-a-v, 2020-07-19:
I was trying to tell:

With 2.7.2, select the generic board in the tool menu, then select the "nonos sdk 2.2.1 legacy" firmware because that was the one which was used with 2.5.2.

But first you must try the tool menu "erase all flash" with your board.

jplavoiemtl, 2020-07-19:
Thank you, that worked well!

closed 2020-07-19 by earlephilhower

gordnhoo, 2022-08-08:
Im having this same problem but nothing fixed it