esp8266/Arduino #7473: FSInfo doesn’t update

zoidby, 2020-07-21:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			        ESP8266
- Core Version:			2.7.2
- Development Env:		Arduino IDE
- Operating System:		Arch Linux

### Settings in IDE

- Module:  			LOLIN(WEMOS) D1 mini Pro
- Flash Mode:			?
- Flash Size: 		       16 MiB
- lwip Variant:                  ?
- Reset Method:		?
- Flash Frequency:	        ?
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:              460800

### Problem Description

I wanted to use this code to monitor the current file system usage over the web:
```
  server.on("/fused", HTTP_GET, [](AsyncWebServerRequest *request){
    FSInfo fsInfo;
    LittleFS.info(fsInfo);
    request->send(200, "text/plain", String(fsInfo.usedBytes));
    });
```

Other code writes new data to the SPIFS periodically, but the value obtained via http://mcu-ip/fused does not update if i refresh the page. It only seems to update after a reset.

Am i using it wrong, is it a bug or is this the expected behaviour?

earlephilhower, 2020-07-22:
Can you please put an MCVE that repro's your problem, @zoidby ?  Something that writes free space before writing, say, a new 100k file, and then reports the new free space afterwards over the Serial port?

earlephilhower, 2020-07-26:
Closing as no MCVE was presented and testing shows this is not the case.  For small files, LFS is smart enough to place the actual file data right inside the directory entries, so creating small files may not cause space used to change.

````
#include <LittleFS.h>

void setup() {
  Serial.begin(115200);
  LittleFS.format();
  LittleFS.begin();
}

char buff[4096];
int cnt = 0;

void loop() {
  FSInfo fsInfo;
  LittleFS.info(fsInfo);
  Serial.printf("Count=%d, Used=%d\n", cnt++, fsInfo.usedBytes);
  sprintf(buff, "file%d.bin\n", cnt);
  File f = LittleFS.open(buff, "w");
  f.write((uint8_t*)buff, 4096);
  f.close();
  delay(1000);
}
````

````
Count=0, Used=16384
Count=1, Used=24576
Count=2, Used=32768
Count=3, Used=40960
Count=4, Used=49152
Count=5, Used=57344
Count=6, Used=65536
Count=7, Used=73728
````

closed 2020-07-26 by earlephilhower

zoidby, 2020-07-26:
I just came here to report that this issue can be closed as it was my own stupidity that caused it. But as it seems i was a couple of minutes late. But i still want to explain where my problem was. I wrote a data-logger that logs 32 bytes every minute and i expected the usedBytes to go up by that every minute. But i forgot about the block size of 8192 bytes ...

I am sorry that i wasted your time with this.

```
#include <LittleFS.h>

const char* filePath = "/testfile.txt";

void setup() {
  Serial.begin(1000000);
  delay(100);

  Serial.print(F("Inizializing LittleFS..."));
  if(LittleFS.begin()) {
    Serial.println(F(" done."));
  } else {
    Serial.println(F(" fail."));
    }

  // If it doesn’t exist, create new file
  if(!LittleFS.exists(filePath)) {
    File f = LittleFS.open(filePath, "w");
    if(!f) {
      Serial.print("Opening ");
      Serial.print(filePath);
      Serial.println(" for writing failed.");
    } else {
      f.println("");
      f.close();
      }
    }

  FSInfo fsInfo;
  LittleFS.info(fsInfo);
  Serial.print("Block size:       ");
  Serial.println(fsInfo.blockSize);
  }

void loop() {
  // Print fs-usage
  FSInfo fsInfo;
  LittleFS.info(fsInfo);
  Serial.print("FS-Used: ");
  Serial.println(fsInfo.usedBytes);
    
  File f = LittleFS.open(filePath, "a");
  if(!f) {
    Serial.print("Appending to ");
    Serial.print(filePath);
    Serial.println(" failed.");
  } else {
    f.println("aaaaaaaaaaaaaaaaaaaaaaaaaaaaa");
    Serial.print("Filesize: ");
    Serial.println(f.size());
    f.close();
    }

  delay(1000);
  }
```