esp8266/Arduino #7474: Test simple Captive Portal example and getting 2 lines at the top of Android phone display regarding connectvitycheck.gstatus.com

davidkjackson54, 2020-07-21:
I am testing the simple captive portal example and it is running fine without many modification except that on an Android phone 2 additional lines are displayed at the top.
**Sign into WiFIAP**
connecitivtycheck.gstatus,com

I have searched but not found any indication as to how to stop those lines from being displayed.
I suspect it is something to do with getting a 204 response but really not sure and I don't see anything in the documentation.

devyte, 2020-07-21:
Issue template ignored, which means almost no context. Please open a new issue and follow the template instructions.
Closing due to not compliant.

closed 2020-07-21 by devyte

davidkjackson54, 2020-07-21:
What more do I need to provide that you don't understand? 
I said I am using the sample captive portal unmodified.  is that not clear enough?

devyte, 2020-07-21:
No, it's not. The template exists for reasons.

JaosnHsieh, 2020-10-08:
> I am testing the simple captive portal example and it is running fine without many modification except that on an Android phone 2 additional lines are displayed at the top.
> **Sign into WiFIAP** [...]

Hi, where can we find "simple captive portal example"?