esp8266/Arduino #7475: ESP8266 webserver allows only 5 clients . how to increase the number of client (upto 20 clients)? (HTTP error : Connection Refused )

rohank27696, 2020-07-22:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266]
- Flash Mode:			[dio|]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4||Higher Bandwidth]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

ESP8266 webserver allows only 5 clients . how to increase the number of client (upto 20 clients)? (HTTP error : Connection Refused )

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2020-07-22:
`MAX_PENDING_CLIENTS_PER_PORT` must be globally defined to 20.
It is in WiFiServer.cpp.

rohank27696, 2020-07-22:
@d-a-v  Thanks for suggestion

But this is not working in my case.

Can you give me alternative  solution?

d-a-v, 2020-07-22:
Did you try to modify in wifiserver.cpp directly?

devyte, 2020-07-22:
Other way around, this issue isn't correct.

closed 2020-07-22 by devyte

rohank27696, 2020-07-22:
@d-a-v  first i removed the library from Arduino /libraries then i edited  wifiserver.cpp and added the library again to Arduino ide.

d-a-v, 2023-05-09:
Partly, the API has changed and we have now `WiFiServer::begin(port, 20)`, but we are still missing `ESP8266WebServer::begin(port, 20)`.
You can create a new issue to request for this change, or propose a pull-request.