esp8266/Arduino #7476: ESP8266 webserver allows only 5 clients . how to increase the number of client (upto 20 clients)? (HTTP error : Connection Refused )

rohank27696, 2020-07-22:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266]
- Flash Mode:			[dio|]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4||Higher Bandwidth]
- Reset Method:		[dtr]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

ESP8266 webserver allows only 5 clients . how to increase the number of client (upto 20 clients)? (HTTP error : Connection Refused )

```cpp

///////////////////////////////////CLIENT CODE///////////////////////////////////
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>

const char *ssid = "QWERTY";
const char *password = "Krishna12";

void setup() {

  Serial.begin(19200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid,password );

        while ((WiFiMulti.run() != WL_CONNECTED)) 
        {
           Serial.print("*");
        }
          
          Serial.println("connected\n\r");
      

}

void loop() {

      WiFiClient client;

      HTTPClient http1;
      char url[500]={0};
      char ipbuff[15]={0};

      sprintf(url,"http://192.168.4.1/dfgfdgfdgdfg");
      Serial.println(url);

      http1.setReuse(false);
      http1.setTimeout(2000);

      if (http1.begin(client,url))
      {
       
        int httpCode = http1.GET();

        Serial.printf("HTTP CODE == %d \n\r",httpCode);
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY)
            {
                String payload = http1.getString();
               
                Serial.println(payload);
          }
          else
          {
             
              Serial.println("HTTP ERROR CODE");
              Serial.printf("[HTTP] GET... failed, error: %s\n", http1.errorToString(httpCode).c_str()); 
 
                   
          }
       }
      else
      {
        Serial.println("HTTP Cannot connect");
      }   

   randNumber=random(0, 35); 
   Serial.println(randNumber*1000);
   client.flush();  
   client.stopAll();
   http1.end();  
   delay(randNumber*1000);

}

///////////////////////////////////SERVER CODE///////////////////////////////////
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include "LittleFS.h"
#include "FS.h"
#include <ESP8266WebServer.h>

extern "C" {
#include "espconn.h"
}

IPAddress staticIP(192, 168, 4, 1); //ESP8266 static ip
IPAddress gateway(192, 168, 4, 180); //IP Address of your WiFi Router (Gateway)
IPAddress subnet(255, 255, 255, 0); //Subnet mask
IPAddress dns(8, 8, 8, 8); //DNS

ESP8266WebServer server(80);

ESP8266WiFiMulti WiFiMulti;

void setup()
{
  Serial.begin(19200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid,password );

        while ((WiFiMulti.run() != WL_CONNECTED)) 
        {
           Serial.print("*");
        }
          
          Serial.println("connected\n\r");
  server.on("/inline",HTTP_GET,handleinlineresponse);
  server.begin();
}

void loop()
{
    server.handleClient();
}
void handleinlineresponse()
{
  String powerstatus = server.arg("deviceid"); 
 
  Serial.println(powerstatus);
  server.send(200, "text/html",powerstatus );
  return ;
  
}
```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2020-07-22:
Duplicate of #7475

devyte, 2020-07-22:
Why do you need 20 clients? Each accepted client takes up heap, I doubt the ESP's heap can handle that many at a time.

d-a-v, 2020-07-22:
It's few hundreds of bytes per connection.
Ref #7475 for how to change the defaults.
The number should be at least the number of hardware devices likely to connect to the server (+some)

d-a-v, 2020-07-22:
... Unless clients retry after a random time when the connection is refused. Or combination of the two. 
This limitation prevents esp to miserably fail when incoming connection rate is greater than connection service time.

rohank27696, 2020-07-22:
@devyte  thanks for replying . i want to connect clients one by one(disconnect one client and connect another one).but i am unable to acheive it

rohank27696, 2020-07-22:
@d-a-v  thanks for replying . i am using random delay for retry.

rohank27696, 2020-07-22:
@d-a-v  @devyte  i have also tried once client has got http response 200 i am disconnecting it  and keeping it in continous loop.
still i am facing same issue

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>

const char *ssid = "QWERTY";
const char *password = "Krishna12";

int passflag =0 ;

void setup() {

  Serial.begin(19200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid,password );

        while ((WiFiMulti.run() != WL_CONNECTED)) 
        {
           Serial.print("*");
        }
          
          Serial.println("connected\n\r");
      

}

void loop() {

      WiFiClient client;

      HTTPClient http1;
      char url[500]={0};
      char ipbuff[15]={0};

      sprintf(url,"http://192.168.4.1/dfgfdgfdgdfg");
      Serial.println(url);

      http1.setReuse(false);
      http1.setTimeout(2000);

      if (http1.begin(client,url))
      {
       
        int httpCode = http1.GET();

        Serial.printf("HTTP CODE == %d \n\r",httpCode);
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY)
            {
                String payload = http1.getString();
               
                Serial.println(payload);
                passflag=1;

          }
          else
          {
             
              Serial.println("HTTP ERROR CODE");
              Serial.printf("[HTTP] GET... failed, error: %s\n", http1.errorToString(httpCode).c_str()); 
 
                   
          }
       }
      else
      {
        Serial.println("HTTP Cannot connect");
      }   

   randNumber=random(0, 35); 
   Serial.println(randNumber*1000);
   client.flush();  
   client.stopAll();
   http1.end();  
   delay(randNumber*1000);

  while(passflag)
  {
      Serial.println("HELLO");
      delay(2000);

  }

}

jandrassy, 2020-07-22:
so those are SoftAP clients - stations. they are set to 4 max at default and can be set to max 8 as WiFi.softAP parameter.
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html#softapgetstationnum

d-a-v, 2020-08-12:
Can you clarify "same issue" ?
Do you have 5 running loops, only one, or more ?
Your `passflag` seems to stop everything in the end of the first success loop.

Instead of 
```
  client.stopAll();
  http1.end();  
```
Can you do
```
  http1.end();  
  client.stop();
```

rohank27696, 2020-08-12:
@d-a-v  i have only one loop in code. i am dumping this code in 8 different modules . giving them different deviceid [ **sprintf(url,"http://192.168.4.1/inline?deviceid=3")**].
and i keeping only one server ( 8 clients and  1 server ).
only 5 clients are able to communicate with Server.  Other clients  i  am gettting HTTP error : Connection Refused .

d-a-v, 2020-08-12:
The server works. I suspect your client is different from the client snippet above: this one does not work without a `delay()` in the connection loop.

Once fixed, I can run more than 5 clients without issues.
I suggest double checking that your clients call `client.stop()` after every http request.

Also please try and report whether your two MCVEs above work.

rohank27696, 2020-08-13:
@d-a-v  thank you for the solution. it Worked . All eight clients were able to communicate the server. 
/////////////////SERVER CODE/////////////////////////////////////////////////////////
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include "LittleFS.h"
#include "FS.h"
#include <ESP8266WebServer.h>

extern "C" {
#include "espconn.h"
}

//IPAddress staticIP(192, 168, 4, 1); //ESP8266 static ip
//IPAddress gateway(192, 168, 4, 180); //IP Address of your WiFi Router (Gateway)
//IPAddress subnet(255, 255, 255, 0); //Subnet mask
//IPAddress dns(8, 8, 8, 8); //DNS

const char *ssid = "GBTL";
const char *password = "Krishna12";

ESP8266WebServer server(80);

ESP8266WiFiMulti WiFiMulti;

void setup()
{
  Serial.begin(19200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid,password );

        while ((WiFiMulti.run() != WL_CONNECTED)) 
        {
           Serial.print("*");
           delay(500);
        }
          
         Serial.println("connected\n\r");

  Serial.println("NETWORK-INFO ");

  Serial.print("Connected to:");
  Serial.println(WiFi.SSID());
  Serial.print("IP Address:");
  Serial.println(WiFi.localIP());         
          
  server.on("/inline",HTTP_GET,handleinlineresponse);
  
  server.begin();
}

void loop()
{
    server.handleClient();
}

void handleinlineresponse()
{
  String powerstatus = server.arg("deviceid"); 
 
  Serial.println(powerstatus);
  server.send(200, "text/html",powerstatus );
  return ;  
}

//////////////////////////////////// CLIENT CODE//////////////////////////////////////////////
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>

const char *ssid = "GBTL";
const char *password = "Krishna12";

int passflag =0 ;

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(19200);
  Serial.println();
  
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid,password );

        while ((WiFiMulti.run() != WL_CONNECTED)) 
        {
           Serial.print("*");
           delay(500);
        }
          
         Serial.println("connected\n\r");
      

}

void loop() {

      WiFiClient client;

      HTTPClient http1;
      char url[500]={0};
      char ipbuff[15]={0};

      sprintf(url,"http://192.168.0.12/inline?deviceid=11");
      Serial.println(url);

      http1.setReuse(false);
      http1.setTimeout(2000);

      if (http1.begin(client,url))
      {
       
        int httpCode = http1.GET();

        Serial.printf("HTTP CODE == %d \n\r",httpCode);
            if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY)
            {
                String payload = http1.getString();
               
                Serial.println(payload);
                passflag=1;

          }
          else
          {
             
              Serial.println("HTTP ERROR CODE");
              Serial.printf("[HTTP] GET... failed, error: %s\n", http1.errorToString(httpCode).c_str()); 
 
                   
          }
       }
      else
      {
        Serial.println("HTTP Cannot connect");
      }   

    int randNumber=random(0, 35); 
   Serial.println(randNumber*1000);
   client.flush();  
   client.stopAll();
   http1.end();  
   delay(randNumber*1000);

  while(passflag)
  {
      Serial.println("HELLO");
      delay(2000);

  }
}

d-a-v, 2020-08-13:
Thanks for the feedback !

closed 2020-08-13 by d-a-v