esp8266/Arduino #7477: Example for Streaming HTTPS not connecting after get request

michaeloconnor1717, 2020-07-23:
- Hardware:			        [NodeMCU 1.0 ESP-12E]
- Core Version:			[2.7.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows10]
### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

I was just trying to run the streaming HTTPS example of the https client and keep getting the error:
"Connection failed"
I would really like some help explaining what I am doing wrong or if this is a problem with the example.

Thanks in advance

```cpp

#include <Arduino.h>

/**
   StreamHTTPClient.ino
    Created on: 24.05.2015
*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("Hogwarts", "Alohomora6444");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    std::unique_ptr<BearSSL::WiFiClientSecure> client(new BearSSL::WiFiClientSecure);

    bool mfln = client->probeMaxFragmentLength("tls.mbed.org", 443, 1024);
    Serial.printf("\nConnecting to https://tls.mbed.org\n");
    Serial.printf("Maximum fragment Length negotiation supported: %s\n", mfln ? "yes" : "no");
    if (mfln) {
      client->setBufferSizes(1024, 1024);
    }

    Serial.print("[HTTPS] begin...\n");

    // configure server and url
    const uint8_t fingerprint[20] = {0xEB, 0xD9, 0xDF, 0x37, 0xC2, 0xCC, 0x84, 0x89, 0x00, 0xA0, 0x58, 0x52, 0x24, 0x04, 0xE4, 0x37, 0x3E, 0x2B, 0xF1, 0x41};

    client->setFingerprint(fingerprint);

    HTTPClient https;

    if (https.begin(*client, "https://tls.mbed.org/")) {

      Serial.print("[HTTPS] GET...\n");
      // start connection and send HTTP header
      int httpCode = https.GET();
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK) {

          // get lenght of document (is -1 when Server sends no Content-Length header)
          int len = https.getSize();

          // create buffer for read
          static uint8_t buff[128] = { 0 };

          // read all data from server
          while (https.connected() && (len > 0 || len == -1)) {
            // get available data size
            size_t size = client->available();

            if (size) {
              // read up to 128 byte
              int c = client->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

              // write it to Serial
              Serial.write(buff, c);

              if (len > 0) {
                len -= c;
              }
            }
            delay(1);
          }

          Serial.println();
          Serial.print("[HTTPS] connection closed or file end.\n");

        }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }

      https.end();
    } else {
      Serial.printf("Unable to connect\n");
    }
  }

  Serial.println("Wait 10s before the next round...");
  delay(10000);
}
```

### Debug Messages

```
19:49:31.422 -> [SETUP] WAIT 3...
19:49:32.442 -> [SETUP] WAIT 2...
19:49:33.428 -> [SETUP] WAIT 1...
19:49:34.925 -> 
19:49:34.925 -> Connecting to https://tls.mbed.org
19:49:34.925 -> Maximum fragment Length negotiation supported: yes
19:49:34.959 -> [HTTPS] begin...
19:49:34.959 -> [HTTPS] GET...
19:49:38.330 -> [HTTPS] GET... failed, error: connection failed
19:49:38.330 -> Wait 10s before the next round...
19:49:48.778 -> 
19:49:48.778 -> Connecting to https://tls.mbed.org
19:49:48.778 -> Maximum fragment Length negotiation supported: yes
19:49:48.778 -> [HTTPS] begin...
19:49:48.778 -> [HTTPS] GET...
19:49:49.390 -> [HTTPS] GET... failed, error: connection failed
19:49:49.390 -> Wait 10s before the next round...
19:49:59.806 -> 
```

earlephilhower, 2020-07-23:
You need to run the example at 160MHz.  ARM uses EC keys for this site, and they take longer than RSA to compute so the ARM server times out their end while the 8266 is still decoding things.

For most SSL connections you'll want to be at 160, though, as the handshake can be 5 seconds or more at 80mhz which is marginal for many servers.

Also, this showed an issue in the ESP8266 code due to #7464 .  See #7478 for the fix, if it's not already merged when you check.

closed 2020-07-23 by earlephilhower

Ale9806, 2021-07-21:
Still Fails at 160MHz

borhanbjit, 2022-01-18:
I am same here