esp8266/Arduino #7481: Build errors xtensa-lx106-elf-gcc: error: unrecognized command line option '-std=c17'

TD-er, 2020-07-23:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Trying to build the 'staging' version of the esp8266/Arduino code and my compiler keeps complaining about not recognizing the `-std=c17` flag.

`xtensa-lx106-elf-gcc: error: unrecognized command line option '-std=c17'`

Tried in PlatformIO several things, like changing the `platform` version.

The current version I'm having (which is not working):

```
platform                  = espressif8266@2.6.0
platform_packages         =
	framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
build_flags               = ${esp82xx_2_6_x.build_flags}
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122
                            -Wno-deprecated-declarations
                            -std=gnu99	
                            -std=c++11
build_unflags             = -std=c17	
                            -std=gnu++17
```

So my question is, what should be done to make it compile in Windows?
By the way, tried it in Linux and without any extras installed (not tested any) it will also not compile there.

MarkyAD, 2020-07-23:
That would mean its using the wrong compiler. You probably have other releases installed and its finding that compiler first.

TD-er, 2020-07-23:
Any idea how to fix that?
My assumption is that it would be included in the toolset that is installed by PlatformIO, as I never installed a compiler specifically.
Also it should be possible to run several compiler versions on the same build machine as I don't want to break anything using the older core lib versions.

MarkyAD, 2020-07-23:
On linux I had remnants of an old arduino-ide install left over in ~/.arduino15 and ~/.arduino15/packages. rm'ing them solved it.

TD-er, 2020-07-23:
Well I'm not using Arduino IDE, but PlatformIO, which usually does a good job in separating things per build env declared in the platformio.ini file.

MarkyAD, 2020-07-23:
And Makefiles do that even better for me, Find your conflicting compiler and youll solve your problem

TD-er, 2020-07-23:
Strictly speaking, you may be right, but the problem is that it is all installed "automagically" by PlatformIO and even though I may be able to fix it for my specific build environment.
But that's going to cause dozens of reported issues of users of my project that are unable to compile it out of the box.
So I want it to be fixed in the toolchain, hence this issue.

MarkyAD, 2020-07-23:
That would indicate a platformIO issue then. Maybe ask there since the latest git master and releases here are working fine and have nothing to do with any "staging" releases.

TD-er, 2020-07-23:
Well something has changed here, as it was working well and some merge to the "master" branch included some build flags what made it cause to no longer build.

It could very well be PlatformIO related, but I am simply not sure.
So as it got "broken" by some changes here, that's where I first start.

MarkyAD, 2020-07-23:
The merge was switching to gcc-10 here, which works fine if you use the correct compiler.

closed 2020-07-23 by TD-er

MarkyAD, 2020-07-23:
Guess you got pointed in the right direction eh?

TD-er, 2020-07-23:
Not sure, I opened an issue at the PlatformIO repo: https://github.com/platformio/platform-espressif8266/issues/224

So we'll see if I got redirected again to the right repo as that happens quite often with these kind of issues. :(

MarkyAD, 2020-07-23:
Sounds to me like platformIO just needs gcc-10 installed. Hope it works out.

mathertel, 2020-09-11:
After deleting the esp8266 board package using the Board manager there were still files in my folder
C:\Users\<me>\AppData\Local\Arduino15\packages.
After deleting these files the compiler warning disappeared.