esp8266/Arduino #7489: https post with ESP8266

masoud935, 2020-07-27:
#### Platform

- Hardware:					[NodeMCU 1.0-12E]
- Core Version:			[2.7.2]
- Development Env:		[Arduino IDE-1.8.13]
- Operating System:		[Windows7-64bit]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			***
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		***
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I loaded the sample library program on the nod.
But the connection to the website failed and I received an "connection failed
" error message.
This is exactly the example "ESP8266WiFi.h".
I even changed the fingerprint to new values, but it did not change.

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#ifndef STASSID
#define STASSID "****"
#define STAPSK  "*********"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const char* host = "api.github.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char fingerprint[] PROGMEM = "B4 F5 71 B4 C0 F3 41 62 77 35 05 AC B2 82 7D 75 C3 09 EF 08";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);

  Serial.printf("Using fingerprint '%s'\n", fingerprint);
  client.setFingerprint(fingerprint);

  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "/repos/esp8266/Arduino/commits/master/status";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
    Serial.println("esp8266/Arduino CI successfull!");
  } else {
    Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}

void loop() {
}

```

### Debug Messages

```
Debug messages go here
```

earlephilhower, 2020-07-27:
The fingerprint you've shown is not correct for api.github.com.  Please use head and if the example doesn't work please submit a PR with the updated fingerprint.  GitHub changes certs on an almost daily basis and fingerprints are not the best way to connect to them.

earlephilhower, 2020-07-27:
Ah, I see you were running an unmodified example which has an old FP for API.GITHUB.COM.  That example, `HTTPSRequest.ino` needs to be changed to just use the CA because GH seems to change their certs very frequently.  Reopening as a real issue.

masoud935, 2020-07-28:
I used the new Github fingerprint. But I still have a problem with this code.
Is there another solution to use the post method for https without using a fingerprint?

earlephilhower, 2020-07-28:
For general coding issues, please use the forums at https://www.esp8266.com or Gitter https://gitter.im/esp8266/Arduino

closed 2020-07-28 by earlephilhower

mbm60, 2020-07-28:
@masoud935 
Masoud jan lotfan be jaye finger print az certificate estefade kon ta tarikh enghezaye roozane nadashte bashe.

masoud935, 2020-07-29:
I used the new Github fingerprint. But I still have a problem with this code.
Is there another solution to use the post method for https without using a fingerprint?

> 
> 
> @masoud935
> Masoud jan lotfan be jaye fingerprint az certificate estefade kon ta tarikh enghezaye roozane nadashte bashe.
tanks. but...
1-man az fingerprint jadid estefade kardam vali bazam kar nemikone. yani mottasel nemishe.
2-lotfan nemone code certificate ke gofti ro bede. man chizi be in esm peyda nakardam.

mbm60, 2020-07-29:
>  1-man az fingerprint jadid estefade kardam vali bazam kar nemikone. yani mottasel nemishe.
>  2-lotfan nemone code certificate ke gofti ro bede. man chizi be in esm peyda nakardam.

Fingerprint har rooz taghir mikone. PR #7490 ke be in issue montasab shode, taghirat lazem ro roye mesale "HTTPSRequest.ino" emaal mikone. version github esp8266 ro update kon va az mesale jadid estefade kon.

PR #7490 (2020-07-28): Use root cert, not fingerprint for api.github.com
The HTTPS example we were using a fingerprint which seems to change
daily as the github.com certificates are regenerated.  Replace this with
a trust anchor based on the ultimate root CA that github.com uses to
sign their certificates.  Assuming they don't change CAs, this
certificate should be good until 2030+

Fixes #7489
commit 63b41bcf (2020-07-28): Use root cert, not fingerprint for api.github.com (#7490)

In the HTTPS example we were using a fingerprint which changes almost
daily as the github.com certificates are regenerated.  Replace this with
a trust anchor based on the ultimate root CA that github.com uses to
sign their certificates.  Assuming they don't change CAs, this
certificate should be good until 2030+

Fixes #7489