esp8266/Arduino #7493: Unexpected behavior when fetching URL using BearSSL::WiFiClientSecure

RodrigoATorres, 2020-07-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[NodeMCU 1.0 ESP-12E]
- Core Version:			[d5187790bd09f58c98603922fab9d2806388751e]
- Development Env:		[Arduino IDE]
- Operating System:		[Debian]

### Settings in IDE

- Module:  			[NodeMCU 1.0 ESP-12E]
- Flash Mode:			[dio]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v1.4|v2 Lower Memory|Higher Bandwidth] (couldn't find)
- Reset Method:		[nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

I was trying to setup a SSL connection using [this example](https://github.com/esp8266/Arduino/blob/92373a98370618dea09718010b30d311a97f3f25/libraries/ESP8266WiFi/examples/BearSSL_Sessions/BearSSL_Sessions.ino) as base. I tried to create a separate function for setting the certificates, in order to keep the code better organized. But when I do that, I start to get an error when running the code.
The most unexpected was that if I call "fetchURL" inside "setupSSL" the code run just fine, but when I call it from outside it throws the error. I don't have a big experience with C++, but I expected both to have the exact same behavior. Couldn't find anything similar on my research both on esp8266 and C++ forums.
Thank you!

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <time.h>

const char *ssid = "myssid";
const char *pass = "mypassword";

const char *   host = "api.github.com";
const uint16_t port = 443;
const char *   path = "/";

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\nConnected");

  // Set up time to allow for certificate validation
  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print("Waiting for NTP time sync: ");
  time_t now = time(nullptr);
  while (now < 8 * 3600 * 2) {
    delay(500);
    Serial.print(".");
    now = time(nullptr);
  }
  Serial.println("");
  struct tm timeinfo;
[... 238 lines cut ...]
```

Decoding stack results
```
0x40225b0d: br_x509_minimal_run at src/x509/x509_minimal.c line 1612
0x40225d82: xm_append at src/x509/x509_minimal.c line 284
0x4021fe74: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 1874
0x4021e335: br_ssl_engine_recvrec_ack at src/inner.h line 553
0x4021deac: jump_handshake at src/inner.h line 2211
0x4021e441: br_ssl_engine_recvrec_ack at src/ssl/ssl_engine.c line 1206
0x402045ca: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 541
0x4021eae0: br_ssl_hs_client_init_main at src/ssl/ssl_hs_client.c line 904
0x4021eb30: br_ssl_hs_client_run at src/ssl/ssl_hs_client.c line 957
0x4021e5aa: br_ssl_engine_hs_reset at src/ssl/ssl_engine.c line 1319
0x402047ab: BearSSL::WiFiClientSecure::_wait_for_handshake() at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 564
0x40204993: BearSSL::WiFiClientSecure::_connectSSL(char const*) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 1122
0x402059c8: __esp_yield() at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 112
0x40206006: __delay(unsigned long) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.cpp line 54
0x40203566: WiFiClient::connect(IPAddress, unsigned short) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h line 144
0x40204a69: BearSSL::WiFiClientSecure::connect(char const*, unsigned short) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp line 231
0x402011bd: fetchURL(BearSSL::WiFiClientSecure*, char const*, unsigned short, char const*) at /home/rodrigo/Arduino/testing/testing.ino line 47
0x402013be: loop() at /home/rodrigo/Arduino/testing/testing.ino line 124
0x402052b8: Print::print(char const*) at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 129
0x40205adc: loop_wrapper() at /home/rodrigo/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 197
```

earlephilhower, 2020-07-30:
Per the documentation, the X509 certificate list must remain alive for the life of the connections using it.  Your code creates and destroys the certificate in a subroutine, so when BSSL tries to use it it hits a use-after-free error.  Make the X509List global.

closed 2020-07-30 by earlephilhower

RodrigoATorres, 2020-07-30:
Thank you very much.
Now I see that the documentation was pretty clear about that, I think I was looking at wrong spot.
Sorry about that.