esp8266/Arduino #7494: Cannot build with LittleFS support  (LittleFS.h not found)

TD-er, 2020-07-31:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[Core 2.7.3 and current master]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Problem Description

When trying to build (on a new computer), I now get these build errors:
```
src\src\Helpers\../../ESPEasy_common.h:26:24: fatal error: LittleFS.h: No such file or directory
```

The context of where this error is reported:
```c++
#ifdef USE_LITTLEFS
  #include <LittleFS.h>
  #define ESPEASY_FS LittleFS
#else 
  #ifdef ESP32
    #include <SPIFFS.h>
  #endif
  #define ESPEASY_FS SPIFFS
#endif
```

I have defined `USE_LITTLEFS` (otherwise I would not get the error) and in the PlatformIO.ini file this is set:
```
lib_ignore                = ESP32_ping, ESP32WebServer, ESP32HTTPUpdateServer, IRremoteESP8266, HeatpumpIR
lib_deps                  = ${esp82xx_common.lib_deps}, LittleFS(esp8266)
```

The file does exist in the path where PlatformIO does keep its framework downloads.
For me it is: `C:\Users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\LittleFS\src`
So it is not a matter of not all sub repositories being fetched.

Any idea what may be wrong here?

TD-er, 2020-07-31:
Just a short update, as a similar problem seems to occur with building ESP32, but then when including SPIFFS.h, which I'm 100% certain is correct as it was changed/fixed by me roughly a week ago on my previous computer.

Given some similar issues I found online, it looks like it may be a git issue not initializing all sub modules.
- [missing spiffs.h (IDFGH-1009)](https://github.com/espressif/esp-idf/issues/990)
- [Installation of esp8266 stage failed with git error](https://community.platformio.org/t/installation-of-esp8266-stage-failed-with-git-error/3184)
-[Error compiling arduino8266 framework](https://community.platformio.org/t/error-compiling-arduino8266-framework/12828)

v-a-d-e-r, 2020-07-31:
@TD-er: No issues here. Compiles like ever...
My env.: Arduino IDE, ESP8266, Core 2.7.3 and ESPeasy of course ;-)

TD-er, 2020-07-31:
And what version of Git?
And you do use LittleFS?

TD-er, 2020-07-31:
OK, part of the build issues do seem to be 'fixable' by me moving some other include that does include the LittleFS.h (or SPIFFS.h for ESP32).
So if that appears to be a bug, then it is a PlatformIO bug and not one for this repo.
But so far I am not able to build with LittleFS on core 2.7.3 but only on the stage branch (core 3.0.0)
And that does seem to be something related to this repo, right?

v-a-d-e-r, 2020-07-31:
Git version? I don't have to care about that!? It was directly offered in the IDE.
[image]
On a clean installed node with LittleFS it seems to run, but on a already running node I had some trouble. Only changing the FS seems no good idea... :-/

TD-er, 2020-07-31:
Well LittleFS is very (very) slow with saving the settings in the way how we use it in ESPEasy.
So that's why I don't have it enabled in the regular builds until that's been dealt with.

earlephilhower, 2020-08-01:
> OK, part of the build issues do seem to be 'fixable' by me moving some other include that does include the LittleFS.h (or SPIFFS.h for ESP32).
> So if that appears to be a bug, then it is a PlatformIO [...]

I'm not sure what the exact issue is, but we really didn't change anything w/LittleFS from the 2.7.2 (or .3) release to the current head.  Especially nothing related to the build system (only bugfixes inside which would be unrelated to PIO finding the headers/libs in the first place).  I'm closing for now so you can work w/the PIO guys and see what's going on.

We're also not planning on working on 2.7.3 anymore, so if 3.0.0-dev works you're golden.  Thx!

closed 2020-08-01 by earlephilhower