esp8266/Arduino #7495: Divide by Zero does not fail when numerator is Zero

JMWilton, 2020-08-01:
Dividing by zero should always throw an exception.  If the numerator is also zero it returns zero instead.

Sample Sketch to reproduce:

#include "Arduino.h"
//NodeMCU 0.9 ESP-12 Module

  long a;
  long b;
  long result;
  
void setup() 
{
  Serial.begin(115200);
  delay (5000);
  a = 0;
  b = 0;
  result = a/b;
  Serial.print("Result1 = ");
  Serial.println(result);
  a =  1;
  Serial.print("Result2 = ");
    delay(5000);
  result = a/b;
  Serial.println(result);
}

void loop()
{
  
}

earlephilhower, 2020-08-01:
Interesting observation!  Unfortunately, you didn't tell us what version you're running.  Is this 2.7.3 or git head?  

Something in GCC 10's intrinsics may have changed if it only fails under git master...

JMWilton, 2020-08-01:
I am using Arduino IDE 1.8.12 to build the code.
I am relatively new to working with ESP8266.  Can you point me in a direction to find the version number (2.7.3 or git head) you are needing?

earlephilhower, 2020-08-01:
Tools->Board->Boards Manager, then filter with "esp" and you should be able to find the number there.

earlephilhower, 2020-08-01:
With the git master on my own system, it hangs and crashes on the first division, as expected.

Please post your sketch output as well.

And I think you mean "Div 0 does not fail when *denominator* is zero"...

JMWilton, 2020-08-01:
2.6.3

earlephilhower, 2020-08-01:
Upgrade to 2.7.3, the latest release, and with that verison please post the output of the sketch, too, so we can see what you're seeing.

JMWilton, 2020-08-01:
Output from 2.7.3.
8:06:16.832 -> =======================================================================
18:06:16.832 -> Result1 = 0
18:06:16.832 -> Result2 = 
18:06:21.816 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:06:21.816 -> 
18:06:21.816 -> Exception (0):
18:06:21.816 -> epc1=0x4000dce5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
18:06:21.816 -> 
18:06:21.816 -> >>>stack>>>
18:06:21.816 -> 
18:06:21.816 -> ctx: cont
18:06:21.816 -> sp: 3ffffdf0 end: 3fffffc0 offset: 0190
18:06:21.850 -> 3fffff80:  3fffdad0 3ffee338 3ffee360 402010a0  
18:06:21.850 -> 3fffff90:  feefeffe feefeffe feefeffe 3ffee3c8  
18:06:21.850 -> 3fffffa0:  3fffdad0 00000000 3ffee388 40201a78  
18:06:21.850 -> 3fffffb0:  feefeffe feefeffe 3ffe84e0 40100b85  
18:06:21.850 -> <<<stack<<<
18:06:21.850 -> 
18:06:21.850 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
18:06:21.884 -> 
18:06:21.884 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
18:06:21.884 -> 
18:06:21.884 -> load 0x4010f000, len 3584, room 16 
18:06:21.884 -> tail 0
18:06:21.884 -> chksum 0xb0
18:06:21.884 -> csum 0xb0
18:06:21.884 -> v5d3af165
18:06:21.884 -> ~ld

Result1 shouldn't be zero but should throw an exception also

earlephilhower, 2020-08-01:
Can you try 1 change to see if it's a GCC core or intrinsics?

Change
````
long a;
long b;
long result;
````
to
````
volatile long a;
volatile long b;
volatile long result;
````

I'm thinking that because the code is so obvious in the sample that the GCC optimizer is constant folding.  Making things volatile will better simulate a division op that's not constant since `volatile` means it's got to re-read before use always and disallow that optimization.

JMWilton, 2020-08-01:
With the volatile change, we are now getting an exception of the first divide as expected.

earlephilhower, 2020-08-01:
Thanks for the update.

This is a GCC code optimizer issue then.  GCC10 (the git master) doesn't have it, so my best guess is that it's something fixed between 4.8 and 10.1.  There's nothing we can do in the core about it (master works and it's GCC related), so this looks like there's nothing we can do here.

#7496 will make the core actually dump a synthetic div-by-zero error in this case going forward.

closed 2020-08-01 by earlephilhower

JMWilton, 2020-08-01:
Thanks for investigating.