esp8266/Arduino #7497: BASIC HTTPS CLIENT example crash down on 2.7.3

i3water, 2020-08-02:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[2.7.3](https://github.com/esp8266/Arduino/releases/tag/2.7.3)
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[WiFiduino]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

code crash down when https request begin.
the same code on 2.7.2/2.7.1 don`t crash down.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

just use the [BasicHttpsClient](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino) example

### Debug Messages

```
13:37:28.360 -> [SETUP] WAIT 4...
13:37:29.360 -> [SETUP] WAIT 3...
13:37:30.359 -> [SETUP] WAIT 2...
13:37:31.355 -> [SETUP] WAIT 1...
13:37:32.351 -> [HTTPS] begin...
13:37:32.351 -> [HTTPS] GET...
13:37:37.501 -> [HTTPS] GET... failed, error: connection failed
13:37:37.501 -> Wait 10s before next round...
13:37:47.512 -> [HTTPS] begin...
13:37:47.512 -> [HTTPS] GET...
13:37:47.790 -> 
13:37:47.790 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
13:37:47.790 -> 
13:37:47.790 -> Exception (28):
13:37:47.790 -> epc1=0x40209ff0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00300030 depc=0x00000000
13:37:47.823 -> 
13:37:47.823 -> >>>stack>>>
13:37:47.823 -> 
13:37:47.823 -> ctx: cont
13:37:47.823 -> sp: 3ffffc30 end: 3fffffc0 offset: 0190
13:37:47.823 -> 3ffffdc0:  3ffef69c 3fff139c 3ffef3e4 40204247  
13:37:47.823 -> 3ffffdd0:  00000119 3ffee700 00000001 40207ebc  
13:37:47.823 -> 3ffffde0:  00000000 00000000 00000000 402084b6  
13:37:47.823 -> 3ffffdf0:  00000000 3ffef3e4 3ffef664 40202f10  
13:37:47.823 -> 3ffffe00:  3ffee5f8 00000d50 3ffffe78 00000000  
13:37:47.858 -> 3ffffe10:  000001bb 3ffef3e4 3ffef69c 00000000  
13:37:47.858 -> 3ffffe20:  000001bb 3ffef3e4 3ffef69c 4020441d  
13:37:47.858 -> 3ffffe30:  4020a6a0 15341e80 4020a6a0 15341e80  
13:37:47.858 -> 3ffffe40:  3ffef3e4 3ffffefc 3ffffef0 40205699  
13:37:47.858 -> 3ffffe50:  ffffffff 3fffff1c 3ffffef0 40205e49  
13:37:47.858 -> 3ffffe60:  70747400 00000073 80fffec0 00000000  
13:37:47.858 -> 3ffffe70:  00000000 00000000 3ffffef0 4020632b  
13:37:47.858 -> 3ffffe80:  3ffe8683 00000000 000a2e2e 40208ee9  
13:37:47.892 -> 3ffffe90:  3ffe8683 00000011 3ffffec0 00000001  
13:37:47.892 -> 3ffffea0:  3ffe8741 00000001 00000000 00000001  
13:37:47.892 -> 3ffffeb0:  4020678c 3ffee618 3ffe8674 40206798  
13:37:47.892 -> 3ffffec0:  4020678c 3ffee618 3ffe8674 00000001  
13:37:47.892 -> 3ffffed0:  3fffff7c 3ffef3e4 3ffee618 40206496  
13:37:47.892 -> 3ffffee0:  3fffff7c 3ffef3e4 3ffee618 40201147  
13:37:47.892 -> 3ffffef0:  00000000 00000000 3ffef3e4 3ffef69c  
13:37:47.892 -> 3fffff00:  000d000f 00201b83 000101bb 40001388  
13:37:47.927 -> 3fffff10:  3ffef6b4 0015001f 00000020 70747468  
13:37:47.927 -> 3fffff20:  3ffe0073 85fef8f4 00000000 40207c98  
13:37:47.927 -> 3fffff30:  80000020 3ffef4b4 0011001f 00201b83  
13:37:47.927 -> 3fffff40:  3fffff00 00000009 80fe8304 00000000  
13:37:47.927 -> 3fffff50:  00000000 00000000 ffffffff 3ffee600  
13:37:47.927 -> 3fffff60:  00000000 7e8c000a 3ffee500 40201c6d  
13:37:47.927 -> 3fffff70:  80000000 00000000 00000000 3ffef400  
13:37:47.962 -> 3fffff80:  002a002f 80000000 00000001 40100154  
13:37:47.962 -> 3fffff90:  3fffdad0 00000000 3ffee664 3ffee6a4  
13:37:47.962 -> 3fffffa0:  3fffdad0 00000000 3ffee664 40207fd4  
13:37:47.962 -> 3fffffb0:  feefeffe feefeffe 3ffe84e4 40100b85  
13:37:47.962 -> <<<stack<<<
13:37:47.962 -> 
13:37:47.962 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
13:37:47.997 -> 
13:37:47.997 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
13:37:47.997 -> 
13:37:47.997 -> load 0x4010f000, len 3584, room 16 
13:37:47.997 -> tail 0
13:37:47.997 -> chksum 0xb0
13:37:47.997 -> csum 0xb0
13:37:47.997 -> v5d3af165
13:37:47.997 -> ~ld
13:37:48.066 -> 
```

d-a-v, 2020-08-02:
Thanks, hotfix on the way...

earlephilhower, 2020-08-02:
@i3water can you pull the new 2.7.4 release and report back?

i3water, 2020-08-03:
@earlephilhower @d-a-v 
Thanks a lot, it can run on 2.7.4 release.

closed 2020-08-03 by earlephilhower