esp8266/Arduino #7498: mDNS not responding in WIFI_AP mode with ESP8266WebServer

ignazkevenaar, 2020-08-02:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266
- Core Version:			2.7.3
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

My goal: A standalone (not connected to another router or the internet) Wifi AP serving a configuration webpage on a `.local` address using mDNS.

I'm trying to setup mDNS on a WIFI_AP-mode ESP. The `ESP8266mDNS` -> `mDNS_Web_Server` example works when I convert it to AP mode. When I try to do the same thing with an ESP8266WebServer, the `.local` address will not respond to anything.

I've turned on "MDNS" in the "Debug Level options" to look for any suspicious messages, of which there is one pasted below. 

When I try to reach the url, you can see the debug messages come flowing in, so something is clearly happening.

Other things I've tried:
* Setting the second parameter of `MDNS.begin()` to `WiFi.softAPAP()`.
* Setting the Wifi hostname using `wifi_station_set_hostname("esp8266.local")`.
* Adding a service to the mDNS server using `MDNS.addService("http", "tcp", 80);`.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#ifndef STASSID
#define STASSID "ESPap"
#define STAPSK  "thereisnospoon"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

ESP8266WebServer server(80);

void setup(void) {
  Serial.begin(115200);

  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);

  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while (1) {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");

  server.on("/", []() {
    server.send(200, "text/plain", "hello world");
  });

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  MDNS.update();
}
```

### Debug Messages

When I turn on the MDNS debug level, the only "suspicious" output is this:

```
[MDNSResponder] _callProcess (18882, triggered by: 192.168.4.2)
[MDNSResponder] _parseMessage (Time: 18887 ms, heap: 45976 bytes, from 192.168.4.2(5353), to 224.0.0.251(5353))
[MDNSResponder] _readRRQuestion
[MDNSResponder] _readRRQuestion esp8266.local Type:0x0001 Class:0x0001 Multicast
[MDNSResponder] _replyMaskForHost: 0x1
[MDNSResponder] _parseQuery: Host reply needed 0x1
[MDNSResponder] _readRRQuestion
[MDNSResponder] _readRRQuestion esp8266.local Type:0x001C Class:0x0001 Multicast
[MDNSResponder] _parseQuery: Host reply needed 0x1
[MDNSResponder] _parseQuery: Known answers(1):
[MDNSResponder] _readRRAnswer: -empty- Type:0x0029 Class:0x05A0 TTL:4500, RDLength:18 generic 
[MDNSResponder] _parseQuery: Sending answer(0x1)...
[MDNSResponder] _sendMDNSMessage_Multicast: Will send to '224.0.0.251'.
[MDNSResponder] _prepareMDNSMessage
[MDNSResponder] _prepareMDNSMessage: ID:0 QR:1 OP:0 AA:1 TC:0 RD:0 RA:0 R:0 QD:0 AN:1 NS:0 AR:0
[MDNSResponder] _writeMDNSAnswer_A (192.168.4.1)
[MDNSResponder] _sendMDNSMessage_Multicast: FAILED!
[MDNSResponder] _sendMDNSMessage: FAILED!
[MDNSResponder] _parseQuery: FAILED!
[MDNSResponder] _parseMessage: Done (FAILED after 91 ms, ate 0 bytes, remaining 45976)
```

d-a-v, 2020-08-12:
There will soon be a new version of LEAmDNS which will be interface agnostic.
In the meantime, you can try current version with `MDNS.begin("esp8266", WiFi.softAPIP())`.

nikk16, 2020-08-12:
I think you need to use DNS for AP mode, not mDNS.

d-a-v, 2020-08-12:
> I think you need to use DNS for AP mode, not mDNS.

That is true. While mDNS can/should/will work on any network interface (STA, AP), resolving mDNS names on lightweight client devices may not be straightforward (is anyone using an ESP as AP for computers?).

A DNS server is available in this repo's libraries.

d-a-v, 2020-10-15:
@ignazkevenaar can you retry with current git version ?

ignazkevenaar, 2020-10-17:
@d-a-v Sorry for taking a while to get back to you!
I've tried with the current git version and I don't see any improvement using my MCVE sketch.
I have tried using the DNS example and setting the DNS server to my preferred domain name, which seems to work for my application. 👍

d-a-v, 2020-11-09:
> which seems to work for my application. :+1:

Thanks for the report !
I will close this issue then.

closed 2020-11-09 by d-a-v

nikk16, 2021-10-31:
> @ignazkevenaar I have the same trouble, can you give some example with dns host?

google it