esp8266/Arduino #7504: example code "HTTPSRequest.ino" got exception (9) in ESP8266

Furlly, 2020-08-05:
#### Platform

- Hardware:					[ESP-01]
- Core Version:			[2.7.3]
- Development Env:		[Arduino IDE-1.8.13]
- Operating System:		[Windows10-64bit]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[dtr]
- Flash Frequency:	[26Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I loaded the example code "HTTPSRequest.ino", changed the ssid and password of wifi, changed fingerprint of github to newest one. Meet Fatal exception 9(LoadStoreAlignmentCause) .

It is the limitation  of my ability of debug to find the program executing the "client.connect(host, httpsPort)", and goes to function “WiFiClientSecure::_connectSSL(const char* hostName)“ in file<WiFiClientSecurebearSSL.cpp>. 
Using DEBUG_BSSL, I found the program stopped during line 1096: "if (_sk && _sk->isRSA()) {", but don't know how to continue debugging and how to solve it. If I put "_sk->isRSA()" previous, the exception goes previous.
Appreciate if any help.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#ifndef STASSID
#define STASSID "XXXX"    //not showed
#define STAPSK  "XXXX"    //not showed
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const char* host = "api.github.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
//const char fingerprint[] PROGMEM = "5F F1 60 31 09 04 3E F2 90 D2 B0 8A 50 38 04 E8 37 9F BC 76";
const char fingerprint[] PROGMEM = "DB 6F DD BE 9A FE 42 0D 78 27 6D 5E 2D 04 44 A5 50 61 4C 63";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);

  Serial.printf("Using fingerprint '%s'\n", fingerprint);
  client.setFingerprint(fingerprint);

  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "/repos/esp8266/Arduino/commits/master/status";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
    Serial.println("esp8266/Arduino CI successfull!");
  } else {
    Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}

void loop() {

}

```

### Debug Messages

```
part of serial output:
WiFi connected
IP address: 
192.168.10.105
connecting to api.github.com
Using fingerprint 'DB 6F DD BE 9A FE 42 0D 78 27 6D 5E 2D 04 44 A5 50 61 4C 63'
[hostByName] request IP for: api.github.com
[hostByName] Host: api.github.com IP: 18.179.245.253
:ref 1
BSSL:_connectSSL: start connection
[BSSL_connection]:is_rsa:                 //this command I added is just before code "if (_sk && _sk->isRSA()) "
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x402091b8, epc2=0x00000000, epc3=0x00000000, excvaddr=0xfeefeffe, depc=0x00000000

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x402091b8 epc2=0x00000000 epc3=0x00000000 excvaddr=0xfeefeffe depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb70 end: 3fffffc0 offset: 0190
3ffffd00:  3ffe87c1 3ffee7c8 3ffffea0 4020467b  
3ffffd10:  3ffe0000 00000000 0000000a 40100c21  
3ffffd20:  00000000 00000000 3ffeeda8 4021ae14  
3ffffd30:  000008a0 3ffee8d8 3fff1508 40100c21  
3ffffd40:  00000014 3ffef9ac 000000ff 00000000  
3ffffd50:  00000006 3ffeed2c 00000020 40100f40  
3ffffd60:  40100b0c 00000008 00302064 3ffef9ac  
3ffffd70:  3ffef944 3ffeeda8 00000000 4021ae5c  
3ffffd80:  3ffeeda8 00000000 00000000 4010037a  
3ffffd90:  00000000 40212e60 00000359 4021ae82  
3ffffda0:  3ffeeda8 00000000 00000000 4021ba17  
3ffffdb0:  0002be31 0000005a 00000008 40216f66  
3ffffdc0:  3ffeeda8 00000000 00000001 40212d65  
3ffffdd0:  401054d1 0002db5d 4d916872 00000000  
3ffffde0:  007a1200 2cc5f6c5 0002db00 00000000  
3ffffdf0:  40105755 0002db5d 3ffee8b0 00000000  
3ffffe00:  3ffee1a0 3ffee8b0 00000001 3ffee7a8  
3ffffe10:  00000012 3ffee8b0 00000001 40206ff4  
3ffffe20:  00000000 00000001 3ffee7a8 40207602  
3ffffe30:  00000000 3ffffea0 3ffef72c 40202b48  
3ffffe40:  00000000 00000d50 00000014 000001bb  
3ffffe50:  3ffffea0 3ffe87c1 3ffee7c8 000001bb  
3ffffe60:  3ffffea0 3ffe87c1 00000001 4020490b  
3ffffe70:  40209810 fdf5b312 40209810 fdf5b312  
3ffffe80:  3ffe84dc 3ffee790 3ffee7c8 3ffe864b  
3ffffe90:  3ffe84dc 4025428a 3ffee7c8 40201168  
3ffffea0:  402096b0 00000000 00003a98 feefeffe  
3ffffeb0:  00000000 3ffef72c 3fff1654 3ffefaf4  
3ffffec0:  00000000 00000000 3fff1654 00000000  
3ffffed0:  00000000 3fff679c 3fff6d74 00000000  
3ffffee0:  00000000 3fff23b4 3ffefb1c 3fff650c  
3ffffef0:  3fff6774 00000000 00000000 00000000  
3fffff00:  00004145 00000255 feef0000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 6fdb0100 fe9abedd  
3fffff30:  27780d42 042d5e6d 6150a544 fe00634c  
3fffff40:  00000000 feefeffe 00000000 00000000  
3fffff50:  feefef00 00000000 00000000 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffee854  
3fffffa0:  3fffdad0 00000000 3ffee814 40207104  
3fffffb0:  feefeffe feefeffe 3ffe8500 40101231  
<<<stack<<<

```

d-a-v, 2020-08-05:
Can you retry with release 2.7.4 ?

Furlly, 2020-08-05:
@d-a-v 
Yeah，I tried the core version 2.7.4 and this issue was fixed.
Thank you for your advise!

closed 2020-08-05 by Furlly