esp8266/Arduino #7506: Suggestion: Setting HTTP_MAX_CLOSE_WAIT to 0 avoids Chrome network errors when web pages have external resources.

marciot, 2020-08-05:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git). _I have not tested, but I see HTTP_MAX_CLOSE_WAIT is still 2000 in the latest code._
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP32
- Core Version:			Unknown
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			ESP32 Dev Module
- Flash Mode:			QIO
- Flash Size: 		        4MB
- lwip Variant:                  Unknown
- Reset Method:		ck
- Flash Frequency:	        80Mhz
- CPU Frequency:		240MHz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I am building a project based on the SDWebServer example and I had small web page on the SD card that included links to four external files "js/main.css", "js/main.js", "js/jquery.js" and "js/delete.png".

I noticed that in Chrome, some resources would fail to load and I would see "ERR_CONNECTION_RESET" in the network panel. Firefox did not seem to have this problem.

[image]

I enabled core debugging messages and noticed the ESP32 wasn't even seeing requests for the missing files at all (see debug logs below). It seems that Chrome was fetching a new resource prior to the ESP32 having called `handleClient()`. This was made worse by the fact that it seemed like requests were taking quite a lot of time to close.

I started to dig through the source code of the ESP32 WebServer and I found a `HTTP_MAX_CLOSE_WAIT`. On a hunch, I set this to 0 figuring that if all data had been served, the ESP32 should begin waiting for the next connection rather than wait for the current client to disconnect.

This hunch seems to have been correct. My web page now completely loads without the errors I was seeing before and loaded a bit faster:

[image]

I do not whether there are any side effects to closing connections like I did, but since it helped with multi-file pages, I wanted to document it here.

I do realize that this fix is somewhat of a kludge. The fundamental problem that the ESP32 web server cannot handle multiple simultaneous connections; setting  HTTP_MAX_CLOSE_WAIT to zero does not solve this problem, but makes the problem less likely to happen.

Given that it should not cause any problems, and solves a few, I wonder whether HTTP_MAX_CLOSE_WAIT of 0 should be the default.

### Debug Messages

```
I (34) wifi: wifi driver task: 3ffb5784, prio:23, stack:3584, core=0
I (152) wifi: wifi firmware version: 7997e4b
I (152) wifi: config NVS flash: enabled
I (152) wifi: config nano formating: disabled
I (152) wifi: Init dynamic tx buffer num: 32
I (156) wifi: Init data frame dynamic rx buffer num: 32
I (161) wifi: Init management frame dynamic rx buffer num: 32
I (167) wifi: Init management short buffer num: 32
I (171) wifi: Init static rx buffer size: 1600
I (176) wifi: Init static rx buffer num: 16
I (179) wifi: Init dynamic rx buffer num: 32
I (236) wifi: mode : sta (30:ae:a4:1e:e1:18)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Connecting to mlt_dsl
I (363) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
I (363) wifi: state: init -> auth (b0)
[... 71 lines cut ...]
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.92
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: Pragma
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.105 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: */*
[V][Parsing.cpp:219] _parseRequest(): headerName: Referer
[V][Parsing.cpp:220] _parseRequest(): headerValue: http://192.168.1.92/
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /js/jquery.js
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[I][web-server.cpp:113] getFromSdCard(): WebApp: Rewrote path as /webapp/js/jquery.js
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
[D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
```

devyte, 2020-08-06:
I think you're at the wrong repo, this one is the Arduino core for ESP8266. 
Closing.

closed 2020-08-06 by devyte

marciot, 2020-08-06:
@devyte: Good point. Sorry about that.